neuropsychology/NeuroKit #218: EMG - Moar activation onsets detectors

DominiqueMakowski, 2020-05-21:
Stumbled across this nice script https://gist.github.com/tostasmistas/747f4585198411c8c4bda5f312f27dfb implementing several methods for EMG pulse onsets detection by @tostasmistas. This could definitely benefit from being packaged (unless its author has any reservations)! It could be an easy re-implementation task for contributors. 

- [ ] [unnamed method](https://gist.github.com/tostasmistas/747f4585198411c8c4bda5f312f27dfb#file-emg-py-L97) 
- [ ] [Hodges, 1996](https://gist.github.com/tostasmistas/747f4585198411c8c4bda5f312f27dfb#file-emg-py-L155) 
- [ ] [Bonato, 1998](https://gist.github.com/tostasmistas/747f4585198411c8c4bda5f312f27dfb#file-emg-py-L255) 
- [ ] [Lidierth, 1986](https://gist.github.com/tostasmistas/747f4585198411c8c4bda5f312f27dfb#file-emg-py-L416) 
- [ ] [Abbink, 1998](https://gist.github.com/tostasmistas/747f4585198411c8c4bda5f312f27dfb#file-emg-py-L574) 
- [ ] [Solnik, 2010](https://gist.github.com/tostasmistas/747f4585198411c8c4bda5f312f27dfb#file-emg-py-L730) 
- [ ] [Silva, 2012](https://gist.github.com/tostasmistas/747f4585198411c8c4bda5f312f27dfb#file-emg-py-L850) 
- [ ] [Londral, 2013](https://gist.github.com/tostasmistas/747f4585198411c8c4bda5f312f27dfb#file-emg-py-L934)

zen-juen, 2020-05-25:
Added the PELT method (https://arxiv.org/pdf/1101.1438.pdf) which identifies changepoints in a signal (see `nk.signal_changepoints()`) into `emg_activation()` as one of the methods to identify emg onsets and offsets. 

While `nk.signal_changepoints()` works quite nicely for the raw emg signal, as such:
```
emg = nk.emg_simulate(duration=10, burst_number=5)
nk.signal_changepoints(emg, change="var", show=True)
```
[image]

It doesn't quite as well for the linear envelope of emg due to the greater change in variance within activated periods.
```
# Simulate signal and obtain amplitude
cleaned = nk.emg_clean(emg)
emg_amplitude = nk.emg_amplitude(cleaned)
nk.signal_changepoints(emg_amplitude, change="var", show=True)
```
[image]

For now I set the onset/offset threshold as the minimum of all detected changepoints by PELT, which gives:
[image]

If anyone has any inputs on this matter / know of more optimal ways to use the PELT method for emg activation detection feel free to voice out!

Battlesenior, 2021-01-29:
Thank you for your contributions to EMG research!If you have a new method, please share!