neuropsychology/NeuroKit #219: Unused module import

vansjyo, 2020-05-22:
**Describe the bug**
An unused module has been imported in `NeuroKit/neurokit2/complexity/fractal_correlation.py`

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '[link'](https://github.com/neuropsychology/NeuroKit/blob/bf2be3b370b7edc8b12adac4e7508c29e26e4d94/neurokit2/complexity/fractal_correlation.py)
2. See line 7:
 https://github.com/neuropsychology/NeuroKit/blob/bf2be3b370b7edc8b12adac4e7508c29e26e4d94/neurokit2/complexity/fractal_correlation.py#L7.

**Expected behaviour**
Module not needed in the function hence it can be removed.

DominiqueMakowski, 2020-05-22:
Nice catch, want to make a PR to the dev branch to fix that?

vansjyo, 2020-05-22:
Sure, will make one.

vansjyo, 2020-05-22:
PR #220

closed 2020-05-22 by vansjyo