neuropsychology/NeuroKit #223: EDA features: aucorrelation, number of change points

DominiqueMakowski, 2020-05-23:
From [van Halem, S., van Roekel, E., Kroencke, L., Kuper, N., & Denissen, J. Moments That Matter? On the Complexity of Using Triggers Based on Skin Conductance to Sample Arousing Events Within an Experience Sampling Framework. European Journal of Personality](https://onlinelibrary.wiley.com/doi/full/10.1002/per.2252?campaign=wolearlyview):

> We used multilevel regression analyses to compare both categories on five different outcomes measures related to the skin conductance signal: **(i) mean, (ii) standard deviation, (iii) autocorrelation, (iv) number of peaks, and (v) number of change points**. Going beyond simply extracting mean levels of the signal, these parameters together better capture the complex and dynamic characteristics of physiological time series data. 

- [ ] Autocorrelation

> For the autocorrelation measure, to avoid ceiling effects (e.g. autoregressive correlations approaching 1), we used relatively broad lags of four seconds to calculate the autocorrelation. 

- [ ] Peak detection method

> We specified a peak when a consistent increase of 0.5 seconds would follow a consistent decrease of 0.5 seconds. Our method of identifying peaks is comparable with standard methods of extracting frequencies of non‐specific skin conductance responses, although in contrast these methods include absolute cut‐offs values, such as minimum amplitudes of 0.05 micro Siemens. 

- [ ] Number of change points

> Finally, we calculated the number of change points using the change point package in R. A change point is defined by a substantial change in the distributional properties of the skin conductance data in terms of mean and variance. We used the cpt.meanvar function with a penalty value of 10 000 (Killick & Eckley, 2014).

DominiqueMakowski, 2020-06-05:
@zen-juen where are we with this?

zen-juen, 2020-06-08:
@DominiqueMakowski finished with the last bit

added a new peaks detection method called `_eda_findpeaks_vanhalem2020` in `eda_findpeaks()`

zen-juen, 2020-06-24:
can close? @DominiqueMakowski 😄

DominiqueMakowski, 2020-06-24:
can can

closed 2020-06-24 by zen-juen