neuropsychology/NeuroKit #231: Poincaré plot improvements (HRV)

DominiqueMakowski, 2020-06-01:
I'm talking about the plot on the right:

[image]

We could improve it by:

- [ ] Add 2D density (Contour plot?) 
- [ ] Add marginal densities

[image]

I'd keep the points but would add the contour behind them. 

The main issue here is that while it is straightforward to do with [seaborn](https://python-graph-gallery.com/82-marginal-plot-with-seaborn/), we'd have to do it directly with matplotlib (because we don't want to add seaborn as a dep).

zen-juen, 2020-06-01:
Plotted scatterpoints only outside the boundaries of sd1 and sd2. kernel densities might be misleading about the spread of the data so I included the hist bars as well.

I still feel like it can be aesthetically improved though im not sure how @DominiqueMakowski thoughts? 😄 

[image]

DominiqueMakowski, 2020-06-01:
it's nice, I it could benefit from 1) lighter lower values that the backgrounds really whiter, 2) a bit bigger points with some alpha so that we can see overlaps, 3) restore the black oval outline from the Poincaré plot, 4) remove the outline of the boxes of the marginal densities, 5) make them closer to the figure, 6) display the figure title above the top marginal box (so that the title is for the "whole" fig)

DominiqueMakowski, 2020-06-01:
PS: this dataset is super misleading, better use ecg_simulate or else :)

zen-juen, 2020-06-01:
With `ecg_simulate()` : 😃 

```
ecg = nk.ecg_simulate(duration=240, method="ecgsyn")
peaks, info = nk.ecg_peaks(ecg, sampling_rate=1000)
hrv = nk.hrv_nonlinear(peaks, sampling_rate=1000, show=True)
```

[image]

DominiqueMakowski, 2020-06-01:
gorgeous

how does it with the others in `nk.hrv(show=True)`?

DominiqueMakowski, 2020-06-01:
very minor, can we make the histogram filled with #ccdff0 and the density line #1b6aaf 😬

zen-juen, 2020-06-01:
```
hrv_indices = nk.hrv(peaks, sampling_rate=1000, show=True)
```
will give: 
[image]

DominiqueMakowski, 2020-06-01:
👍 

How come the marginal plots still above title and far from plot? Isn't it generated from the same code?

zen-juen, 2020-06-01:
Because the axes (and hence including the title + gridspecs) for the poincare plot have to be redefined in `nk.hrv()` when combining all the plots together

Let me see if I can assign replicate the same result, but gridspec

closed 2020-06-02 by zen-juen