neuropsychology/NeuroKit #236: Should interpolated data be used in analysis?

Mitchellb16, 2020-06-02:
Apologies for the two issues in a row, just trying to clear this up for myself and a Google search didn't yield much. 

I noticed that here in rsp_intervalrelated.py:
`rate = data["RSP_Rate"].values`
`amplitude = data["RSP_Amplitude"].values`
`output["RSP_Rate_Mean"] = np.mean(rate)`
`output["RSP_Amplitude_Mean"] = np.mean(amplitude)`

Meaning all of the values, including the interpolated values, are being included in calculation of average amplitudes and rates. My understanding is that interpolation is more or less a very precise estimation that is helpful when we want to make a smooth plot, but intuitively it makes me think that any error in interpolation would skew the means of these two measures. Would it not be better to just use the values that are recorded at the peaks (which is where they are calculated)?

Thanks for any help you can give in understanding this!

DominiqueMakowski, 2020-06-03:
That's a good point. For ECG, the same is done, i.e., `ECG_Rate_Mean` is also computed on the interpolated signal, however, the raw mean of the RR intervals are obtained through the HRV indices (`MeanNN`).

I looked whether there is the equivalent for RSP, but not currently, so indeed we could add in `rsp_rrv()` the `MeanBB` index (the mean of the Breath-to-breath intervals).

The question then is whether we should remove the means of the rate (for ecg and rsp). While I agree that these are not the same, I also think that they give different info, one being the average of the intervals and the other one being the average *rate* (and the rate, although derived from the intervals, is a continuous process).

TLDR; I'll add the mean of the BB intervals in `rsp_rrv` as it's indeed missing, but won't remove the mean of the rate (unless there are other reasons for it) as this gives slightly different information (at least, *conceptually* different). What do you think?

Mitchellb16, 2020-06-05:
Okay that makes sense. I asked this because I've been seeing some pretty crazy interpolated rate values in my signal, being negative or very large (>10e6), but when I groupby(RSP_Peaks).get_group(1) and look at rates there they are much more realistic (but oddly enough I have still seen a couple of negative values). This may be more than one problem going on. I could maybe share a short file if you'd be interested in looking into it.

DominiqueMakowski, 2020-06-05:
Definitely, you can share the .txt data here if you want (or by email), I'll give it a look.

In theory interpolation should work fine and not produce any crazy values, because the interpolation *should* be bounded within the range of the 2 values it connects. So if you don't have crazy values in the first place I'm not sure how it interpolation can produce some, but maybe there's a well-hidden bug somewhere, we need to investigate 🕵️  

Also for reference, aside from being conceptually (and numerically) different, computing the mean of the rate instead of the mean of intervals has the advantage of "stability". For a given length of recording, the higher the heart rate (for instance), the more beats you have and the more "values" of interval you have. This, in turn, makes the standard deviation a function of the mean (i.e., the mean is more "precise" for higher rates), whereas this issue is not present for interpolated rates (because the number of data points is the same).

Mitchellb16, 2020-06-06:
I sent over a file to the email listed in your account. It's a tad large (100mb) as I wanted to make sure the issue was replicable (cleaning steps on a short segment of the file may yield different results than the entire file. Let me know if you have any questions/didn't receive it.

DominiqueMakowski, 2020-06-06:
received it 👌 , will look into it asap!

DominiqueMakowski, 2020-06-06:
closing here in favour of #242

closed 2020-06-06 by DominiqueMakowski