neuropsychology/NeuroKit #238: 🚀 we can fix these (pylint issues)

DominiqueMakowski, 2020-06-03:
```
2020-06-03T11:19:19.2764444Z ************* Module neurokit2
2020-06-03T11:19:19.2765749Z neurokit2/__init__.py:33:222: W1401: Anomalous backslash in string: '\c'. String constant might be missing an r prefix. (anomalous-backslash-in-string)
2020-06-03T11:19:19.2766385Z neurokit2/__init__.py:1:0: C0199: First line empty in module docstring (docstring-first-line-empty)
2020-06-03T11:19:19.2767000Z neurokit2/__init__.py:65:0: C0116: Missing function or method docstring (missing-function-docstring)
2020-06-03T11:19:19.2767210Z ************* Module neurokit2.stats.rescale
2020-06-03T11:19:19.2767783Z neurokit2/stats/rescale.py:5:0: W0102: Dangerous default value [] as argument (dangerous-default-value)
2020-06-03T11:19:19.2768379Z neurokit2/stats/rescale.py:44:0: W0102: Dangerous default value [] as argument (dangerous-default-value)
2020-06-03T11:19:19.2768591Z ************* Module neurokit2.stats.hdi
2020-06-03T11:19:19.2769144Z neurokit2/stats/hdi.py:9:0: W9015: "kwargs, show" missing in parameter documentation (missing-param-doc)
2020-06-03T11:19:19.2770185Z neurokit2/stats/hdi.py:9:0: W9016: "show" missing in parameter type documentation (missing-type-doc)
2020-06-03T11:19:19.2770814Z neurokit2/stats/hdi.py:68:4: W0612: Unused variable 'fig' (unused-variable)
2020-06-03T11:19:19.2771029Z ************* Module neurokit2.stats.fit_loess
2020-06-03T11:19:19.2771635Z neurokit2/stats/fit_loess.py:53:11: R1716: Simplify chained comparison between the operands (chained-comparison)
2020-06-03T11:19:19.2771837Z ************* Module neurokit2.stats.correlation
2020-06-03T11:19:19.2772411Z neurokit2/stats/correlation.py:7:0: W9015: "x, y" missing in parameter documentation (missing-param-doc)
2020-06-03T11:19:19.2773009Z neurokit2/stats/correlation.py:7:0: W9016: "x, y" missing in parameter type documentation (missing-type-doc)
2020-06-03T11:19:19.2773570Z neurokit2/stats/correlation.py:36:7: W0612: Unused variable 'p' (unused-variable)
2020-06-03T11:19:19.2773783Z ************* Module neurokit2.stats.fit_mixture
2020-06-03T11:19:19.2774354Z neurokit2/stats/fit_mixture.py:6:0: W9017: "threshold" differing in parameter documentation (differing-param-doc)
2020-06-03T11:19:19.2774976Z neurokit2/stats/fit_mixture.py:6:0: W9018: "threshold" differing in parameter type documentation (differing-type-doc)
2020-06-03T11:19:19.2775199Z ************* Module neurokit2.stats
2020-06-03T11:19:19.2775772Z neurokit2/stats/__init__.py:1:0: C0199: First line empty in module docstring (docstring-first-line-empty)
2020-06-03T11:19:19.2775988Z ************* Module neurokit2.stats.fit_polynomial
2020-06-03T11:19:19.2776599Z neurokit2/stats/fit_polynomial.py:63:2: W0511: TODO: add cross-validation or some kind of penalty to prevent over-fitting? (fixme)
2020-06-03T11:19:19.2777219Z neurokit2/stats/fit_polynomial.py:62:0: C0116: Missing function or method docstring (missing-function-docstring)
2020-06-03T11:19:19.2777440Z ************* Module neurokit2.stats.fit_error
2020-06-03T11:19:19.2778310Z neurokit2/stats/fit_error.py:83:0: C0116: Missing function or method docstring (missing-function-docstring)
2020-06-03T11:19:19.2778928Z neurokit2/stats/fit_error.py:87:0: C0116: Missing function or method docstring (missing-function-docstring)
2020-06-03T11:19:19.2779515Z neurokit2/stats/fit_error.py:91:0: C0116: Missing function or method docstring (missing-function-docstring)
2020-06-03T11:19:19.2780310Z neurokit2/stats/fit_error.py:92:4: R1705: Unnecessary "else" after "return" (no-else-return)
2020-06-03T11:19:19.2780530Z ************* Module neurokit2.stats.mutual_information
2020-06-03T11:19:19.2781115Z neurokit2/stats/mutual_information.py:8:0: W9015: "x, y" missing in parameter documentation (missing-param-doc)
2020-06-03T11:19:19.2781718Z neurokit2/stats/mutual_information.py:8:0: W9016: "x, y" missing in parameter type documentation (missing-type-doc)
2020-06-03T11:19:19.2782320Z neurokit2/stats/mutual_information.py:64:0: C0199: First line empty in function docstring (docstring-first-line-empty)
2020-06-03T11:19:19.2782943Z neurokit2/stats/mutual_information.py:118:0: C0199: First line empty in function docstring (docstring-first-line-empty)
2020-06-03T11:19:19.2783546Z neurokit2/stats/mutual_information.py:132:0: W9015: "X" missing in parameter documentation (missing-param-doc)
2020-06-03T11:19:19.2784150Z neurokit2/stats/mutual_information.py:132:0: W9016: "X" missing in parameter type documentation (missing-type-doc)
2020-06-03T11:19:19.2784761Z neurokit2/stats/mutual_information.py:155:4: W0105: String statement has no effect (pointless-string-statement)
2020-06-03T11:19:19.2785348Z neurokit2/stats/mutual_information.py:132:0: W9011: Missing return documentation (missing-return-doc)
2020-06-03T11:19:19.2785937Z neurokit2/stats/mutual_information.py:132:0: W9012: Missing return type documentation (missing-return-type-doc)
2020-06-03T11:19:19.2786150Z ************* Module neurokit2.stats.mad
2020-06-03T11:19:19.2786723Z neurokit2/stats/mad.py:32:4: W0621: Redefining name 'mad' from outer scope (line 5) (redefined-outer-name)
2020-06-03T11:19:19.2786939Z ************* Module neurokit2.epochs.epochs_plot
2020-06-03T11:19:19.2787256Z neurokit2/epochs/epochs_plot.py:85:2: W0511: TODO: Custom legend (fixme)
2020-06-03T11:19:19.2787989Z neurokit2/epochs/epochs_plot.py:55:4: R1705: Unnecessary "else" after "return" (no-else-return)
2020-06-03T11:19:19.2788220Z ************* Module neurokit2.epochs.eventrelated_utils
2020-06-03T11:19:19.2788833Z neurokit2/epochs/eventrelated_utils.py:34:0: W0102: Dangerous default value {} as argument (dangerous-default-value)
2020-06-03T11:19:19.2789470Z neurokit2/epochs/eventrelated_utils.py:73:0: W0102: Dangerous default value {} as argument (dangerous-default-value)
2020-06-03T11:19:19.2789670Z ************* Module neurokit2.epochs
2020-06-03T11:19:19.2790245Z neurokit2/epochs/__init__.py:1:0: C0199: First line empty in module docstring (docstring-first-line-empty)
2020-06-03T11:19:19.2790465Z ************* Module neurokit2.epochs.epochs_create
2020-06-03T11:19:19.2791131Z neurokit2/epochs/epochs_create.py:9:0: W9015: "baseline_correction, epochs_end, epochs_start" missing in parameter documentation (missing-param-doc)
2020-06-03T11:19:19.2791846Z neurokit2/epochs/epochs_create.py:9:0: W9016: "baseline_correction, epochs_end, epochs_start" missing in parameter type documentation (missing-type-doc)
2020-06-03T11:19:19.2792583Z neurokit2/epochs/epochs_create.py:88:7: R1701: Consider merging these isinstance calls to isinstance(data, (list, np.ndarray, pd.Series)) (consider-merging-isinstance)
2020-06-03T11:19:19.2792819Z ************* Module neurokit2.eeg.mne_channel_extract
2020-06-03T11:19:19.2793412Z neurokit2/eeg/mne_channel_extract.py:5:0: W9015: "name" missing in parameter documentation (missing-param-doc)
2020-06-03T11:19:19.2794037Z neurokit2/eeg/mne_channel_extract.py:5:0: W9016: "name" missing in parameter type documentation (missing-type-doc)
2020-06-03T11:19:19.2794678Z neurokit2/eeg/mne_channel_extract.py:5:0: W9017: "channel_names" differing in parameter documentation (differing-param-doc)
2020-06-03T11:19:19.2795321Z neurokit2/eeg/mne_channel_extract.py:5:0: W9018: "channel_names" differing in parameter type documentation (differing-type-doc)
2020-06-03T11:19:19.2795903Z neurokit2/eeg/mne_channel_extract.py:36:14: W0612: Unused variable 'time_index' (unused-variable)
2020-06-03T11:19:19.2796118Z ************* Module neurokit2.eeg
2020-06-03T11:19:19.2796806Z neurokit2/eeg/__init__.py:1:0: C0199: First line empty in module docstring (docstring-first-line-empty)
2020-06-03T11:19:19.2797023Z ************* Module neurokit2.eeg.mne_channel_add
2020-06-03T11:19:19.2797699Z neurokit2/eeg/mne_channel_add.py:6:0: W9015: "channel_name, sync_index_channel, sync_index_raw" missing in parameter documentation (missing-param-doc)
2020-06-03T11:19:19.2798394Z neurokit2/eeg/mne_channel_add.py:6:0: W9016: "channel_name, sync_index_channel, sync_index_raw" missing in parameter type documentation (missing-type-doc)
2020-06-03T11:19:19.2799013Z neurokit2/eeg/mne_channel_add.py:52:8: C0415: Import outside toplevel (mne) (import-outside-toplevel)
2020-06-03T11:19:19.2799232Z ************* Module neurokit2.rsp.rsp_fixpeaks
2020-06-03T11:19:19.2799827Z neurokit2/rsp/rsp_fixpeaks.py:6:0: W9015: "peaks, troughs" missing in parameter documentation (missing-param-doc)
[... cut ...]
2020-06-03T11:19:37.4021814Z neurokit2/ppg/ppg_simulate.py:259:2: W0511: TODO (fixme)
2020-06-03T11:19:37.4022108Z neurokit2/ppg/ppg_simulate.py:127:8: W0612: Unused variable 'fig' (unused-variable)
2020-06-03T11:19:37.4022494Z neurokit2/ppg/ppg_simulate.py:193:0: R1710: Either all return statements in a function should return an expression, or none of them should. (inconsistent-return-statements)
2020-06-03T11:19:37.4023089Z neurokit2/ppg/ppg_simulate.py:231:0: C0199: First line empty in function docstring (docstring-first-line-empty)
```

zen-juen, 2020-06-05:
Pylint detects both style and semantic errors, the former having been fixed by `flake8`. However, we had many semantic errors that have to be manually fixed. 

We disabled some errors (some of which are false positives) such as:
- **E0401**: import error (done and checked by `flake8`)
- **R1705**: unnecessary else after return
- **R1720**: unnecessary else after raise
- **R0913**: too many arguments in a function or method 
- **R0801**: duplicate code across multiple files
- **R0914**: too many local variables in a function or method
- **R0912**: too many branches in a function or method
- **R0915**: too many statements in a function or method
- **E203**: whitespace before ":"
- **E303**: Too many blank lines
- **E0401**: Unable to import
- **C0330**: bad continuation
- **C0114**: missing docstrings for modules
- **C0103**: name doesn't conform to naming rules of its type
- **C0415**: import used in places other than module toplevel (we have some in our files to check for missing necessary imports)
- **C0301**: line too long (`black` ensures all lines are within 120 characters with #noqa added to ignore long url lines but `pylint` still detects these urls)
- **C1801**: not preferable to use `len(sequence)` in certain conditions
- **C0111**: When a module/function/class has no docstring
- **C0330**: wrong hanging indentation before block (https://github.com/psf/black/issues/48 - we have codes like these too)
- **W0511**: checker that complains about "TODO" and similar terms in comments
- **W0102**: dangerous default value (when mutable value as list or dictionary is detected in a default value for an argument)
- **W9006**:  raises not documented as being raised

zen-juen, 2020-06-05:
We're now left with these:

(for some reason the kwargs error pops up even if **kwargs is in parameters) 🤔  
```
************* Module neurokit2.stats.hdi
neurokit2/stats/hdi.py:9:0: W9015: "kwargs" missing in parameter documentation (missing-param-doc)
************* Module neurokit2.stats.mutual_information
neurokit2/stats/mutual_information.py:161:4: W0105: String statement has no effect (pointless-string-statement)
************* Module neurokit2.rsp.rsp_simulate
neurokit2/rsp/rsp_simulate.py:105:0: C0199: First line empty in function docstring (docstring-first-line-empty)
************* Module neurokit2.rsp.rsp_intervalrelated
neurokit2/rsp/rsp_intervalrelated.py:112:8: W0101: Unreachable code (unreachable)
neurokit2/rsp/rsp_intervalrelated.py:120:8: W0101: Unreachable code (unreachable)
************* Module neurokit2.ecg.ecg_delineate
neurokit2/ecg/ecg_delineate.py:225:4: W0613: Unused argument 'debug' (unused-argument)
neurokit2/ecg/ecg_delineate.py:327:4: W0613: Unused argument 'debug' (unused-argument)
neurokit2/ecg/ecg_delineate.py:389:82: W0613: Unused argument 'debug' (unused-argument)
************* Module neurokit2.ecg.ecg_findpeaks
neurokit2/ecg/ecg_findpeaks.py:1:0: C0302: Too many lines in module (1056/1000) (too-many-lines)
neurokit2/ecg/ecg_findpeaks.py:523:13: R1716: Simplify chained comparison between the operands (chained-comparison)
neurokit2/ecg/ecg_findpeaks.py:542:13: R1716: Simplify chained comparison between the operands (chained-comparison)
neurokit2/ecg/ecg_findpeaks.py:680:13: R1716: Simplify chained comparison between the operands (chained-comparison)
************* Module neurokit2.complexity.utils
neurokit2/complexity/utils.py:72:19: E1136: Value 'embedded' is unsubscriptable (unsubscriptable-object)
************* Module neurokit2.complexity.entropy_fuzzy
neurokit2/complexity/entropy_fuzzy.py:6:0: W9015: "kwargs" missing in parameter documentation (missing-param-doc)
************* Module neurokit2.complexity.complexity_optimize
neurokit2/complexity/complexity_optimize.py:349:4: W0632: Possible unbalanced tuple unpacking with sequence defined at line 2 of : left side has 2 label(s), right side has 1 value(s) (unbalanced-tuple-unpacking)
neurokit2/complexity/complexity_optimize.py:417:8: R1723: Unnecessary "else" after "break" (no-else-break)
************* Module neurokit2.complexity.entropy_approximate
neurokit2/complexity/entropy_approximate.py:7:0: W9015: "kwargs" missing in parameter documentation (missing-param-doc)
************* Module neurokit2.complexity.complexity_dimension
neurokit2/complexity/complexity_dimension.py:9:0: W9015: "kwargs" missing in parameter documentation (missing-param-doc)
neurokit2/complexity/complexity_dimension.py:145:79: E1136: Value 'y2' is unsubscriptable (unsubscriptable-object)
neurokit2/complexity/complexity_dimension.py:151:24: E1136: Value 'y2' is unsubscriptable (unsubscriptable-object)
neurokit2/complexity/complexity_dimension.py:151:36: E1136: Value 'y2' is unsubscriptable (unsubscriptable-object)
neurokit2/complexity/complexity_dimension.py:155:0: W9015: "kwargs" missing in parameter documentation (missing-param-doc)
neurokit2/complexity/complexity_dimension.py:203:79: E1136: Value 'y2' is unsubscriptable (unsubscriptable-object)
neurokit2/complexity/complexity_dimension.py:206:16: E1136: Value 'y2' is unsubscriptable (unsubscriptable-object)
************* Module neurokit2.complexity.complexity_delay
neurokit2/complexity/complexity_delay.py:159:47: E1136: Value 'embedded' is unsubscriptable (unsubscriptable-object)
neurokit2/complexity/complexity_delay.py:159:63: E1136: Value 'embedded' is unsubscriptable (unsubscriptable-object)
neurokit2/complexity/complexity_delay.py:164:23: E1136: Value 'embedded' is unsubscriptable (unsubscriptable-object)
neurokit2/complexity/complexity_delay.py:202:8: E1136: Value 'embedded' is unsubscriptable (unsubscriptable-object)
neurokit2/complexity/complexity_delay.py:203:8: E1136: Value 'embedded' is unsubscriptable (unsubscriptable-object)
neurokit2/complexity/complexity_delay.py:204:8: E1136: Value 'embedded' is unsubscriptable (unsubscriptable-object)
************* Module neurokit2.complexity.complexity_simulate
neurokit2/complexity/complexity_simulate.py:5:0: W9015: "kwargs" missing in parameter documentation (missing-param-doc)
************* Module neurokit2.complexity.fractal_dfa
neurokit2/complexity/fractal_dfa.py:8:0: W9015: "kwargs" missing in parameter documentation (missing-param-doc)
neurokit2/complexity/fractal_dfa.py:9:0: W0613: Unused argument 'kwargs' (unused-argument)
************* Module neurokit2.complexity.entropy_sample
neurokit2/complexity/entropy_sample.py:6:0: W9015: "kwargs" missing in parameter documentation (missing-param-doc)
************* Module neurokit2.complexity.entropy_multiscale
neurokit2/complexity/entropy_multiscale.py:9:0: W9015: "kwargs" missing in parameter documentation (missing-param-doc)
neurokit2/complexity/entropy_multiscale.py:116:0: W0613: Unused argument 'kwargs' (unused-argument)
************* Module neurokit2.signal.signal_power
neurokit2/signal/signal_power.py:10:0: W9015: "kwargs" missing in parameter documentation (missing-param-doc)
************* Module neurokit2.signal.signal_fixpeaks
neurokit2/signal/signal_fixpeaks.py:12:0: R1710: Either all return statements in a function should return an expression, or none of them should. (inconsistent-return-statements)
************* Module neurokit2.signal.signal_plot
[... cut ...]
neurokit2/hrv/hrv_frequency.py:179:4: E1123: Unexpected keyword argument 'sampling_rate' in function call (unexpected-keyword-arg)
************* Module neurokit2.data.data
neurokit2/data/data.py:40:8: W0621: Redefining name 'data' from outer scope (line 7) (redefined-outer-name)
neurokit2/data/data.py:44:4: W0612: Unused variable 'file' (unused-variable)
************* Module neurokit2.ppg.ppg_simulate
neurokit2/ppg/ppg_simulate.py:230:0: C0199: First line empty in function docstring (docstring-first-line-empty)
```

DominiqueMakowski, 2020-06-09:
Has been fixed

closed 2020-06-09 by DominiqueMakowski