neuropsychology/NeuroKit #245: Benchmarking ECG preprocessing

DominiqueMakowski, 2020-06-07:
This morning I got quite excited by the great [API](https://github.com/berndporr/ECG-GUDB) by @berndporr to access the GUDB database.

One thing lead to another, I ended up with [this script](https://github.com/neuropsychology/NeuroKit/blob/dev/data/gudb/download_gudb.py) that one can run and that basically downloads, formats (in a *"neurokit-benchmarking friendly way"*) and saves the dataset as csv.

We could extend this to other databases to have scripts to download them and format them in the same "format" that we can later easily process.

With this in mind, I then wrote a new function, [benchmark_ecg_preprocessing](https://github.com/neuropsychology/NeuroKit/blob/dev/neurokit2/benchmark/benchmark_ecg.py), to which one can feed a processing function and either an ECG signal + Rpeaks or the path to the folder where a database is saved (assuming that it has the *"neurokit-benchmarking friendly"*), and will run the function and compute benchmarking indices like duration and score (currently the score is calculated in a quite [basic way](https://github.com/neuropsychology/NeuroKit/blob/4f388e6fdfc3e7d9baa38f48ca93b3d59323587b/neurokit2/benchmark/benchmark_ecg.py#L134)).

From there, it was super easy to simply run 3 different R detectors on Bernd's database and make some plots:

👉 [**see this README**](https://github.com/neuropsychology/NeuroKit/tree/dev/benchmarks/ecg_preprocessing#errors)

Some figs:

[image]

[image]

So far, it looks like 'neurokit' is the best method in terms of duration and accuracy 😋 Good thing is, one can add more methods in a few lines. 

Coming back to our idea of making a study to find the optimal preprocessing parameters (#78), this could be a first step. We could start by identifying the best detector on raw data, then pick it and find the best cleaning steps (filtering/detrending parameters). @berndporr we'd love to collaborate and have your expertise and input for that if you're interested and find time ☺️ 

To summarize, the steps to reproduce is:

1. Run [this script](https://github.com/neuropsychology/NeuroKit/blob/dev/data/gudb/download_gudb.py) in a folder to download and format the GUDB database
2. Define whatever processing functions that returns the R-peaks, e.g.,

```python
def my_method(ecg, sampling_rate):
    signal, info = nk.ecg_peaks(ecg, sampling_rate=sampling_rate, method="engzeemod2012")
    return info["ECG_R_Peaks"]
```
3. Run `nk.my_method(method, "path/to/the/GUDB/folder")`. It will return a dataframe of results.

What are your thoughts regarding the API of the benchmarking etc.? (cc @JanCBrammer )

JanCBrammer, 2020-06-07:
@DominiqueMakowski, glad to see that the in-house detector seems to perform well :sunglasses:

>     1. download and format the GUDB database

I feel like we should take advantage of the fact that we can now use the API to stream the datasets one by one into our benchmarking pipeline. Downloading the full dataset and storing it locally or in memory would somehow defeat the purpose of such an API?

> compute benchmarking indices like duration and score (currently the score is calculated in a quite [basic way](https://github.com/neuropsychology/NeuroKit/blob/4f388e6fdfc3e7d9baa38f48ca93b3d59323587b/neurokit2/benchmark/benchmark_ecg.py#L134)).

For the score I suggest we use the [scoring utilities of the wfdb package](https://github.com/MIT-LCP/wfdb-python/blob/b361877d98252de5467f441ffc335167e09cd9b5/wfdb/processing/evaluate.py#L12). These can be used to compare two sets of annotations (i.e., the manual "ground truth" annotation against the detector annotation) and provide TP, FP, and FN scores. These allow us to evaluate the detectors in terms of specificity (how many of the detector peaks are ground truth peaks?) and sensitivity (how many of the ground truth peaks were identified by the detector?). The execution time is also an interesting aspect to evaluate. It would be interesing to evaluate how a detector's run time scales with the size of the ECG dataset (big O), or simply to average execution times over multiple runs on the same dataset (there's probably too much variability in timing a single run).

DominiqueMakowski, 2020-06-07:
> I feel like we should take advantage of the fact that we can now use the API to stream the datasets one by one into our benchmarking pipeline. Downloading the full dataset and storing it locally or in memory would somehow defeat the purpose of such an API?

Yeah, true, although I imagine that it's quite hard to implement 🤔 and would necessitate patience and a solid connection if we combine downloading + processing, as these two steps take time. What I had in mind for a first step is to have is to have super clear guidelines/functions that would download/format the datasets locally for people (as the main issue is often IMO to get the datasets in shape - and in our case, in similar shape for different databases - so that they can be easily analyzed)

> i.e., the manual "ground truth" annotation against the detector annotation) and provide TP, FP, and FN scores. These allow us to evaluate the detectors in terms of specificity (how many of the [...]

Brilliant, I'll try to take a look at it I'm curious how these indices are exactly computed.

> The execution time is also an interesting aspect to evaluate. It would be interesting to evaluate how a detector's run time scales with the size of the ECG dataset

Indeed, this aspect can be improved, as you say it'd be good to repeat each run + do it for subsets of different lengths to see how it scales. That being said, it's already informative for a first rough draft (and consistent, in the fig above it appears as quite obvious for instance that engzeemod is consistently slower than NK or gamboa)

berndporr, 2020-06-07:
We benchmarked the different detectors:
https://www.biorxiv.org/content/10.1101/722397v3
the problem is that current "gold standard" is just an all/nothing within a time window which is usually chosen at 10% of the heartrate so that they all come out nicely at 99%. That also shows in this twitter thread about HRV:
https://github.com/berndporr/hrv
...because the HRV is about 10%.
I have an MSc student who has the task to come up over the summer with a better measure. I'd prefer a good old gaussian approach and rather see how the standard deviations differ. Then one can do statistical testing on them. I'll copy him in as it's his MSc. ;)

berndporr, 2020-06-07:
PS: the API also allows to add a local path to the ECG data. Added that in 1.0.3.

DominiqueMakowski, 2020-06-09:
> We benchmarked the different detectors:
https://www.biorxiv.org/content/10.1101/722397v3

@berndporr Of course we have this study in mind. I think the goal here was to replicate/expand (with some more detectors, and also investigate not just the detectors per se but the preprocessing parameters like filtering/detrending). 

> That also shows in this twitter thread about HRV:

Very interesting, thanks!

Okay so I reran the script and added a bit more data and improved the analysis to have some inferential modelling (taking into account the group-variability related to databases and individuals). The results appear as quite robust no matter what's the database and analysis technique, and they are a good news for @JanCBrammer (the inhouse detectors seems to be 👌 👍 ) (@TiagoTostas the one that you added isn't too bad either tho 😁). 

[image]

There a few minor differences with Bernd's study, but could be explained by the differences in the methodology, @berndporr what do you think?

- [**See here for details and more figures**](https://github.com/neuropsychology/NeuroKit/tree/dev/benchmarks/ecg_preprocessing#benchmarking-of-ecg-preprocessing-methods)

Potential next steps include: 

- [ ] Selecting the best R-peak detector; then testing the benefits of mean-detrending & normalization (3 conditions)
- [ ] Selecting again the best pipeline; then somehow test different parameters of high-pass filtering + complex detrending (this one is tricky 'cause has a lot of combinations between filter type, filter order etc)
- [ ] Selecting again the best pipeline; then test different parameters of lowpass filtering

DominiqueMakowski, 2020-06-09:
On a side note, I'd like to add the [LUDB](https://physionet.org/content/ludb/1.0.0/) database, but somehow I didn't manage to download it... anyone more proficient with wfdb could show me how to download the whole database?

JanCBrammer, 2020-06-22:
@DominiqueMakowski, haven't found a moment yet to fully process/appreciate the benchmarking results and future plans. Will take a look in the coming days. But I like what I see so far, awesome stuff!

DominiqueMakowski, 2021-04-29:
https://github.com/eleanorforsyth/JMX-ECG-Benchmark

New algorithm for peak detectors benchmarking

alexxony, 2021-08-06:
> This morning I got quite excited by the great [API](https://github.com/berndporr/ECG-GUDB) by @berndporr to access the GUDB database.
> 
> One thing lead to another, I ended up with [this script](https://github.com/neuropsychology/NeuroKit/blob/dev/data/gudb/download_gudb.py) that one can run and that basically downloads, formats (in a _"neurokit-benchmarking friendly way"_) and saves the dataset as csv.
> 
> We could extend this to other databases to have scripts to download them and format them in the same "format" that we can later easily process.
> 
> With this in mind, I then wrote a new function, [benchmark_ecg_preprocessing](https://github.com/neuropsychology/NeuroKit/blob/dev/neurokit2/benchmark/benchmark_ecg.py), to which one can feed a processing function and either an ECG signal + Rpeaks or the path to the folder where a database is saved (assuming that it has the _"neurokit-benchmarking friendly"_), and will run the function and compute benchmarking indices like duration and score (currently the score is calculated in a quite [basic way](https://github.com/neuropsychology/NeuroKit/blob/4f388e6fdfc3e7d9baa38f48ca93b3d59323587b/neurokit2/benchmark/benchmark_ecg.py#L134)).
> 
> From there, it was super easy to simply run 3 different R detectors on Bernd's database and make some plots:
> 
> 👉 [**see this README**](https://github.com/neuropsychology/NeuroKit/tree/dev/benchmarks/ecg_preprocessing#errors)
> 
> Some figs:
> 
> [image]
> 
> [image]
> 
> So far, it looks like 'neurokit' is the best method in terms of duration and accuracy 😋 Good thing is, one can add more methods in a few lines.
> 
> Coming back to our idea of making a study to find the optimal preprocessing parameters (#78), this could be a first step. We could start by identifying the best detector on raw data, then pick it and find the best cleaning steps (filtering/detrending parameters). @berndporr we'd love to collaborate and have your expertise and input for that if you're interested and find time ☺️
> 
> To summarize, the steps to reproduce is:
> 
> 1. Run [this script](https://github.com/neuropsychology/NeuroKit/blob/dev/data/gudb/download_gudb.py) in a folder to download and format the GUDB database
> 2. Define whatever processing functions that returns the R-peaks, e.g.,
> 
> ```python
> def my_method(ecg, sampling_rate):
>     signal, info = nk.ecg_peaks(ecg, sampling_rate=sampling_rate, method="engzeemod2012")
>     return info["ECG_R_Peaks"]
> ```
> 
> 1. Run `nk.benchmark_ecg_preprocessing(my_method, "path/to/the/GUDB/folder")`. It will return a dataframe of results.
> 
> What are your thoughts regarding the API of the benchmarking etc.? (cc @JanCBrammer @TiagoTostas)

 see this README is not found link.

closed 2022-12-03 by stale[bot]