neuropsychology/NeuroKit #247: signal_formatpeaks(): desired length to interpolate signal

Tam-Pham, 2020-06-09:
While the issue of signal_interpolation() is discussed in #242 and #243, I also noticed that in `_signal_formatpeaks_sanitize()`, the `desired length` for interpolation is set as `len(peaks_signal)`, which would be the length of the entire signal. 
https://github.com/neuropsychology/NeuroKit/blob/a8d2981155b031e88188b126e3120c74730bf792/neurokit2/signal/signal_formatpeaks.py#L71

However, in its use case in `signal_period()`, where `period` is interpolated between `peaks`, shouldn't the `desired_length` be the length between the `first_peak` and `last_peak` instead of the length of the entire signal?

@JanCBrammer @DominiqueMakowski. What do you guys think?

JanCBrammer, 2020-06-09:
@Tam-Pham, that is a good point and it made me review `signal_interpolate()` again. It would be ok to set `desired length` to the length of the signal if `signal_interpolate()` could extrapolate y values before the first peak (x_val[0]) and after the last peak (x_val[-1]). However, there are two issues with `signal_interpolate()` as it is currently implemeted:

1. extrapolation before x_val[0] or after x_val[-1] is not possible. This used to be possible with the `new_x` argument which I removed some time ago (my bad).

2. the scipy implementation of the Akima1DInterpolator does not allow for extrapolation. We need to use [monotonic cubic spline](https://docs.scipy.org/doc/scipy/reference/generated/scipy.interpolate.PchipInterpolator.html) instead. The latter ensures monotone interpolation (see #243) and also allows extrapolation beyond x_val[0] and x_val[-1]. Again, my bad, I didn't realize that `akima` cannot extrapolate.

I won't have time today to fix this, but we need to adress this asap since a lot of crucial functionality depends on `signal_interpolate()`.

DominiqueMakowski, 2020-06-09:
Agreed with @JanCBrammer, good catch @Tam-Pham !

JanCBrammer, 2020-06-09:
I just pushed a [fix of `signal_interpolate()`](https://github.com/neuropsychology/NeuroKit/pull/244/commits/394262f7ee89d45b291db4e318ebe7030dfaa90e). Please review. Now it's a matter of making sure that an appropriate `new_x` is passed in, which basically is everywhere we call `signal_period()` or `signal_rate()`.

JanCBrammer, 2020-06-09:
@Tam-Pham, @DominiqueMakowski, I think we need to review the arguments to `signal_interpolate()`.

Currently we have
```
(x_values, y_values, desired_length)
```
where desired_length is an integer that indicates over how many samples the y_values should be interpolated. However, the argument is ill-determined regarding the start and end of the interpolation range. I.e., the argumentntells us that the interpolation range should span desired_length samples. But does the range start at zero or at x_values[0]? And where does it end? Currently, we assume that the range starts at x_values[0] and ends at x_values[-1]. However, for many functions (e.g, `signal_period()`) we need to _extra_polate before and after the first and last x_value (otherwise the event-related analysis become invalid).

I suggest we replace `desired_length` with `new_x`:
```
(x_values, y_values, x_new)
```
With x_new being the a vector. This way, both the bounds of the interpolation range (start, end) as well as its length are clearly specified.

What do you think?

DominiqueMakowski, 2020-06-09:
A long time ago we indeed had both `new_x` and `desired_length` ([here](https://github.com/neuropsychology/NeuroKit/blob/v0.0.6/neurokit2/signal/signal_interpolate.py)) 😁 

I think that we could still use both:
```python
signal_interpolate( x_values, y_values, desired_length=None, x_new=None)
```

In which at least one of `desired_length` and `x_new` is necessary. If desired_length is none, then use `x_new`, if `x_new` is None, then assume that start and end are xval[0] and xval[-1]. and If both are provided and are not matching (i.e., desired_length != len(x_new)), complain. This way we don't break the API (we keep desired_length which is quite convenient for quick stuff, but we have x_new for precise stuff.

Or a simplified version would be to keep only `x_new`, but if an integer is passed, then it is considered as the `desired_length` and proceeds. So it would indeed simplify the API and allow for quick stuff like `signal_interpolate(x, y, 1000)` when you don't wanna think about creating a linspace.

Tam-Pham, 2020-06-09:
@JanCBrammer yes I definitely agree with you. Hard coding the starting and ending values (forced at x[0] and x[-1]) definitely restricts the flexibility of this function.

For clarity, I would tend to support having two separate arguments, one for desired_length and one for new_x.

JanCBrammer, 2020-06-09:
Ok, so then we basically keep the [implementation](https://github.com/neuropsychology/NeuroKit/pull/244/commits/394262f7ee89d45b291db4e318ebe7030dfaa90e) that I pushed earlier today:
```
signal_interpolate( x_values, y_values, new_x=None, desired_length=None )
```
?

Then we "only" have to make sure now that we pass through `x_new` to all the places where we either directly or indirectly call `signal_interpolate()`. Should we divide and conquer this task? 

Here's a list of all the functions that directly call `signal_interpolate()` (ignore the color highlighting):

[image]

I can take care of `signal_period()` and all the functions that call it (i.e., that indirectly call `signal_interpolate()`) at some point this week.

DominiqueMakowski, 2020-06-10:
> Should we divide and conquer this task?

Yes

Tho again, if `desired_length` is only used for convenience purposes, but not used directly in any of our functions, we might as well just merge it with `x_new`. In the docstring we can add  `If an integer is passed, will be considered as the desired length of the interpolated signal between its first and last value.` and basta, so it's less confusing than to have an optional argument that we ourselves never use

Tam-Pham, 2020-06-10:
Let's agree on the arguments first before getting started. 

To me having both or just 1 `x_new` argument wouldn't make that big of a difference since both are meant to capture the `new_x`. Just that if we are having one argument for both, let's make sure we document them carefully. 

>  ` If an integer is passed, will be considered as the desired length of the interpolated signal between its first and last value.`

This is fine!

I can take `rsp_amplitude()` and `ecg_quality()`

JanCBrammer, 2020-06-10:
Agreeing with you both. Dropping `desired_lenght` and preserving its purpose in `x_new` (and documenting it well) seems better than what we have now.

To be clear, we replace

```
desired_length=len(signal)
```

with

```
x_new=np.arange(len(signal))
```

anywhere we call `signal_interpolate()` directly or indireclty?

Tam-Pham, 2020-06-10:
I checked again and those that call `signal_interpolate()` directly are all below: 

[image]

Tam-Pham, 2020-06-10:
I'm contemplating between this:
[image]
where we don't use `desired_length` at all;  len(signal) is used to generate `new_x`

And this:
[image]

@JanCBrammer @DominiqueMakowski what do you guys think?

JanCBrammer, 2020-06-10:
I'm somewhat torn between the options. The first one is clean and explicit, but it makes the `desired_length` variable returned by `signal_formatpeaks()` obsolete. Which means that we would ideally also re-write the latter. The seconds options still makes use of the returned `desired_length` variable, but is much less clear/explicit imo. So basically the way I see it we can be clear at the cost of a little more work, or less clear and save some work :grin: I would prefer clarity.

DominiqueMakowski, 2020-06-11:
I feel like we should also change the order of the two firs arguments and have `y_values`, `x_values` and `x_new`. It feels more natural for the "signal" values (the y) to be passed as first

Tam-Pham, 2020-06-11:
Thanks @JanCBrammer for cleaning up everything so nicely! I will cleanup our documentation that uses `signal_interpolate()` directly or indirectly. 

> I feel like we should also change the order of the two firs arguments and have `y_values`, `x_values` and `x_new`. It feels more natural for the "signal" values (the y) to be passed as first

I think otherwise though 😅 Especially since `scipy.interpolate()` is using (x, y), passing `x_values` before `y_values` feels more natural 😄

DominiqueMakowski, 2020-06-11:
> Especially since scipy.interpolate() is using (x, y),

That's an argument from authority tho it doesn't mean it's the best option 😏

From the perspective of consistency, in all of the signal_* functions, the first argument is the signal (i.e., the y values), whereas for instance in events_* functions the first arg are the events. It feels natural for the signal you want to interpolate to write `signal_interpolate(signal, x_axis, new_axis)`

Tam-Pham, 2020-06-11:
> That's an argument from authority tho it doesn't mean it's the best option 😏

😅  

your thoughts? @JanCBrammer  ^^

JanCBrammer, 2020-06-11:
That's a tough one again.. :smile:

I agree with both of you that breaking convention is bad from an API perspective. It basically comes down to which convention we want to keep/break: NeuroKit or SciPy? Since `signal_interpolate()` wraps only SciPy functions, I'd prefer to preserve its original API `signal_interpolate(x_values, y_values)`. That also maps more intuitively (subjectively speaking of course) to the inteprolation steps:
1. create a function _f_ that maps original x_values to original y_values
2. use _f_ to map new_x to interpolated y_values

DominiqueMakowski, 2020-06-11:
[image]

Tam-Pham, 2020-06-11:
While inspecting the failures, I realized that we should remove `desired_length` in `signal_period()`  and `signal_rate()` too. 

https://github.com/neuropsychology/NeuroKit/blob/b0de7732160cc4a04c6d5dba252bcaf7a551d02c/neurokit2/signal/signal_period.py#L8

Since `_signal_formatpeaks_sanitize()` is not returning any `desired_length`, this variable is defined as `None`, causing the code to break (we can't `len(a-none-type)`). 

Hence, I'm going to remove the `desired_length` and interpolate on the range of len(peaks) instead:

`period = signal_interpolate(peaks, period, new_x=np.arange(len(peaks)), method=interpolation_method)`

DominiqueMakowski, 2020-06-11:
> It basically comes down to which convention we want to keep/break: NeuroKit or SciPy?

Isn't the purpose of this wrapper is to create our own API for this function 🤔 I think that people using nk's function don't have in mind the scipy functions (otherwise they'd just use scipy directly)

> Since signal_interpolate() wraps only SciPy functions

for now, but I wouldn't be surprised if other methods from other functions could make their way into it. 

> That also maps more intuitively (subjectively speaking of course) to the inteprolation steps:

Good point, however the steps are rather 1) I have a signal 2) I create a function that maps x_values to it, 3) I interpolate over a new range

soooo.... (signal, x_axis, x_new) 🙃

DominiqueMakowski, 2020-06-11:
[image]

JanCBrammer, 2020-06-11:
> inspecting the failures

@Tam-Pham, only the doctests fail because I haven't updated the examples. The actual code works (confirm by running tests locally). We need to have `desired_length` in `signal_period()` and `signal_rate()`, otherwise we don't know over which range to interpolate. The NoneType issue has other origins that I already fixed :blush:

JanCBrammer, 2020-06-11:
> 1) I have a signal 2) I create a function that maps x_values to it

That sir, is not how conventional function arguemtsworks in my mind 😄 It's f(x) = y, not f(y) = x, right?

Tam-Pham, 2020-06-11:
> We need to have `desired_length` in `signal_period()` and `signal_rate()`, otherwise we don't know over which range to interpolate. 

You are right! I got confused at one point and thought we can use `len(peaks)` instead of `len(signal)`, hence negating the need for `desired_length` 😅 

Lemme run through the docstring to fix them all😄

JanCBrammer, 2020-06-11:
> I got confused at one point and thought we can use `len(peaks)` instead of `len(signal)`, hence negating the need for `desired_length`

I went through this exact same confusion yesterday, and it took a quite gnarly session with the debugger to get over it :grin:. I highly recommend the Visual Studio Code IDE. Its debugger and search utilities are super handy for this kind of stuff (started using these things recently).

DominiqueMakowski, 2020-06-11:
<sub>*\*mutters to himself\* I'm sure this wouldn't have happen if `signal` was the first argument*</sub>

DominiqueMakowski, 2020-06-13:
Note that [`heartpy`](https://github.com/paulvangentcom/heartrate_analysis_python/blob/master/heartpy/preprocessing.py#L187) seems to be using an alternative: "interpolates peaks between the clipping segments using cubic spline interpolation. It takes the clipping start +/- 100ms to calculate the spline."

JanCBrammer, 2020-06-13:
@DominiqueMakowski, maybe that's a separate function that we could add to the `signal` module? Not sure what you mean by "alternative", do you mean for adding it to `signal_interpolate()`?

DominiqueMakowski, 2020-06-13:
I don't really know, I was just putting it here for reference 😁 

while I suspect that the purpose of this "clipping interpolation" is to avoid the caveats related to spline interpolation, I have no idea about the benefits/cost of this particular method. 

But if it's worth having, we could indeed add it as a method `clippingcubic` or something for `signal_interpolate`. But again, maybe we don't need and our solution works just as good?

JanCBrammer, 2020-06-13:
> the purpose of this "clipping interpolation" is to avoid the caveats related to spline interpolation

By looking at the [corresponding heartpy documentation](https://python-heart-rate-analysis-toolkit.readthedocs.io/en/latest/algorithmfunctioning.html#clipping-detection-and-interpolation) I think the "clipping interpolation" adresses a different problem. It works directly on the raw cardiac signal with the goal to restore lost peaks (by using cubic spline interpolation, i.e., it's not an alternative to cubic spline interpolation).

Also, I think we should have this discussion over at #243. #247 is more related to the API of `signal_interpolate()` whereas #243 is about the interpolation methods.

Mitchellb16, 2020-06-16:
> The NoneType issue has other origins that I already fixed 😊

@JanCBrammer were these fixed in the most recent PR? I'm running 0.0.37 and leaving defaults in `signal_rate()` and `signal_phase()` for `desired_length` throws errors for when `desired_length` is needed down the line.

`period = signal_interpolate(peaks, period, x_new=np.arange(desired_length), method=interpolation_method)` 

 Is there any reason why we can't just have them default to `len(signal)` (as they are in `rsp_process()`)? Or even just leave them as a required parameter?

JanCBrammer, 2020-06-17:
@Tam-Pham, @DominiqueMakowski, re-opening this since it seems that there are still two issues with the new `signal_interpolate()` API.

1. We have to better document the `desired_length` argument in those functions that call `signal_interpolate()`. Currently, the first sentence in the docstrings is misleading https://github.com/neuropsychology/NeuroKit/blob/65147605838947aa609e8116bfd5e80a60d5f419/neurokit2/signal/signal_period.py#L21-L24

2. We need a better default for `desired_length`. Currently we use `None` and that makes the call to `signal_interpolate()` throw an error:
https://github.com/neuropsychology/NeuroKit/blob/65147605838947aa609e8116bfd5e80a60d5f419/neurokit2/signal/signal_period.py#L67-L68

I suggest we set `desired_length` to the number of elements in `peaks` by default. Then we return the non-interpolated period by default.

DominiqueMakowski, 2020-06-17:
Agreed with the default desired_length.

Tam-Pham, 2020-06-17:
@Mitchellb16 for functions like `signal_rate()`, `signal` is not an argument and hence we can't have `desired_length` default as `len(signal)`. And therefore, as we illustrated in the examples, we hope the users can input the `desired_length`. 
```
rate = nk.signal_rate(peaks=info["Peaks"], desired_length=len(signal))
```
However, like you said, it can be really inconvenient (I sometimes forget too 😢) and should definitely be fixed to be more user-friendly.

@JanCBrammer 
> We have to better document the desired_length argument in those functions that call signal_interpolate(). Currently, the first sentence in the docstrings is misleading

Agree, the documentation of `desired_length` seems like an old one that we didn't update after making the changes. 

> We need a better default for desired_length. Currently we use None and that makes the call to signal_interpolate() throw an error:

Agree that we should set a default for `desired_length` to avoid the errors. 
If we were to set the default to `len(peak)` and the returned signal is a non-interpolated one, I'm thinking we should inform the users so. Perhaps we can print a message when `desired_length` = `len(peaks)`, saying that the output is non-interpolated; if users would like to perform interpolation between the peaks, please set `desired_length` to `len(signal)` instead.

JanCBrammer, 2020-06-17:
> If we were to set the default to `len(peak)` and the returned signal is a non-interpolated one, I'm thinking we should inform the users so. Perhaps we can print a message when `desired_length` = [...]

@Tam-Pham, that's exactly what I had in mind. Although we probably don't need the warning. Maybe it's better to simply clarify in the docstrings that this is happening (the description of the `desired_lenght` argument is probably a good place to do so, as well as the description of the returned values).

JanCBrammer, 2020-06-17:
Another solution would be to remove the `peaks` argument entirely and replace it with `signal`. Then we can simply call `signal_findpeaks(signal)` internally. May be more intuitive, especially since the function is callesd `signal_period()` not `peaks_period()`. Although that might be a step too far?

Tam-Pham, 2020-06-17:
> Another solution would be to remove the peaks argument entirely and replace it with signal

I strongly support with this option. It makes everything simpler, especially for the users (they can just use the default options). And I don't see any problem with doing that (as of now)

JanCBrammer, 2020-06-18:
> I strongly support this option

@DominiqueMakowski, do you have objections? Otherwise we can start working on this I guess? @Tam-Pham, should we do this on a feature branch? I won't be able to look into this until Monday though.

Btw, technically we should have this discussion in a new issue since it's now about the `signal_period()` and `signal_rate()` API, not the `signal_interpolate()` API 🙈

DominiqueMakowski, 2020-06-18:
Nope, seems okay to me! We can discuss it in detail in the related PR once it's created

PR #244 (2020-06-15): 0.0.37
close #247 close #255
closed 2020-06-28 by DominiqueMakowski