neuropsychology/NeuroKit #249: Missing rsp peaks, any parameters to tune to help this?

Mitchellb16, 2020-06-11:
Shown below is a short sample of an issue I have with a few of my respiratory files, followed by a segment directly after the issue segment that seems to be fine. I'm not sure if these peaks are getting washed out during detrending or signal_formatpeaks() maybe? Looking for any parameters I might be able to tune to help this, thanks!

[image]

[image]

DominiqueMakowski, 2020-06-12:
@Mitchellb16 what method are you currently using? the default one?

JanCBrammer, 2020-06-12:
@DominiqueMakowski I'm a little hestitant to flag this as a bug. I suspect the code works as intended. It's more likely that the signal needs to be processed differently.

@Mitchellb16, could you show us the script that produced this result? Then we can start ruling things out and formulating hypotheses as to what might need to be changed.

DominiqueMakowski, 2020-06-12:
My bug-flagging is quite liberal ☺️

JanCBrammer, 2020-06-12:
> My bug-flagging is quite liberal

I err more on the "innocent until proven guilty" side of things.

Mitchellb16, 2020-06-12:
certainly!

`sig_detrended = nk.signal_detrend(raw_sig)`

    sig_filtered = nk.signal_filter(sig_detrended, highcut=5, sampling_rate=10000)
    peak_signal, info = nk.rsp_peaks(sig_filtered,sampling_rate=10000, amplitude_min=.5)

    # Get additional parameters
    phase = nk.rsp_phase(peak_signal)
    amplitude = nk.rsp_amplitude(sig_filtered, peak_signal)
    rate = nk.signal_rate(peak_signal, sampling_rate=samp_rate)

    # Prepare output
    signals = pd.DataFrame({"RSP_Raw": raw_sig,
                            "RSP_Clean": sig_filtered,
                            "RSP_Amplitude": amplitude,
                            "RSP_Rate": rate})
    signals = pd.concat([signals, phase, peak_signal], axis=1)

JanCBrammer, 2020-06-12:
@Mitchellb16, from a first glance at this two things come to mind:

1. decrease the `amplitude_min` argument you're passing to `rsp_peaks()`. The latter internally chucks out trough-peak pairs with too small of an amplitude difference, based on a criterion that takes `amplitude_min` into account (by doing so you increase the risk of false positives though).
2. do not use `signal_detrend()` and instead apply a highpass filter with the `lowcut` argument set to .05Hz. Currently you are lowpass filtering with a highcut of 5Hz, so you could use a bandpass with lowcut .05Hz and highcut 5Hz.

Mitchellb16, 2020-06-12:
Will give that a try, thank you @JanCBrammer

Mitchellb16, 2020-06-12:
Brilliant! seems to have done the trick! @JanCBrammer , may I ask why we should no longer detrend the signal in this case? 

[image]

JanCBrammer, 2020-06-13:
> may I ask why we should no longer detrend the signal in this case?

@Mitchellb16, the breathing extrema detection algorithm relies on the raw breathing data being zero-centered. Internally we achieve this with highpass filtering in `rsp_clean()`. However, I'm guessing that you're not using `rsp_clean()` since it's optmized for human data (you're working with rodent data if I remember correctly?). Zero-centering can certainly also be achieved with detrending, but I'm not sure if `signal_detrend()` has side effects that might not "play well" with the `rsp_peaks()`. So in a nutshell, `rsp_findpeaks()` works best with signals returned from `rsp_clean()`.

But I have a suspicion that the way you detrend the data doesn't matter so much and that it's mainly about the `amplitude_min` argument. Have you tried only changing this parameter while leaving the rest of your pipeline the same? If you apply both steps that I mentioned in the earlier post at the same time you won't know which one "made it work".

closed 2020-06-16 by JanCBrammer