neuropsychology/NeuroKit #253: HRV – SD2/SD1 ratio in hrv_nonlinear

richrobe, 2020-06-12:
**Question and context**
Why is the library computing the SD2/SD1 ratio in hrv_nonlinear.py whereas literature is always computing the SD1/SD2 ratio? Is it on purpose? If not, I'd be happy to change that :-)

DominiqueMakowski, 2020-06-12:
I think my basis for this was this repo https://github.com/Aura-healthcare/hrvanalysis/blob/master/hrvanalysis/extract_features.py#L540

but then you might be right, I didn't double-check in the literature and other implementations. If you can add some references in the docstring to justify SS1/SD2 then please go ahead and fix it 😁 

[heartpy](https://github.com/paulvangentcom/heartrate_analysis_python/blob/master/heartpy/analysis.py#L644) this one has SD1/SD2 too so I think we can safely change it

richrobe, 2020-06-16:
Yes, heartpy has SD1/SD2, [pyhrv](https://github.com/PGomes92/pyhrv) as well, and the reference that was already referenced in the docstring (Shaffer and Ginsberg, 2017). I added an additional reference I used for my previous change (PR #252).

closed 2020-06-28 by DominiqueMakowski