neuropsychology/NeuroKit #274: Release NeuroKit2 as a conda package

milesrb, 2020-06-18:
This is an excellent Python package! Thank you for making it open source.  

Could we please make the neurokit package available from the conda package manager? This could be done through conda-forge: `conda install neurokit2 -c conda-forge`

As per the installation instructions for Neurokit2, the default way of installing Python and the scientific Python stack is through the Anaconda distribution of Python. Conda packages provide automatic updates (e.g. `conda update --all`) and better package dependencies management (as compared to pip), which could make the user's life a lot easier. Therefore, it would be great if neurokit2 were also a conda package

DominiqueMakowski, 2020-06-19:
Hi @milesrb, that seems like a good idea, let me see what I can do

DominiqueMakowski, 2020-06-19:
I tried, but I didn't manage to build/upload it. Here are the steps I followed (note that I don't use anaconda so I might have overlooked something):

I'm on windows 10. I took inspiration from [this guide](https://medium.com/@giswqs/building-a-conda-package-and-uploading-it-to-anaconda-cloud-6a3abd1c5c52)

1. install miniconda
2. Open anaconda prompt
3. Run `conda skeleton pypi neurokit2`

```
Warning, the following versions were found for neurokit2
0.0.1
0.0.3
0.0.6
0.0.8
0.0.9
0.0.10
0.0.12
0.0.15
0.0.18
0.0.28
0.0.34
0.0.35
0.0.36
0.0.37
Using 0.0.37
Use --version to specify a different version.
Using url https://files.pythonhosted.org/packages/de/31/a621a606f24a5ad06f5fc56a961fefc6ec372329a7a2b489b26ea7c58f86/neurokit2-0.0.37.tar.gz (14.8 MB) for neurokit2.
Downloading neurokit2
PyPI URL:  https://files.pythonhosted.org/packages/de/31/a621a606f24a5ad06f5fc56a961fefc6ec372329a7a2b489b26ea7c58f86/neurokit2-0.0.37.tar.gz
Unpacking neurokit2...
done
working in C:\Users\domma\AppData\Local\Temp\tmpa7ro3ei8conda_skeleton_neurokit2-0.0.37.tar.gz
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... done

## Package Plan ##

  environment location: D:\domma\miniconda3\conda-bld\skeleton_1592534034055\_h_env

The following NEW packages will be INSTALLED:

    ca-certificates: 2020.1.1-0
    certifi:         2020.4.5.2-py37_0
    openssl:         1.1.1g-he774522_0
    pip:             20.1.1-py37_1
    python:          3.7.7-h81c818b_4
    pyyaml:          5.3.1-py37he774522_0
    setuptools:      47.3.0-py37_0
    sqlite:          3.32.2-h2a8f88b_0
    vc:              14.1-h0510ff6_4
    vs2015_runtime:  14.16.27012-hf0eaf9b_2
    wheel:           0.34.2-py37_0
    wincertstore:    0.2-py37_0
    yaml:            0.1.7-hc54c509_2
    zlib:            1.2.11-h62dcd97_4

Preparing transaction: ...working... done
Verifying transaction: ...working... done
Executing transaction: ...working... done
Applying patch: 'C:\\Users\\domma\\AppData\\Local\\Temp\\tmpa7ro3ei8conda_skeleton_neurokit2-0.0.37.tar.gz\\pypi-distutils.patch'

Leaving build/test directories:
  Work:
 D:\domma\miniconda3\conda-bld\skeleton_1592534034055\work
  Test:
 D:\domma\miniconda3\conda-bld\skeleton_1592534034055\test_tmp
Leaving build/test environments:
  Test:
source activate  D:\domma\miniconda3\conda-bld\skeleton_1592534034055\_test_env
  Build:
source activate  D:\domma\miniconda3\conda-bld\skeleton_1592534034055\_build_env

        Error:
            Cannot use 'git' (not a git repo and/or patch) and did not find 'patch' in: D:\domma\miniconda3\conda-bld\skeleton_1592534034055\_build_env\Scripts;D:\domma\miniconda3\conda-bld\skeleton_1592534034055\_build_env\Library\mingw-w64\bin;D:\domma\miniconda3\conda-bld\skeleton_1592534034055\_build_env\Library\usr\bin;D:\domma\miniconda3\conda-bld\skeleton_1592534034055\_build_env\Library\bin;D:\domma\miniconda3\Scripts;D:\domma\miniconda3\Library\mingw-w64\bin;D:\domma\miniconda3\Library\usr\bin;D:\domma\miniconda3\Library\bin;D:\domma\miniconda3;D:\domma\miniconda3\Library\mingw-w64\bin;D:\domma\miniconda3\Library\usr\bin;D:\domma\miniconda3\Library\bin;D:\domma\miniconda3\Scripts;D:\domma\miniconda3\bin;D:\domma\miniconda3;D:\domma\miniconda3\Library\mingw-w64\bin;D:\domma\miniconda3\Library\usr\bin;D:\domma\miniconda3\Library\bin;D:\domma\miniconda3\Scripts;D:\domma\miniconda3\bin;D:\domma\miniconda3\condabin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\iCLS;C:\Program Files\Intel\Intel(R) Management Engine Components\iCLS;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;C:\Program Files\Intel\WiFi\bin;C:\Program Files\Common Files\Intel\WirelessCommon;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;C:\Users\domma\AppData\Local\Microsoft\WindowsApps;C:\Users\domma\AppData\Roaming\TinyTeX\bin\win32;C:\Users\domma\AppData\Local\GitHubDesktop\bin
            You can install 'patch' using apt-get, yum (Linux), Xcode (MacOSX),
            or conda, m2-patch (Windows),
```

Any help would be welcomed!

hoechenberger, 2020-12-03:
Hello, I have just published a `conda-forge` package of NeuroKit2.

Package: https://anaconda.org/conda-forge/neurokit2
Feedstock with recipe: https://github.com/conda-forge/neurokit2-feedstock

A bot will automatically update the recipe whenever a new version of NeuroKit2 has been published on PyPI. It will only require manual intervention if dependencies change.

- Should this be added to the installation instructions?
- I'll gladly add interested parties as co-maintainers to the feedstock.

JanCBrammer, 2020-12-04:
@hoechenberger, amazing, thanks! You can add me as co-maintainer. We'll add `conda` to the installation instructions asap.

hoechenberger, 2020-12-04:
@JanCBrammer Done, you should receive a notification / invitation :)

JanCBrammer, 2020-12-04:
Closing for now. We can re-open in case `conda`-related stuff comes up.

closed 2020-12-04 by JanCBrammer

DominiqueMakowski, 2020-12-04:
Awesome @hoechenberger and thanks a lot 🙌