neuropsychology/NeuroKit #275: Enable Markdown files to be added to the Sphinx docs

DominiqueMakowski, 2020-06-20:
We have `.md` files (the `README.md` that are in the benchmark folder) that it'd be nice to include in the docs. 

See this issue: https://github.com/sphinx-doc/sphinx/issues/2840

Waiting for this to be merged https://github.com/sphinx-doc/sphinx/pull/7739

rpanderson, 2020-06-27:
Current alternatives for achieving this are:
- [MyST](https://myst-parser.readthedocs.io/en/latest/): A fully-functional markdown flavor and parser for Sphinx.
- Use `m2r` (not actively maintained) for the `mdinclude` directive and `recommonmark` for everything else, per [this comment and example](https://github.com/readthedocs/recommonmark/issues/191#issuecomment-622369992) by @orsinium.

DominiqueMakowski, 2020-06-28:
Thanks for the suggestions @rpanderson, I didn't know about MyST! Let me give it a try

DominiqueMakowski, 2020-06-28:
I tried a few things but didn't manage to make it work 😞 

The issue is that I'd like the documentation to mainly be built using .RST as it is for now, but only in some places include .md files. In particular, the .md files that I'd like to include are the [README.md's](https://github.com/neuropsychology/NeuroKit/tree/master/benchmarks) that are present in the benchmark subfolders. (Note that these are not the package's README, but files that are called README so it gets displayed when visiting this page on github.)

For now, what I tried is, in the documentation, add an [index.rst](https://github.com/neuropsychology/NeuroKit/blob/dev/docs/benchmarks/index.rst) that points to another [.rst](https://github.com/neuropsychology/NeuroKit/blob/dev/docs/benchmarks/ecg_preprocessing.rst) that is supposed to include the .md file (similarly as it works for instance in [this case](https://github.com/neuropsychology/NeuroKit/blob/dev/docs/contributing/contributing.rst)). But I wasn't successful in making it work, and as you said the problem with `m2r` is that it's not maintained and therefore problematic to rely on.

Any suggestions are very much welcome ☺️

rpanderson, 2020-06-28:
That *appears* possible using MyST, but I haven't got enough experience with it to readily diagnose your issue. I am familiar with applying my second recommendation to the use case you mention, however (using m2r), which I implemented a working example of in the [mdinclude-example branch of my fork](https://github.com/rpanderson/NeuroKit/commit/d9c80097a939696821fc4667ff3600e876ad1676), which is built [here](https://rpanderson-neurokit2.readthedocs.io/en/latest/).

To make this work, I also had to:
* Use HTML for any hyperlinks in bold/italics;
* Relative link figures from docs directory, i.e. prepend ../../benchmarks/ecg_preprocessing to image paths; and
* Remove line blank lines from references div.

I too would prefer to use MyST for this use case and will let you know if/when I implement it.

DominiqueMakowski, 2020-06-28:
> which I've implemented a working example of in the mdinclude-example branch of my fork

@rpanderson splendid! Could you please consider making a PR with it! I'd be very happy to merge it :)

> Relative link figures from docs directory

Does that mean that when viewing the README.md from github, the figures won't be visible anymore 'cause the relative link won't work? If so, I'd say it's acceptable for now as it's better to have them nicely displayed on the website than lost within github

rpanderson, 2020-06-28:
> Could you please consider making a PR with it!

Done. But you'll have to [change the base branch](https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/changing-the-base-branch-of-a-pull-request) to dev :-P

> Does that mean that when viewing the README.md from github, the figures won't be visible anymore 'cause the relative link won't work?

Nope. By coincidence, this README.md and the docs source using `mdinclude` are at the same level of the repository, so the relative `../../` takes you to root in both cases, and the figures render on GitHub.

PR #291 (2020-06-28): [docs] Use of mdinclude directive to add .md files to sphinx docs
# Description

Partially resolves #275.

# Proposed Changes

Use the `mdinclude` directive of m2r, per https://github.com/readthedocs/recommonmark/issues/191#issuecomment-622369992 for including benchmarks/ecg_preprocessing/README.md on the Benchmarks page. To make this work, I had to additionally:
* Use HTML for any hyperlinks in bold/italics;
* Remove blank lines from references div; and
* Relative link figures from docs directory, i.e. prepend ../../benchmarks/ecg_preprocessing to image paths.

The latter does not break image [... cut]
closed 2020-06-29 by DominiqueMakowski