neuropsychology/NeuroKit #279: add eog_plot()

DominiqueMakowski, 2020-06-23:
@zen-juen now that we have a minimal EOG pipeline we could add _plot function with like the signal and the blinks and below the rate.

zen-juen, 2020-06-23:
Potentially we could also add more labels of features once we are able to delineate potental blinks into saccades, fixations etc. 😄 

[image]

DominiqueMakowski, 2020-06-24:
good job @zen-juen 🥇 yeah we'll hopefully improve that with new features in time!

DominiqueMakowski, 2020-06-28:
We could add a "pane" on the right similarly to ecg_plot where we plot all the detected blink epochs

zen-juen, 2020-06-29:
```
# Get data
 eog_signal = nk.data('eog_100hz')

 # Process
eog_signals, peaks = nk.eog_process(eog_signal, sampling_rate=100)

nk.eog_plot(eog_signals, peaks, sampling_rate=100)

```

[image]

closed 2020-06-30 by zen-juen