neuropsychology/NeuroKit #281: HRV - SD1 and SD2: discrepancies between implementations

DominiqueMakowski, 2020-06-23:
Alright folks I came across another weirdness related to HRV and the Poincaré measures SD1 and SD2. I checked different existing implementations and they seem to differ, and I'm not sure which one is the right one.

Let's assume these are the RRi:

```python
import numpy as np

rri = np.array([125, 250, 100, 300])
```

Below are the different implementations (which ***should*** be correct) and their result: 

## [hrv](https://github.com/rhenanbartels/hrv/blob/develop/hrv/classical.py#L272) and [hrvanalysis](https://github.com/Aura-healthcare/hrvanalysis/blob/2becba351955075e5e1cf9d2e6b2c0195344d8d0/hrvanalysis/extract_features.py#L537)

```python
diff_rri = np.diff(rri)
sd1 = np.sqrt(np.std(diff_rri, ddof=1) ** 2 * 0.5)
sd2 = np.sqrt(2 * np.std(rri, ddof=1) ** 2 - 0.5 * np.std(diff_rri, ddof=1) ** 2)
print("hrv\nSD1 = " + str(sd1), "\nSD2 = " + str(sd2))
````
```
hrv
SD1 = 130.3041314259324 
SD2 = 40.82482904638623
```
## [pyhrv](https://github.com/PGomes92/pyhrv/blob/master/pyhrv/time_domain.py#L343)

```python
x1 = rri[:-1]
x2 = rri[1:]
sd1 = np.std(np.subtract(x1, x2) / np.sqrt(2), ddof=1)
sd2 = np.std(np.add(x1, x2) / np.sqrt(2), ddof=1)
print("PyHRV\nSD1 = " + str(sd1), "\nSD2 = " + str(sd2))
````
```
PyHRV
SD1 = 130.30413142593238 
SD2 = 17.677669529663675
```

## [heartpy](https://github.com/paulvangentcom/heartrate_analysis_python/blob/master/heartpy/analysis.py#L695)

```python
x_plus = rri[:-1]
x_minus = rri[1:]
x_one = (x_plus - x_minus) / np.sqrt(2)
x_two = (x_plus + x_minus) / np.sqrt(2)
sd1 = np.sqrt(np.var(x_one))
sd2 = np.sqrt(np.var(x_two))
print("HeartPy\nSD1 = " + str(sd1), "\nSD2 = " + str(sd2))
````
```
HeartPy
SD1 = 106.39287779003085 
SD2 = 14.433756729740633
```

@richrobe any thoughts? Also pinging my fellow @rhenanbartels ☺️

DominiqueMakowski, 2020-06-23:
[Ciccone, 2017](https://onlinelibrary.wiley.com/doi/abs/10.1002/mus.25573):

[image]

richrobe, 2020-06-23:
The title of Ciccone et al. is very misleading, I also stumbled across that... They say "RMSSD and SD1 are identical heart rate variability metrics", but in fact that's not true. They are identical with respect to a **constant** 1 / sqrt(2). They even report that in their own paper: (Eq. 10).

Of course both measures carry the identical physiological information, but to allow comparability with other work the correct definition for SD1 should be used. Either by just taking: SD1 = 1/sqrt(2) * RMSSD or by using the Formula from [Brennan, 2001](https://ieeexplore.ieee.org/abstract/document/959330)

richrobe, 2020-06-23:
But relationship could be used to test which implementation is correct, right? By comparing their result for RMSSD with their result for SD1, multiplied by sqrt(2)... both should be the same if SD1 is computed correctly :)

DominiqueMakowski, 2020-06-23:
> he title of Ciccone et al. is very misleading

Yeah that's what started my investigation haha

> But this relationship could be used to test which implementation is correct

In theory yes but it doesn't seem right...

```python
rmssd = np.sqrt(np.mean(diff_rri ** 2)) 
rmssd / np.sqrt(2)
```
```
114.1088661469096
```

🤔

richrobe, 2020-06-24:
heartpy computes the uncorrected variance (0 degrees of freedom). hrv and pyhrv compute the corrected standard deviation (1 degree of freedom).

If you add ddof=1 to their code you will get the same result as pyhrv:
```
x_plus = rri[:-1]
x_minus = rri[1:]
x_one = (x_plus - x_minus) / np.sqrt(2)
x_two = (x_plus + x_minus) / np.sqrt(2)
sd1 = np.sqrt(np.var(x_one, ddof=1))
sd2 = np.sqrt(np.var(x_two, ddof=1))
print("HeartPy\nSD1 = " + str(sd1), "\nSD2 = " + str(sd2))
```

```
HeartPy
SD1 = 130.30413142593238 
SD2 = 17.677669529663675
```

It makes such a large difference because the N in our example is quite low. For larger N's the difference would eventually almost vanish.

richrobe, 2020-06-24:
To be honest, I have no idea what pyhrv and heartpy do in their computation of SD2, especially the element-wise addition of successive RR intervals. I would "trust" the implementation of hrv(analysis) because it uses the formula from [Brennan, 2001](https://ieeexplore.ieee.org/abstract/document/959330/) (Eq. 12). It's been published in TMBE which is a very good journal and has been cited over 900 times, so I'd go with that ;)

DominiqueMakowski, 2020-06-24:
With @Tam-Pham we just arrived at the same conclusion regarding the ddof=1 for SD1, but indeed for SD2 it is quite mysterious! But thanks for your light, now I'm more confident in our implementation (which is the same as hrv) - always good to double-check ☺️ 

But again I mean I keep repeating the same stuff over and over (cf. the issues about complexity indices), but the fact that one can find so many divergences in established implementations of physio processing is simply mindblowing and really scarry

Tam-Pham, 2020-06-24:
So with a closer look at  [Brennan, 2001](https://ieeexplore.ieee.org/abstract/document/959330/) with @hungpham2511 , we believe that the x1 and x2 in heartpy implementation can be found in Eq7.

[image]

With `theta = pi/4` (phase angle of identity line), sin(theta) and cos(theta) are both `1/sqrt(2)`. Multiplying the matrix will give us exactly their implementation

```
x_plus = rri[:-1]
x_minus = rri[1:]
x_one = (x_plus - x_minus) / np.sqrt(2)
x_two = (x_plus + x_minus) / np.sqrt(2)
```
And the below text:
[image]

is basically saying that `sd1 = std(x1)` and `sd2 = std(x2)` (thanks @hungpham2511 for explaining it to me, I will never get that myself 😁 )

So the main point is, our implementation and heartpy's are equivalent. Perhaps the above example has too few data points, hence the discrepancy? 

Anyway, I tried with one of our sample data

```
data = nk.data("bio_resting_5min_100hz")
```
and SD1 and SD2 from both are very similar

```
HeartPy
SD1 = 49.283740132770504
SD2 = 72.51277315542195
```
```
NeuroKit
SD1 = 49.283740132770504
SD2 = 72.62846122819343
```
It's good that @DominiqueMakowski brought up all the different implementations so that we have got a chance to compare and review ours. 

IMO, pyhrv's and heartpy's implementation would be preferred over our current version and hrv's. Even though the latter are so much clearer in the implementation, they went through more calculations and hence basically more prone to noises.

DominiqueMakowski, 2020-06-25:
Well TBH `72.51277315542195` and `72.62846122819343` is a big difference (~0.1157), so it's not a trivial thing due to rounding or precision. So @Tam-Pham if I understand, you say that essentially pyhrv's `sd2 = np.std(np.add(x1, x2) / np.sqrt(2), ddof=1)` and heartpy's `np.sqrt(np.var(x_two))` are equivalent (only diff being the ddof for the SD).

And this is different from hrv's (and ours) `np.sqrt(2 * np.std(rri, ddof=1) ** 2 - 0.5 * np.std(diff_rri, ddof=1) ** 2)`, which is "more prone to noises" (?).

So based on that the "best" implementation would be pyhrv's `sd2 = np.std(np.add(x1, x2) / np.sqrt(2), ddof=1)`?

Tam-Pham, 2020-06-25:
> So based on that the "best" implementation would be pyhrv's `sd2 = np.std(np.add(x1, x2) / np.sqrt(2), ddof=1)?`

Yes, I would agree with this. Should we make a revision for our implementation?

DominiqueMakowski, 2020-06-25:
Yes, I think so. @richrobe what do you think?

closed 2020-06-28 by DominiqueMakowski