neuropsychology/NeuroKit #284: New EDA-ECG/HRV index: Sympathovagal balance (SV)

DominiqueMakowski, 2020-06-24:
- [ ] [Ghiasi et al., 2018](https://ieeexplore.ieee.org/abstract/document/8512932)

> Particularly, sympathetic activity is estimated from
the EDA power calculated within the 0.045-0.25Hz bandwidth
(EDASymp), whereas parasympathetic dynamics is measured
instantaneously through a point-process modeling framework
devised for heartbeat dynamics (HFpp).

> significantly higher statistical discriminant
power than the standard LF/HF ratio, also revealing different
dynamics between female and male subjects

zen-juen, 2020-07-01:
Based on a literature search, it seems like there are a few variations of how EDAsymp is calculated differently in these two following papers:

- [Ghiasi et al., 2018](https://ieeexplore.ieee.org/abstract/document/8512932)
> EDA signals were downsampled to 50 Hz and normalized
by applying z-score calculation. Then, a 2th order Butterworth bandpass filter (0.01-0.5 Hz) was employed to remove
any very low frequency trend. The time-frequency representation of EDA was calculated through the standard shorttime Fourier transform approach with Welch periodograms. A Blackman window of 60s (overlap of 59s) was employed
to divide the signal into segments and obtain the timefrequency representation. Such a time-frequency plane is
then integrated within the (0.045-0.25 Hz) band in order to estimate the sympathetic autonomic activity EDASymp(k)
where k are discrete time samples within a time window (0, T]

Note also that Ghiasi et al. cite Posada-Quintero et al. for more details on EDA PSD analysis

- [Posada-Quintero et al., 2016](https://biosignal.uconn.edu/wp-content/uploads/sites/2503/2018/09/09_Posada_2016_AnnalsBME.pdf)
> For frequency-domain analysis, EDA signals were
down-sampled to 2 Hz. Before down-sampling, the data was filtered with an 8th-order Chebyshev Type I
low-pass filter (0.8 Hz). Down-sampling from 400 to 2 Hz was performed in two steps (using consecutive
down-sampling factors of 1/20 and 1/10, respectively). Finally, signals were high-pass filtered (0.01 Hz, Butterworth, 8th order) to remove any trend. The power spectra of EDA signals were calculated using Welch’s
periodogram method with 50% data overlap. A Blackman window (length of 128 points) was applied
to each segment, the Fast Fourier Transform was calculated for each windowed segment, and the power
spectra of the segments were averaged. Note that the down-sampling frequency for EDA was half of the
sampling frequency for HRV. Total power [lS2
] and the power within the frequency bands of interest (VLF = 0–0.045 Hz,
LF = 0.045–0.15 Hz, HF1 = 0.15–0.25 Hz, HF2 = 0.25–0.4 Hz and VHF = 0.4–0.5 Hz) were computed. After determining the frequency range for sympathetic assessment, EDASymp and EDASympn (n.u.) = EDASymp/(total power) were computed.

I'm not yet sure if these differences are experiment-specific

DominiqueMakowski, 2020-07-01:
So it seems like there are two different things, there is 1) EDASymp which is roughly the equivalent of HRV but from EDA (EDV?) and 2) the Sympathovagal balance (SV) which is "EDASymp/HFpp", so the EDV normalized by the heart rate. Is that correct?

zen-juen, 2020-07-01:
To be more exact, EDAsymp is computed as the power of the EDA signal used as an index of sympathetic activity (just that this power is computed differently in Ghiasi and Posado-Quintero) so no, not really the equivalent of HRV though related! 

The main motivation seems to be this ([Posada-Quintero et al., 2016](https://biosignal.uconn.edu/wp-content/uploads/sites/2503/2018/09/09_Posada_2016_AnnalsBME.pdf)) :
>  Specifically, we are interested in examining if EDA analysis yields similar frequency
dynamics at the HRV’s prescribed LF range of 0.045– 0.15 Hz. If such relation occurs, we can overcome the
main disadvantage of LF domain analysis of HRV because EDA is not influenced by the parasympathetic
branch of the autonomic nervous system, and we could provide a better quantitative evaluation of sympathetic
tone. 

and yes you're right in that SV is the ratio between EDAsymp vs. HF power from HRV - the calculation of HFpp seems quite complicated and I've been trying to wrap my head around it 😨

zen-juen, 2020-07-07:
So I compared the two implementations and there seems to be a slight discrepancy in the output, where Posada-Quintero et al.'s method results in:
```
indexes_posada = nk.eda_sympathetic(eda, sampling_rate=100, method='posada', show=False)

indexes_posada
Out[294]: {'EDA_Symp': 8.007495794727076e-05, 'EDA_SympN': 0.00010296497878935038}
```
and Ghiasi et al.'s method in:
```
indexes_ghiasi = nk.eda_sympathetic(eda, sampling_rate=100, method='ghiasi', show=False)

indexes_ghiasi
Out[296]: {'EDA_Symp': 9.657522523654743e-05, 'EDA_SympN': 0.00024128086623066448}
```

also note that the former's implementation involves `signal.scipy.welch` whereas the latter involves `signal.scipy.spectrogram`

DominiqueMakowski, 2020-07-07:
> also note that the former's implementation involves signal.scipy.welch whereas the latter involves signal.scipy.spectrogram

Could indeed be the source. But it's good that we have both methods! Which one do you think we should have as "default" ?

zen-juen, 2020-07-07:
For now I’ve set Posada-Quintero et al.’s method as the default simply because its specific algorithmic steps are more commonly referenced in the (rather limited) EDAsymp literature (eg downsampling rate of 2Hz, data overlap of 50%)

DominiqueMakowski, 2020-07-08:
downsampling to 2hz is super harsh tho 😕 the other method has it too?

zen-juen, 2020-07-08:
Ghiasi's method downsamples it to 50Hz. Posada uses 2Hz downsampling rate in several of his published papers on EDA sympathetic index and one [study](https://journals.plos.org/plosone/article/file?id=10.1371/journal.pone.0198328&type=printable) of his in particular uses 4Hz:

> 
> Time-invariant and time-varying spectral analyses have been deployed recently on EDA data
> [16,17]. For this study, EDA data were down sampled to 4 Hz prior to spectral analysis.

Their justification is that 2Hz is sufficient given that the dynamics of the EDA spectrum are largely confined to frequencies less than 0.4 Hz

DominiqueMakowski, 2020-07-08:
And what are the advantages of downsampling at all? Like would it work if we don't downsample at all?

DominiqueMakowski, 2020-07-19:
@zen-juen can we close this?

zen-juen, 2020-07-19:
@DominiqueMakowski The EDAsymp index is more or less finished but not the Sympathovagal balance (SV) index - though the latter can be reserved for a separate issue 😄 your call

DominiqueMakowski, 2020-07-19:
SV is "just" EDAsymp / HRV-HF right? can be implemented quite fast so we can leave that open 'til then

zen-juen, 2020-07-19:
I wasn't too sure if it was as straightforward as just dividing EDAsymp by HRV-HF because based on Ghiasi et al.

> Particularly, sympathetic activity is estimated from the EDA power calculated within the 0.045-0.25Hz bandwidth (EDA Symp ), whereas parasympathetic dynamics is measured instantaneously through a point-process modeling framework devised for heartbeat dynamics (HF pp ). We test our new index SV = EDA Symp /HF pp 

which then goes on to define HFpp as :
[image]
[image]
[image]

DominiqueMakowski, 2020-07-19:
I think it's the same, he basically defines (@Tam-Pham must confirm) the Fourier transform of the RR intervals which gives the PSD, and the computation of single-point values for different ranges.

What it's intriguing is that they define the HF range as [0.14-0.45], whereas it is most often defined as [0.15, 0.4]. Wonder where they got that from. Especially since LF is til 0.15. So like frequency of 0.145 is both LF and HF 🤔 ? Probably a mistake from their part, I'd say.

Tam-Pham, 2020-07-19:
> the Fourier transform of the RR intervals which gives the PSD, and the computation of single-point values for different ranges.

I just read the paper and it seems like that's the basic idea. 

EDA_symp is the power in the (0.045, 0.25) frequency band. However, instead of a single value for an entire signal, the paper seems to suggest to calculate an EDA_symp value for each time segment - EDASymp(k) 

[image]

signal_timefrequency() returned is a matrix where the columns are the discrete time segments (e.g. segment 1, segment 2...) and the rows are the frequencies. So to calculate EDASymp(k), we can take the mean along axis 0 (columns)

[image]

For each time segment, a SV(k) value is calculated by taking EDASymp(k)/ HF(t) (t=k). Here HF is HRV-HF of that segment.

Would we want to return EDA_symp and SV as time series or as single values across all time segments as of now?

>  they define the HF range as [0.14-0.45], whereas it is most often defined as [0.15, 0.4].

😨 true...seems more like a mistake.

@zen-juen I have finalized the stft function in `signal_timefrequency()` so I think you can safely internally import it. It's just that I'm currently setting the default window as `window='hann'`. Maybe we can add an argument to allow us to use `window='blackman'` for EDAsymp.

Tam-Pham, 2020-07-20:
Hmm, looking at the paper again this morning, I have to agree with Zen that calculating HFpp is not that straightforward. It might not be the same as HRV-HF 😅

Will go and stare at the formulas

closed 2021-05-22 by stale[bot]