neuropsychology/NeuroKit #292: Standardize branch & PR names

DominiqueMakowski, 2020-06-29:
As of the last release, we tried to move towards a different workflow. Rather than committing like savages directly to dev, we tried to create separate branches (and PRs) more "feature specific", and merge them only once it passes the checks. We hope that it's easier to work with (less cross-breaks) and also easier to track changes, as each PR is compiled into the release changelog:

https://github.com/neuropsychology/NeuroKit/releases

This makes it seem like we should standardize the branch names and the PR names. For now, I tried to have all PRs starts with `[type] title` but as you can see, we should standardize the categories titles. Anyway, let me know your thoughts and experience on what is the easier to work with (in the end, the goal is to minimize the burden for contributors so that the pipeline is not too convoluted or hard to remember).

Tam-Pham, 2020-06-30:
I find the naming convention of `[type] title` makes the purpose of the PR pretty clear. 

We could agree on a few general categories so as not to have too many. And perhaps to briefly define each of them. For instance:
- **fix**: for hot fixes or any breaking changes that could not be put under other categories
- **docs**: to add/ improve docstrings, fix doctests or any other docstring related matters
- **feature**: to add new features
...

Also, since we are currently having quite a few labels for Issues, could we also add similar labels for PRs and these labels could be the categories that that PR belongs to?

DominiqueMakowski, 2020-07-06:
For PRs:

- **[Fix]** bug fix etc
- **[Feature]** new feature or function
- **[Docs]** typos, documentation and new articles
- **[Change]** changes in args, names, functions etc (often breaking changes)
- **[Improvement]** refining of code, efficiency improvements etc.

For branches, same but named 'fix/something', 'feature/something', 'docs/something' etc.

commit bda4f488 (2020-07-19): branch naming convention

close #292
closed 2020-08-05 by Tam-Pham