neuropsychology/NeuroKit #309: ECG delineate CWT mode crashes if no consecutive T and P peaks are found

raimonpv, 2020-07-06:
**Describe the bug**
```
Traceback (most recent call last):
  File "*/neurokit2/ecg/ecg_delineate.py", line 121, in ecg_delineate
    waves = _ecg_delineator_cwt(ecg_cleaned, rpeaks=rpeaks, sampling_rate=sampling_rate)
  File "*/neurokit2/ecg/ecg_delineate.py", line 468, in _ecg_delineator_cwt
    tpeaks, ppeaks = _peaks_delineator(ecg, rpeaks, sampling_rate=sampling_rate)
  File "*/neurokit2/ecg/ecg_delineate.py", line 625, in _peaks_delineator
    tpeaks, ppeaks = zip(*[(g[0], g[-1]) for g in significant_peaks_groups])
  File "*/neurokit2/ecg/ecg_delineate.py", line 625, in <listcomp>
    tpeaks, ppeaks = zip(*[(g[0], g[-1]) for g in significant_peaks_groups])
IndexError: list index out of range
```

**To Reproduce**
1. Run ecg_delineate function with "cwt" mode with an ECG signal with no consecutive T and P peaks.

**Expected behaviour**
Right now it is considered that the first peak found in _find_tppeaks is a T-peak, the last one a P-peak. If just one peak is found, it is considered that the T and P peaks are the same and if no peak is found it is returning an empty list that makes crash the `_peaks_delineator` function. I expect to return a Nan instead in order that the function doesn't crash and the resulting arrays of P and T peaks is the same length as the R peaks.
 
**Specifications**
- System: Linux
- Python version: 3.7.7
- NeuroKit Version: 0.0.37

Tam-Pham, 2020-07-07:
Thanks @raimonpv for reporting the bug. We will fix it asap and follow up here. 

In the meantime, if needed for your analysis, please rely on the `dwt` or `peaks` methods in `ecg_delineator()`.

raimonpv, 2020-07-07:
@Tam-Pham I had already fixed it and can make a PR with the code tomorrow morning if it's okey with you

DominiqueMakowski, 2020-07-07:
@raimonpv by all means, please PR :)

Tam-Pham, 2020-07-07:
That’s fantastic! We can further discuss the fixes in your PR. 💪🏻💪🏻💪🏻 looking forward to it

raimonpv, 2020-07-09:
Closed by #314

closed 2020-07-09 by raimonpv