neuropsychology/NeuroKit #310: ecg_process() function ValueError: NeuroKit error: we expect the user to provide a vector, i.e., a one-dimensional array (such as a list of values). Current input of shape: ()

solomonmanuelraj, 2020-07-07:
Hi,
In the python 3.7 - 32 bits environment i have run the following sample code given for ecg_process function. 

>>> import neurokit2 as nk
>>>
>>> ecg = nk.ecg_simulate(duration=15, sampling_rate=1000, heart_rate=80)
>>> signals, info = nk.ecg_process(ecg, sampling_rate=1000)

When i run the nk.ecg_process() statement (signals, info = nk.ecg_process(ecg, sampling_rate=1000)) i received the following error.

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\charl\anaconda3\envs\py37_32\lib\site-packages\neurokit2\ecg\ecg_process.py", line 106, in ecg_process
    ecg_cleaned=ecg_cleaned, rpeaks=rpeaks, sampling_rate=sampling_rate
  File "C:\Users\charl\anaconda3\envs\py37_32\lib\site-packages\neurokit2\ecg\ecg_delineate.py", line 119, in ecg_delineate
    waves = _ecg_delineator_peak(ecg_cleaned, rpeaks=rpeaks, sampling_rate=sampling_rate)
  File "C:\Users\charl\anaconda3\envs\py37_32\lib\site-packages\neurokit2\ecg\ecg_delineate.py", line 683, in _ecg_delineator_peak
    Q_index, Q = _ecg_delineator_peak_Q(rpeak, heartbeat, R)
  File "C:\Users\charl\anaconda3\envs\py37_32\lib\site-packages\neurokit2\ecg\ecg_delineate.py", line 720, in _ecg_delineator_peak_Q
    Q = signal_findpeaks(-1 * segment["Signal"], height_min=0.05 * (segment["Signal"].max() - segment["Signal"].min()))
  File "C:\Users\charl\anaconda3\envs\py37_32\lib\site-packages\neurokit2\signal\signal_findpeaks.py", line 120, in signal_findpeaks
    info["Onsets"] = _signal_findpeaks_findbase(info["Peaks"], signal, what="onset")
  File "C:\Users\charl\anaconda3\envs\py37_32\lib\site-packages\neurokit2\signal\signal_findpeaks.py", line 191, in _signal_findpeaks_findbase
    bases = as_vector(bases)
  File "C:\Users\charl\anaconda3\envs\py37_32\lib\site-packages\neurokit2\misc\type_converters.py", line 44, in as_vector
    "list of values). Current input of shape: " + str(shape)
ValueError: NeuroKit error: we expect the user to provide a vector, i.e., a one-dimensional array (such as a list of values). Current input of shape: ()

when i run in 64 bit python environment this statement is working fine. Why it is giving error in 32_bit environment
[image]

DominiqueMakowski, 2020-07-07:
Weird, indeed, as it works for me:

[image]

Could you run `nk.__version__` and tell us which version are you using? Thanks

solomonmanuelraj, 2020-07-07:
Hi,
thanks for your quick help. You are running in 64 bit python. It is working fine in 64 bit python. 

In 32 bit python it is giving error specially ecg_process() function. ecg_simulate() function is working fine.

I have attached the screen shot to show case the 32 bits python environment where i am running. I have a constraint it need to execute in the 32 bit python environment . 

concern is why it is working in 64 bits environment and not in 32 bits environment
[image]

solomonmanuelraj, 2020-07-07:
Kindly help me to solve this issue

DominiqueMakowski, 2020-07-07:
Could you try first updating the package from github and then try again just to make sure it's not the cause of the problem:

`pip install https://github.com/neuropsychology/NeuroKit/zipball/master --upgrade`

solomonmanuelraj, 2020-07-07:
Thanks for the link. I have updated the package (0.0.38) using github link. Received the same error. I have attached the screenshot for more details.

thanks for your help.

[image]

DominiqueMakowski, 2020-07-07:
It's quite bizarre, because the error seems to be located in signal_findpeaks(), as if it didn't find any peaks. Can you try running the following:

```python
>>> import neurokit2 as nk
>>>
>>> ecg = nk.ecg_simulate(duration=15, sampling_rate=1000, heart_rate=80)
>>> nk.signal_plot(ecg)
>>> nk.ecg_findpeaks(ecg, sampling_rate=1000)
```

And attach here the figure and the potential error message?

solomonmanuelraj, 2020-07-07:
Above code is executing fine.

Attached the results.

Thanks for your help. I need to use ecg_process() function. is it any other way we can run ecg_process() function?

i am using the ecg_process() output for further processing.

Thanks for your help.
[image]

DominiqueMakowski, 2020-07-07:
Okay, we're getting there, can you try the following:

```python
>>> ecg = nk.ecg_simulate(duration=15, sampling_rate=1000, heart_rate=80)
>>> cleaned = nk.ecg_clean(ecg, sampling_rate=1000)
>>> _, rpeaks = nk.ecg_peaks(cleaned, sampling_rate=1000)
>>> nk.ecg_delineate(cleaned, rpeaks, sampling_rate=1000, method="peak")
```

solomonmanuelraj, 2020-07-07:
Attached the code

[image]

DominiqueMakowski, 2020-07-07:
Okay so the error originates from the `ecg_delineate()` function. Can you inspect the cleaned signal (plot it) and the 'rpeaks' object (to confirm that there are some detected peaks)?

JanCBrammer, 2020-07-07:
Pretty sure this is a type issue that originates here:
https://github.com/neuropsychology/NeuroKit/blob/cb37d83ee20d6a13a91c4848aa435f41e979e203/neurokit2/misc/type_converters.py#L29-L32

and then causes the `ValueError` here:
https://github.com/neuropsychology/NeuroKit/blob/cb37d83ee20d6a13a91c4848aa435f41e979e203/neurokit2/misc/type_converters.py#L35-L41

Try this for illustration:
```Python
import numpy as np

out = np.array([42])

print(out.shape)

out = np.array(42)

print(out.shape)
```

DominiqueMakowski, 2020-07-08:
@JanCBrammer typical python ^^, any recommendations for the most elegant fix?

JanCBrammer, 2020-07-08:
We'd first have to confirm that it is actually a type issue by inspecting what's happening in a 32 bit environment (I believe those can be set up with conda). @solomonmanuelraj, how did you set up your 32 bit environment? Maybe easiest if we replicate what you did.

solomonmanuelraj, 2020-07-08:
I have followed the below stackoverflow link .
https://stackoverflow.com/questions/17781266/how-can-i-use-virtualenv-to-use-32-bit-and-64-bit-python-in-windows

set CONDA_FORCE_32BIT=1
conda create -n py37_32 python=3.7.6 anaconda

Above commends I have used. 

Thanks for your help.

Let me know how this issue can be solved.

JanCBrammer, 2020-07-08:
@solomonmanuelraj, ok cool, thanks! I guess you set up the environment on a 64 bit machine?

Out of curiosity, why do you have the 32-bit constraint?

Will try to replicate your issue in the coming days (no promises though).

DominiqueMakowski, 2020-07-08:
@Tam-Pham do you know how to set conda envs so we can also try replicating it?

solomonmanuelraj, 2020-07-08:
The deployment machine architecture is 32 bit architecture. That is the reason I am creating 32 bits environment in the 64 bits python machine. I have followed the above stockoverflow link and I have mentioned the two lines I am using in the anoconda prompt to create the virtual environment called py37_32.
After I create the py37_32 environment, I activate it through conda activate py37_32. In the py37_32 environment I am running the ecg_process().

Thanks for your help.
Let me know how to solve this problem.

solomonmanuelraj, 2020-07-08:
First I installed python 64 bits version 3.7.6 in windows 10 machine. After that I created environment variable py37_32 for 32 bits python environment

JanCBrammer, 2020-07-08:
@solomonmanuelraj, thanks for the detailled instructions. I was able to replicate your issue and [fixed it](https://github.com/neuropsychology/NeuroKit/pull/289/commits/3836e62ac56905155ca69964f4acf45f2dc2701c). Once version 0.0.39 is on PyPI you can update and `ecg_process()` should (hopefully) no longer break in your 32 bit environment.

solomonmanuelraj, 2020-07-08:
Nice to know this good news. Let me know when version 0.0.39 will be available?

DominiqueMakowski, 2020-07-09:
@solomonmanuelraj you can install the dev version by running `pip install https://github.com/neuropsychology/neurokit/zipball/dev --upgrade`

closed 2020-07-09 by DominiqueMakowski

solomonmanuelraj, 2020-07-10:
It's working fine. Thanks for your help