neuropsychology/NeuroKit #317: Allow PSD to return different methods at once

DominiqueMakowski, 2020-07-08:
Since signal_psd returns a dataframe anyway, we could allow the methods arg to be passed a list of multiple methods (and an "all" string), in which case it would return the Frequencies x-axis and then one column per method. That would be some syntactic sugar useful for instance for plotting and comparing.

Tam-Pham, 2020-07-08:
I checked and unfortunately as the frequency arrays of different methods are not the same and hence the outputs have all different lengths (they have different resolutions) , we can't combine them into 1 dataframe.

DominiqueMakowski, 2020-07-08:
mmh could we "upsample" to match the most fined-grained one of the output? (i.e., pad fill and stretch)

closed 2021-05-22 by stale[bot]