neuropsychology/NeuroKit #323: Realtime ECG/PPG Signals Validation

solomonmanuelraj, 2020-07-09:
Hi All,
I need your help to classify if the captured ECG signal is valid or not. In real time data capturing, some times ECG and PPG sensors data are not captured properly and ECG /PPG signals contains lot of noise. Even after applying different types of filters the signals quality remains bad which will impact the peaks identification.

Is there any method/steps available to classify/differentiate captured ECG /PPG signals are valid or not. If it is not a valid signal then we need to capture the signal again.

thanks for your help.

DominiqueMakowski, 2020-07-09:
NeuroKit currently doesn't have any realtime features. As for offline ECG signal quality, you can use `nk.ecg_quality()`, though only a relatively basic method is implemented for now. 

If applicable, visual inspections is still the best method for detecting bad signals, especially for signals like ECG where a standard pattern is quite characteristic.

solomonmanuelraj, 2020-07-09:
Ok. Thanks for this information.

closed 2020-07-09 by DominiqueMakowski

solomonmanuelraj, 2020-07-10:
ecg_quality() method gives index value between 0 and 1 vector. How to interpret this values? 0 /0.5 / 0.75 / 1 means what ? Is it a cut off values we need to use on this vector values to classify the ECG signal is valid or not valid?. We need to automate this decision making. We can not see the ECG visualisation charts and decide in the time of automation. How we can automate this decision making with some acceptable accuracy? Thanks for your help. The portal is helping me lot to solve my doubts.

DominiqueMakowski, 2020-07-10:
the method currently implemented in `ecg_quality()` is a relative measure of quality, in the sense that it computes the (normalized) "distance" between each QRS complex from the average QRS complex from the signal. You can see it as an index of deviation from the mean. 

As per the documentation of the function:

> "Compute a continuous index of quality of the ECG signal, by interpolating the distance
    of each QRS segment from the average QRS segment present in the data. This index is
    therefore relative, and 1 corresponds to heartbeats that are the closest to the average
    sample and 0 corresponds to the most distance heartbeat, from that average sample."

In other words, the smaller the worst, and no threshold exist, it's currently ultimately up to you.