neuropsychology/NeuroKit #326: issues for analysis of various sampling rate signals

minsooyeo, 2020-07-10:
hello ! 

firstly, i thanks a lot for wonderful your project !

during do my own project with neurokit2, i found several problems on neurokit2 for calculating bio-medical signal features and where is problem

_below is tutorial of 'neurokit2.rsp.rsp_rrv'_ : link --> https://neurokit2.readthedocs.io/en/latest/functions.html#neurokit2.rsp.rsp_rrv
==============================================================
>>> import neurokit2 as nk
>>>
>>> rsp = nk.rsp_simulate(duration=90, respiratory_rate=15)
>>> rsp, info = nk.rsp_process(rsp)
>>> rrv = nk.rsp_rrv(rsp, show=True) 
==============================================================

but, if i want to the signal with 32 sampling hz
==============================================================
>>> import neurokit2 as nk
>>> from scipy import signal
>>> rsp = nk.rsp_simulate(duration=90, respiratory_rate=15)
>>> rsp = signal.resample(rsp, int((len(rsp) / 1000) * 32))
>>> rsp, info = nk.rsp_process(rsp, sampling_rate=32)
>>> rrv = nk.rsp_rrv(rsp, show=True, sampling_rate=32) 
==============================================================

these result are dismatch and i could see a warning 

"Neurokit warning: signal_psd(): The duration of recording is too short to support a sufficiently long window for high frequency resolution. Consider using a longer recording or increasing the `min_frequency`
"

on debugging code, i found a bug

[image]

 sampling rate should be defined dynamically not constanly.

this issue could find another function: **"nk.rsp_analyze"**

[image]

i think these bugs must be consider later version

if similar issues is founded, i will commit ~ : )

many thanks !

DominiqueMakowski, 2020-07-10:
Hi @minsooyeo, thanks for your feedback!

Indeed, that looks like a bug in which the sampling rate is not propagated correctly throughout the functions. 

Would you like to make a pull request with the fix?

minsooyeo, 2020-07-10:
> Hi @minsooyeo, thanks for your feedback!
> 
> Indeed, that looks like a bug in which the sampling rate is not propagated correctly throughout the functions.
> 
> Would you like to make a pull request with the fix?

yes i will deal it

zen-juen, 2020-07-14:
We've merged your PR - thanks for your improvements and feel free to contribute further if you pick up on more issues! 😄

closed 2020-07-14 by zen-juen

minsooyeo, 2020-07-14:
> We've merged your PR - thanks for your improvements and feel free to contribute further if you pick up on more issues! 😄

thank you !