neuropsychology/NeuroKit #329: HR Calculation using ECG and PPG Signals gives different HR values

solomonmanuelraj, 2020-07-11:
Hi All

thanks for publishing Neurokit2 0.0.39 version. I need to calculate Heart Rates [HR] separately from ECG and PPG signals.
 I created simulated ECG and PPG signals with heart rate = 80. When i tried to retrieve HR from simulated data, i got different HR rates [ ECG - HR - 80 and PPG-HR is 75 ]. I used the ecg_process and ppg_process to derive the HR rates. Why ppg_process gives HR as 75 instead of expected HR rate of 80? 

let me know your comments. 

Thanks for your help. Attached the code and output screen shots below
[image]
[image]

JanCBrammer, 2020-07-11:
Two things you can do:

1. Increase the duration (e.g., to 20 seconds).

2. set the `frequency_modulation` argument to `0` when calling `ppg_simulate()`:
```Python
ppg = nk.ppg_simulate(duration=10, sampling_rate=1000, heart_rate=80, frequency_modulation=0)
```

In general, by plotting the signals you can get a good idea of what is going on (which saves you from guessing why the HR is 75, not 80):

```Python
import neurokit2 as nk
import matplotlib.pyplot as plt

ppg = nk.ppg_simulate(duration=10, sampling_rate=1000, heart_rate=80)
signals, _ = nk.ppg_process(ppg, sampling_rate=1000)

nk.ppg_plot(signals)
plt.show()
```

One side note: you can use the fact that the `signals` object contains numpy arrays. For example you can do `signals["PPG_Rate"].mean()` instead of `statistics.mean(signals["PPG_Rate"])`.

DominiqueMakowski, 2020-07-11:
I'd add that the simulated signals are given some amount of variability to mimic naturally occurring variations in heart rate, hence the discrepancy that you observe between the expected and the actual value. That being said, these fluctuations being at random, they should decrease as the signal length increases.

JanCBrammer, 2020-07-11:
@DominiqueMakowski, that got me thinking, maybe it's best to set `frequency_modulation=0` by default to avoid this kind of confusion?

solomonmanuelraj, 2020-07-11:
Thanks for this information. In actual PPG signal value it will calculate it correctly?

DominiqueMakowski, 2020-07-11:
>  maybe it's best to set frequency_modulation=0 by default to avoid this kind of confusion?

I've toyed with this idea, but I would tend to think that people are (mostly) okay with the "naturalistic" default (especially because otherwise for a perfectly stationary signal it creates weird processing issues in regards with HRV - since essentially the rate is a flat line). That being said, maybe we could emphasize it in the docstrings though, that some random fluctuations are to be expected.

also, ecg_simulation doesn't have an argument for changing the degree of modulation, not sure if that's easy to add (@Tam-Pham maybe has some idea). 

On a side note, I wonder whether we should rename args as `heartrate_fluctuation` (and thus `heartrate` instead of heart_rate to avoid two underscores - though we'd need to to a soft deprecation first and allowing for both args to work before removing it in future releases).

JanCBrammer, 2020-07-12:
> Thanks for this information. In actual PPG signal value it will calculate it correctly?

@solomonmanuelraj, yes, that is most likely the case. The algorithm has been tested by the original authors: https://journals.plos.org/plosone/article/comments?id=10.1371/journal.pone.0076585

Be careful though in case you're planning on using it on short segments during real-time processing, the algorithm is not made for this.

JanCBrammer, 2020-07-12:
> otherwise for a perfectly stationary signal it creates weird processing issues in regards with HRV

Right, that makes sense. Although `ppg_simulate()` still introduces some randomess (jitter) in the peak placement even in the abscence of frequency modulation.

> maybe we could emphasize it in the docstrings though, that some random fluctuations are to be expected

Agreed, that seems like a sensible solution.

solomonmanuelraj, 2020-07-12:
@JanCBrammer thanks for your help. Mainly i need to use nk.ppg_process() function to estimate HR using PPG signal. In the real time i want to capture 10 seconds PPG signals and want to calculate HR for 10 seconds PPG data (sampling rate is 200).  When i pass 10 seconds data with sampling rate of 200, nk.ppg_process() can correctly estimate the HR rate? or any risk you are foreseeing because of less data points ( 10 seconds data).

let me know your comments. 

thanks for your help

JanCBrammer, 2020-07-13:
> i want to capture 10 seconds PPG signals and want to calculate HR for 10 seconds PPG data

@solomonmanuelraj, 10 seconds could be a sufficient amount of data (I would not try to process shorter segments though). Try to carefully validate that the processing is valid (especially if your application is safety critical, which I hope it isn't :grin:). For example, if you record the PPG along with the results of the real-time HR estimation, you can process the PPG offline afterwards and compare the real-time and offline HR (i.e., the time series of instantaneous HR). If they match (not exactly, but close enough) you could assume that everything works as intended. Otherwise you have to consider implementing dedicated real-time processing (since using ppg_process() in real-time is a hack).

DominiqueMakowski, 2020-07-14:
Let us know how your realtime processing goes. 

I think we can close that as the initial question has been addressed.

closed 2020-07-14 by DominiqueMakowski