neuropsychology/NeuroKit #330: Respiration rate using sinusoidal cross-correlation based approach

DominiqueMakowski, 2020-07-12:
From [Chuen, L., Sears, D., & McAdams, S. (2016). Psychophysiological responses to auditory change. Psychophysiology, 53(6), 891-904](https://onlinelibrary.wiley.com/doi/abs/10.1111/psyp.12633) (p. 894): 

> A technique involving cross-correlation of the filtered respiratory signal with sinusoidal signals of different frequencies was used to estimate the time-varying frequency of the respiration. To extract instantaneous respiration rate in BrPM from the respiration belt signal over fairly small time periods, for which average interinhalation interval would be difficult to assess, we adapted a spectral analysis technique from Müller (2007) [that] identifies the principal frequency of oscillation by the peak correlation between 10-s frames of the signal and a bank of sinusoidal signals of appropriate frequency. The procedure first downsamples the filtered respiration signal to a 10-Hz sampling rate and calculates the first-order difference, thus representing the rate of change in the expansion and contraction of the thorax. It then computes the cross-correlations between the resulting time series and a bank of 101 sinusoids with frequencies of 5–30 BrPM at 0.25 BrPM increments, using 10-s frames and a hop size of 1 sample (0.1 s). To account for anomalous respiratory behaviors such as yawning, one component of the set of sinusoid was generated at 1 BrPM, resulting in a monotonic set of samples over the 10-s window. Within this window, this component would cross-correlate well with breathing rates of 3 BrPM or less. Frames with absolute correlation coefficients greater than 0.5 were sampled for peak values that fit the short segment of the respiration cycle in phase and frequency. From the resulting series, maxima were extracted, interpolated, and smoothed with a 0.1 Hz 4th-order Butterworth low-pass filter. This technique thereby provides a measure of the time-varying breathing rate. This breathing-rate signal was then upsampled to 256 Hz for comparison with other series in the data set. **This technique provides a more robust estimate of time-varying respiration rate on fairly short signals by making use of more signal information than the onset of each breath**.

If I understand correctly:

1. Downsample to 10Hz and get the gradient. 
2. Slice the signal into 10s windows with 0.1s overlap
3. For each window, get the (cross?)correlation with 101 sinusoids with frequencies of 5/60–30/60 Hz with 0.25/60 increment (`np.arange(5/60, 30.25/60, 0.25/60)`) + 1 sinusoid at 1/60 Hz.
4. Instructions unclear...

Someone with a better understanding can clarify the steps?

minsooyeo, 2020-07-13:
it seem to investigate regularity of respiration

i think the process is below

1. calculation first-order difference (same to np.diff function)
2. calcaulate the cross correlation with sinusoids with frequencies of 5/60–30/60 Hz with 0.25/60 increment 
3. iteration with a hop size of 1 sample
4. find max values on the data (3.)
5. smoothing the data (4.)

this is my code (step 1~2)

```
import numpy as np
import neurokit2 as nk
from matplotlib import pyplot as plt
from scipy.stats import zscore

if __name__ == '__main__':
    # 10 sec rsp
    sample_rsp = nk.rsp_simulate(10, sampling_rate=10)
    # sample_rsp = np.random.randn(100)
    # first-order difference
    sample_rsp_first_order_diff = np.ediff1d(sample_rsp)
    # cross-correlation with sin signal with various periods
    # define result & sin signal time
    result = []
    t = np.linspace(0, 10, 99)
    for tmp_period in np.arange(5/60, 30.25/60, 0.25/60):
        tmp_sin = np.sin(t * (2 * np.pi * (tmp_period)))
        # calculation cross-correleation with 1-leg
        tmp_xcorr = np.corrcoef(sample_rsp_first_order_diff,
                                tmp_sin,
                                'full')[0, 1]
        result.append(tmp_xcorr)
    # visuallization
    plt.subplot(211)
    plt.title('raw data')
    plt.plot(sample_rsp)
    plt.grid()
    plt.subplot(212)
    plt.plot(np.linspace(5/60, 30.25/60, 101), result)
    plt.xlabel('frequecny')
    plt.ylabel('cross correlation')
    plt.title('result')
    plt.grid()

```
respiration signal result 
[image]

random signal result
[image]

cross correlation is greatest at 0.3 hz, it about 0.3 hz

and this is my code for all step

you should install tqdm with pip

`pip install tqdm`

```
# -*- coding: utf-8 -*-
"""
Created on Mon Jul 13 11:38:52 2020

@author: user
"""
import numpy as np
import neurokit2 as nk
from matplotlib import pyplot as plt
from scipy.stats import zscore
from scipy.signal import find_peaks, butter, filtfilt
from scipy.interpolate import interp1d
from tqdm import tqdm

def butter_lowpass(cutoff, fs, order=5):
    nyq = 0.5 * fs
    normal_cutoff = cutoff / nyq
    b, a = butter(order, normal_cutoff, btype='low', analog=False)
    return b, a

def butter_lowpass_filter(data, cutoff, fs, order=4):
    b, a = butter_lowpass(cutoff, fs, order=order)
    y = filtfilt(b, a, data)
    return y

def resp_rate(data, fs=10, window=10, overlab=1):
    # define parameters
    data_raw = int(fs * window)
    data_len = len(data)
    resp_rate = []
    for idx in tqdm(range(0, data_len, overlab)):
        # data binding
        tmp_data = data[idx:idx + data_raw]
        # skip short data
        if len(tmp_data) < data_raw:
            continue
        # calculate 1-order diffrence
        tmp_data_diff = zscore(np.ediff1d(tmp_data))
        # define parameter
        cross_corr_result = []
        t = np.linspace(0, window, len(tmp_data_diff))
        # iteration with different sin signal
        for tmp_period in np.arange(5/60, 30.25/60, 0.25/60):
            # define sin signal
            tmp_sin = np.sin(t * (2 * np.pi * (tmp_period)))
            # calculation of cross correlation
            tmp_xcorr = np.corrcoef(tmp_data_diff,
                                    tmp_sin,
                                    'full')[0, 1]
            # append data (cross correlation)
            cross_corr_result.append(tmp_xcorr)
        # append data (total cross correlation)
        cross_corr_result = np.array(cross_corr_result)
        # find max frequency (for respiration rate)
        max_fre_idx = np.where(cross_corr_result == np.max(cross_corr_result))[0]
        max_fre = np.arange(5/60, 30.25/60, 0.25/60)[max_fre_idx]
        # append data (respiration rate)
        resp_rate.append(max_fre[0])
    return np.array(resp_rate)

if __name__ == '__main__':
    # 600 sec rsp
    sample_rsp = nk.rsp_simulate(1000, sampling_rate=10)
    resp_rate = resp_rate(sample_rsp)
    maxma_values, _ = find_peaks(resp_rate)
    # interpolation signal
    f1 = interp1d(maxma_values, resp_rate[maxma_values])
    interpol_val = f1(np.linspace(maxma_values[0], maxma_values[-1], len(sample_rsp)))
    # smoothing with filter
    smoothing_data = butter_lowpass_filter(interpol_val, 0.1, 10)
    # visullization
    plt.figure()
    plt.subplot(211)
    plt.plot(sample_rsp)
    plt.grid()
    plt.title('raw data')
    plt.subplot(212)
    plt.title('resp rate')
    plt.plot(smoothing_data)
    plt.xlabel('time (1/10 sec)')
    plt.ylabel('frequency')
    plt.grid()
```

[image]

but i dont know why calculate the 1-order difference
and it take a long time... i think it is better use fft method for calculating respiration rate

DominiqueMakowski, 2020-07-14:
Awesome @minsooyeo thanks, it's much clearer now

> but i dont know why calculate the 1-order difference and it take a long time... i think it is better to use fft method for calculating respiration rate

@Tam-Pham what do you think?

Tam-Pham, 2020-07-22:
@minsooyeo Thank you for your input. I will try to implement this method, following your suggestion, in NeuroKit and we can compare it with other methods. To be honest, the description of the method sounds like a mix between short-time fourier transfrom and lombscargle to me (e.g create short time windows then find the best fitting sinusoid at a frequency 🤔). 

There are 2 things that I'm not too sure though:

- the paper mentioned about fitting a sinusoid at 1Brpm to remove anomolies:
> To account for anomalous respiratory behaviors such as yawning, one component of the set of sinusoids was generated at 1 BrPM, resulting in a monotonic set of samples over the 10‐s window. Within this window, this component would cross‐correlate well with breathing rates of 3 BrPM or less

But I'm not sure what's the steps the author is suggesting here. How do we basically remove the anomalies.

> Frames with absolute correlation coefficients greater than 0.5 were sampled for peak values that fit the short segment of the respiration cycle in phase and frequency.

Here, the author seems to suggest that for all sinusoids with xcorr greater than 0.5, we vary its phases to again look for the best fit before picking the maxima. The author was very vague. However, if the above interpretation is correct, it would make sense to me (even to vary the amplitude together with the phase). What do you think?

JanCBrammer, 2020-07-24:
@Tam-Pham, @DominiqueMakowski, @minsooyeo, Chuen, L., Sears, D., & McAdams, S. (2016). mention in the references that they use the [Matlab code](https://github.com/finn42/RespirationTracking) written by https://github.com/finn42. Maybe @finn42 can help filling in some details?

minsooyeo, 2020-07-26:
i read a matlab code: https://github.com/finn42/RespirationTracking,

and i found several misunderstanding about the code which i written

firstly, the reason why calculation of difference of respiration signal is below, however i still can not understand it ...

**"thus representing the rate of change in the expansion and contraction of the thorax."
(it can be found on main text from  @DominiqueMakowski )**

look at the MATLAB code, the procedure of it is below

**(01)** down sampling of respiration signal 
**(02)** calculation of the difference of respiration signal
**(03)** segmentation with 10 sec frame (no overlapping) in other word, iteration with 10 sec frame
**(04)** calculation of correlation with sin waves with 1 and 5~30 bpm (interval 0.25), totally, it is 101 correlations
**(05)** from the result of 04, change values which less than 0.5 to 0.5
e.g) [0.1, 0.2, -0.3, 0.6, 0.1, 0.2, -0.8] --> [0.5, 0.5, 0.5, 0.6, 0.5, 0.5, 0.5]
**(06)** find maxima index of the result of (05) and find the bpm corresponding to maxima index
e.g) corr:  [0.5, 0.5, 0.5, **_0.6_**, 0.5, 0.5, 0.5]
       index: [0, 1, 2, **_3_**, 4, 5, 6]
       bpm: [1, 5, 5.25, **_5.5_**, 5.75, 6, 6.25]

but still i could find several dismatchs with the paper which @DominiqueMakowski  indroduced

**Within this window, this component would cross-correlate well with breathing rates of 3 BrPM or less. Frames with absolute correlation coefficients greater than 0.5 were sampled for peak values that fit the short segment of the respiration cycle in phase and frequency.** 

matlab code calculation abs() but, it is not used ...

it is line 46 ~ 50 of matlab code
```
       46 >> r = corr(sec,V);%,'type','Spearman');
       47 >> Spec(:,i) = r';
       48 >> %max(abs(r))
       49 >> w = abs(r);
       50 >> w(r<0.5) = 0.5; **_% the result of 49 line not used to find less than 0.5 correlation ...._**
```

but, i still have no idea why this procedure is better than traditional method for calculating respiration rate

finn42, 2020-07-26:
Hi! Nice to see people poking at this algorithm. 

I appreciate some of the consternation expressed here over why the process described is so computationally demanding when someone might just want the respiration rate. This really comes down to two issues: why do you want the respiration rate? What time scale of changes are you interested in tracking?

If you only need an average rate every minute, counting the inspiraitons over 60 s intervals might be fine enough. If you have to work with shorter recording intervals, 10s or 20s, breath counting is too coarse. The cross correlation method allows you to increase precision with your choice of comparison waves.

The sequence of differencing and thresholding is specifically to suppress some kinds of noise that are common to respiration sequences when people are breathing quietly. The shallow breaths of people attending to things like music or video often include noise that is close to the respiration rate, but its effect is strongest in the post expiration pause, when the chest and band are relaxed. Thresholding is a quick and dirty way to cut out some of this noise and reduce octave errors. 

This algorithm was constructed to process very short recordings during a particularly passive task with the goal of picking up stimulus effects that are quite small. If your algorithm needn't be that sensitive, feel free to use a simpler tool!

Last note: I have done a lot  more work on respiration signals since constructing this, and haven't looked at it too closely in the last few years, so I won't claim it to be gold standard. I've since stopped using sinusoidal models for respiration analysis because the signals I work with are not efficiently described by fourier deconstructions. If you are leaning towards using a breath counting method for calculating rate, I reviewed a bunch of methods for detecting inspirations in my dissertation (ch 4)  https://pqdtopen.proquest.com/doc/2128010509.html?FMT=ABS 

I'm not going to comb through all the issues above, but if you have any specific questions about respiration or pressing concerns, drop me a line. 

Cheers!
Finn

JanCBrammer, 2020-07-26:
@finn42, thanks for the elaboration, much appreciated. Started skimming the dissertation, looks like a goldmine!

minsooyeo, 2020-07-27:
thanks @finn42 
I learn a lot

closed 2021-05-22 by stale[bot]