neuropsychology/NeuroKit #332: M/EEG Microstates Resources

DominiqueMakowski, 2020-07-13:
There is already awesome work done 😍:

- [mne_microstates](https://github.com/wmvanvliet/mne_microstates)
- [eeg_microstates](https://github.com/Frederic-vW/eeg_microstates)
- [keypy](https://github.com/keyinst/keypy)
- Guide: [Poulsen, 2018 - microstate for EEGlab](https://www.biorxiv.org/content/10.1101/289850v1.full.pdf)

### Clustering methods

- [ ] k-means
- [ ] modified k-means (in [mne_microstates](https://github.com/wmvanvliet/mne_microstates) and [eeg_microstates](https://github.com/Frederic-vW/eeg_microstates/blob/bd202ee5e218c98c6515f3427bb9fc51a8bf9dce/eeg_microstates.py#L518))
- [ ] Atomize and Agglomerate Hierarchical Clustering (AAHC) (in [eeg_microstates](https://github.com/Frederic-vW/eeg_microstates/blob/bd202ee5e218c98c6515f3427bb9fc51a8bf9dce/eeg_microstates.py#L401))
- [ ] Topographic Atomize and Agglomerate Hierarchical Clustering (TAAHC) (<sub>"the only difference between TAAHC and AAHC is in how they measure the quality of the clustering. Where the AAHC uses the GEV, TAAHC uses the sum of correlations as seen in (6). This means that TAAHC does not account for the strength of the maps, but only for the similarity of their topographies"</sub>, [Poulsen, 2018](https://www.biorxiv.org/content/10.1101/289850v1.full.pdf))
- [ ] PCA
- [ ] ICA

### Features extraction

#### Static properties

- [x] Proportion
- [x] Duration
- [ ] Global explained variance (GEV)
- [ ] ...

#### Dynamic properties

- [ ] Transition probabilities
- [ ] Transition matrix symmetry
- [ ] ...

#### Complexity properties

- [ ] Shannon entropy
- [ ] Empirical entropy rate
- [ ] ...

zen-juen, 2020-07-16:
Though [marjin's method](https://github.com/wmvanvliet/mne_microstates/blob/master/microstates.py) and [frederic's method](https://github.com/Frederic-vW/eeg_microstates/blob/master/eeg_microstates.py) are supposedly implementing the same modified K-means clustering algorithm, the two produces quite different results:

```

DominiqueMakowski, 2020-07-16:
even with the same parameters and input? (i.e., same GFP peak indices and same data?)

zen-juen, 2020-07-16:
Yes even if I do use the same GFP peak indices it is still different 🤔

DominiqueMakowski, 2020-07-16:
With the same data submitted to clustering? Coz I think by default Frederic's version standardizes the data etc. so make sure that preprocessing steps are not included

DominiqueMakowski, 2020-07-16:
and if differences persist, we can try going line by line with the same data to identify at which line the difference occurs, and the see how

zen-juen, 2020-07-20:
Finally I found the part in the kmeans algorithm where frederic's and marjin's method differs 😄 

The main difference is that Frederic's method incorporates the eigenvector method whilst marjin's doesn't (though if you see his code [here](https://github.com/wmvanvliet/mne_microstates/blob/master/microstates.py#L138), the eigenvector was calculated 
 initially though later commented out!) 

Specifically it is this step that is absent from marjin's method:

[image]

DominiqueMakowski, 2020-07-20:
I see, interesting. @wmvanvliet do you by any chance remember the reason for commenting this step out? I.e., the normalization of the eigenvector if I understand correctly?

wmvanvliet, 2020-07-20:
Not sure. If you feel it should be in there, feel free to submit a PR :)

DominiqueMakowski, 2020-07-20:
Haha tbh for now I don't have the slightest idea about what's wrong or what's right. I guess that's why we're trying first to set up a ground for comparison and exploration. Will keep you updated then!

@zen-juen I think you can re-add it then, as the goal here is to stay true to the original publication.

wmvanvliet, 2020-07-20:
Also, make sure that the random values used for initialization of the maps are the same across different algorithms.

zen-juen, 2020-07-21:
Slight difference in the way residual noise assignment in the two algorithms!

Marjin's:
```
        # Estimate residual noise
        act_sum_sq = np.sum(np.sum(states[segmentation].T * data, axis=0) ** 2)
        residual = np.abs(data_sum_sq - act_sum_sq)
        residual /= np.float(n_samples * (n_channels - 1))

        # Next iteration
        prev_residual = residual
        i += 1
```

Frederic's (n_gfp = n_samples here):
```
        # Estimate residual noise and next iteration
        residual = prev_residual
        act_sum_sq = np.sum(np.sum(states[segmentation, :] * data, axis=1) ** 2)
        prev_residual = data_sum_sq - act_sum_sq
        prev_residual /= (n_gfp * (n_channels - 1))
        n_iter += 1
```

closed 2021-07-21 by stale[bot]