neuropsychology/NeuroKit #338: typeerror with nk.ecg_rsp function

saakshipathania, 2020-07-21:
The scipy.signal.butter function is producing error in this function

**Describe the bug**
INPUT
 edr = nk.ecg_rsp(ecg_rate, sampling_rate=100)

ERROR
TypeError: butter() got an unexpected keyword argument 'fs'

**Expected behaviour**
I expected the function to produce an EDR signal samples at frequency 100 hertz.

**Screenshots**
[image]

**System Specifications**

OS: Windows (WindowsPE 64bit) 
- Python: 3.7.1 
- NeuroKit2: 0.0.39

Tam-Pham, 2020-07-21:
Hi @saakshipathania 

May I check with you which version of Scipy you are running on? It seems like your version might be before Scipy 1.2.0 and as`fs` argument should be keyword argument in `scipy.signal.butter()` from Scipy 1.2.0 onwards.

saakshipathania, 2020-07-21:
Hi @Tam-Pham 
Thank you! My version was indeed older than Scipy 1.2.0. Will upgrade right away.

closed 2020-08-06 by DominiqueMakowski