neuropsychology/NeuroKit #340: More PPG and Blood Pressure related resources

zen-juen, 2020-07-22:
Chanced upon this [paper](https://www.sciencedirect.com/science/article/abs/pii/S1746809418302209) on blood pressure estimation from PPG signals and thought it might be useful for enhancing our PPG features 😄 

From raw PPG signals, they estimate
- Mean Arterial Pressure (MAP)
- Diastolic Blood Pressure (DBP)
- Systolic Blood Pressure (SBP) 
(Though note their results on the appropriateness of the algorithms)

Also see this [issue](https://github.com/neuropsychology/NeuroKit/issues/312) regarding Nabian et al.'s package on Blood Pressure preprocessing pipelines

DominiqueMakowski, 2021-08-20:
- [ ] [Characterising the Photoplethysmography Pulse Waveform for Use in Human
Neuroscience: The Hybrid Excess and Decay (HED) Model](https://www.biorxiv.org/content/10.1101/2021.08.19.456935v1.full.pdf)

CrazyPrincipal, 2022-01-18:
Any recent developments on this issue?
Been watching and expecting a result: increased estimation of blood pressure from the PPG signal.

closed 2022-12-21 by stale[bot]