neuropsychology/NeuroKit #351: Issues with the Examples in the NeuroKit2 Paper

TomDonoghue, 2020-08-08:
I am reviewing the Neurokit2 paper (I know this un-blinds me - but I was going to / will sign my review anyways). The paper is also currently available in the `paper` directory in this repository. Since the following is a code issue (I think), and the paper appears to be developed here, I thought I would note an issue I found here (I hope that's fine with the maintainers). 

Basically, I am having some trouble replicating the examples in the paper, which I believe should be 'plug & play', and provide expected outputs, based on what's shown in the paper. 

Output of `nk.version()`:

```
- OS: Darwin ( 64bit) 
- Python: 3.8.3 
- NeuroKit2: 0.0.40 

- NumPy: 1.18.5 
- Pandas: 1.0.5 
- SciPy: 1.5.0 
- sklearn: 0.23.1 
- matplotlib: 3.2.2
```

## Event-related Example

For the first example, of the event related analysis, the example appears to run correctly, but the results seem quite different from what I expected, given what is described in the paper. 

Copy of the code I ran (from the paper)
```
# Load the package
import neurokit2 as nk

# Download the example dataset
data = nk.data("bio_eventrelated_100hz")

# Process the data
df, info = nk.bio_process(ecg=data["ECG"], rsp=data["RSP"],
                          eda=data["EDA"], sampling_rate=100)

# Find events
conditions = ["Negative", "Neutral", "Neutral", "Negative"]
events = nk.events_find(event_channel=data["Photosensor"],
                        threshold_keep='below',
                        event_conditions=conditions)

# Epoch the data
epochs = nk.epochs_create(data=df, events=events,
sampling_rate=100, epochs_start=-0.1, epochs_end=4)

# Extract event related features
results = nk.bio_analyze(epochs) # Show subset of results
results[["Condition", "ECG_Rate_Mean", "RSP_Rate_Mean", "EDA_Peak_Amplitude"]]
```

Results I get:
[image]

Expected results (from the paper):
[image]

## Resting State Example

For the resting state example, the code from the paper unexpectedly hits an error. 

Code I ran (from the paper)
```
# Load the package
import neurokit2 as nk

# Download the example dataset
data = nk.data("bio_resting_5min_100hz")

# Process the data
df, info = nk.bio_process(ecg=data["ECG"], rsp=data["RSP"],
sampling_rate=100)

# Extract features
results = nk.bio_analyze(df)

# Show subset of results
results[["ECG_Rate_Mean", "HRV_RMSSD", "RSP_Rate_Mean", "RSA_P2T_Mean"]]
```

Error I get:

[image]

## Conclusion

I didn't spend too much time to digging into the issue here - I'm not sure if these suggest issues in the code, and/or perhaps some problem with what was described in the paper, etc.

DominiqueMakowski, 2020-08-09:
Hi @TomDonoghue, 

Thank you for commenting directly here and reporting this issue! We will fix it asap ☺️

zen-juen, 2020-08-09:
Hi @TomDonoghue!

You'd have to specify `sampling_rate=100` in `nk.bio_analyze()` as it defaults to a sampling rate of 1000Hz!

Thanks so much for pointing this out in the paper though, definitely an oversight on our part 😅

TomDonoghue, 2020-08-09:
@zen-juen - both code examples in the paper do already the sampling rate, as is also seen in the code I used and posted above, so that does not seem to be explain it, unless the `sampling_rate=100` input is wrong.

zen-juen, 2020-08-09:
In the code you posted above, `sampling_rate=100` is specified in `nk.bio_process()` and `nk.epochs_create()`, but not in `nk.bio_analyze()`. 

Though yes the dissimilarity persists in `ecg_rate_mean` and `rsp_rate_mean` in event-related analysis, and `rsa_p2t_mean` in interval-related analysis 🤔 

Going to put the exact comparisons here so that we can fix this:

## Event-related
```
results = nk.bio_analyze(epochs, sampling_rate=100) # Show subset of results
results[["Condition", "ECG_Rate_Mean", "RSP_Rate_Mean", "EDA_Peak_Amplitude"]]

  Condition  ECG_Rate_Mean  RSP_Rate_Mean  EDA_Peak_Amplitude
1  Negative      -2.010553      -0.152965            0.926195
2   Neutral      -3.127719       1.403998            0.409627
3   Neutral       1.339599      -0.335359            0.019098
4  Negative      -3.552375       1.973311            1.056855

```
[image]

## Interval-related
```
results = nk.bio_analyze(df, sampling_rate=100)
results[["ECG_Rate_Mean", "HRV_RMSSD", "RSP_Rate_Mean", "RSA_P2T_Mean"]]

   ECG_Rate_Mean  HRV_RMSSD  RSP_Rate_Mean  RSA_P2T_Mean
0      86.392105  38.837766      15.735302      0.068919
```
[image]

TomDonoghue, 2020-08-09:
Ahh, sorry, I didn't pay enough attention to which function was being talked about. I can confirm that when I add the sampling rate input to the `bio_analyze` call both run, and provide the same outputs as you posted, which are a bit different from what is printed in the paper.

Tam-Pham, 2020-08-09:
Hi @TomDonoghue 

We are so glad you got in touch to clarify the discrepancies. 

The differences in` ecg_rate_mean` and `rsp_rate_mean` in event-related analysis, and `rsa_p2t_mean` in interval-related analysis are due to the recent major changes that we made to the low-level functions, specifically the default interpolation method used in `signal_period()` and `signal_rate()`. 

We have realized that the previous interpolation methods used were not suitable for the physiological signals, especially for signals that are noisy. The issue was first brought to our attention in #242 and the solution was discussed and finalized in #243. 

Due to an oversight on our part, we didn't update the printed output in the paper after making these major changes to the package. Thank you so much for pointing them out 😃

DominiqueMakowski, 2020-08-09:
Thanks a lot Tom! In any case, we will fix that in the manuscript for the next round whatever it will be ☺️

TomDonoghue, 2020-08-09:
Thanks for the very quick notes & fixes here - the updates / changes / differences all seem to make sense to me! 

I look forward to seeing the updates in the paper, in the next round (for clarity - I filed my review earlier, which mentions this issue, but as a minor thing to fix, and not as some big problem or anything like that).

closed 2020-11-17 by Tam-Pham