neuropsychology/NeuroKit #352: Memory Error in nk.signal_decompose(x, method="ssa",n_components= 5)

saakshipathania, 2020-08-10:
memory error indicated in the line
-> 1562         u, s, vh = gufunc(a, signature=signature, extobj=extobj)

I am dealing with heart beat interval signals with length in the range(16-41000)

[image]

DominiqueMakowski, 2020-08-10:
Hi @saakshipathania, thanks for the report. Could you run `nk.version()` and tell us your system specifications? Also, could you confirm if this error appears with rather long signals (but not shorter ones)?

JanCBrammer, 2020-08-11:
@saakshipathania, thanks for the report. Could you post the entire traceback (copy-paste it from the terminal)?

DominiqueMakowski, 2020-08-12:
@saakshipathania I will copy paste your new issue here as it's best to avoid duplicates:

[image]

**System Specifications**

RAM- 4GB
Internal Storage- 1TB
python: 3.7.1 (default, Dec 10 2018, 22:54:23) [MSC v.1915 64 bit (AMD64)]
neurokit: 0.0.39

And copy paste the output.

----

Now, my guess it's that it's possible that it's a legit memory error, i.e., your array is too big for your computer's memory.

The problem occurs during the SVD decomposition, which is a costly procedure.

There seem to be several documented workarounds, that I'll investigate:

https://stackoverflow.com/questions/7143723/applying-svd-throws-a-memory-error-instantaneously
https://scicomp.stackexchange.com/questions/7566/memory-efficient-implementations-of-partial-singular-value-decompositions-svd

I'll keep you updated.

DominiqueMakowski, 2020-08-12:
@saakshipathania I made an update which might help, you can install the dev version by running `pip install https://github.com/neuropsychology/neurokit/zipball/dev` and let us know if it still errors.

saakshipathania, 2020-08-12:
- [ ] 

> @saakshipathania I made an update which might help, you can install the dev version by running `pip install https://github.com/neuropsychology/neurokit/zipball/dev` and let us know if it still errors.
@DominiqueMakowski  Thank you the updated version fixed the problem!

closed 2020-08-12 by DominiqueMakowski