neuropsychology/NeuroKit #353: Manuscript Revision Checklist

zen-juen, 2020-08-11:
The Neurokit manuscript was sent back for revision albeit with pretty positive and helpful suggestions for changes 😄 @DominiqueMakowski @Tam-Pham  @JanCBrammer @sangfrois Putting a checklist here to track our progress before resubmitting!

## Major Issues
### Fix **examples**
- [ ] Reviewer 1: Invert the examples? Reviewer thinks it's weird to see that the second example is a copy of the first, with one less step. Perhaps it would be more appropriate to see the simpler example first, and then add the epoching step later on
- [ ] Reviewer 2: Inconsistent example results (https://github.com/neuropsychology/NeuroKit/issues/351), examine and address the improved results, provide expected results with the current release version + specify code versions used to run the examples

### Describing and expanding on **Functionality**
Talk about the functionality available early in the paper for readers not familiar with nk
- [ ] What analyses can be applied, not what modalities are supported
- [ ] Additional descriptions of available tooling (examples are useful but largely describe higher level functionality (clean, process, etc), which are somewhat opaque as to what processes are applied to the data) - can talk about nk's functions for simulating data, plotting options
- [ ] Add extra paragraph (page 4) succinctly listing out functionality available

> e.g. "NeuroKit2 provides functionality such data and event management; signal processing including filtering, resampling, ...; spectral analyses such as computing power spectral densities; statistical analyses such as ...; complexity analyses such as ..., plotting such as ...". 

- [ ] If there are more workflows described by prior publications available, such as the 'khodadad2018' method noted later in the paper, some more of these could be mentioned and cited, as examples.

### Noting Programming Requirements & Interface
- [ ] Make intro and discussion clearer that nk is used as a Python interface + what level of Python required from users (specifically, that nk *does* require some programming)
  -  Not very clear how nk addresses the challenge of GUIs not being reproducible (we also state that many algorithms are inaccessible to those without programming experience, which may suggest nk has a GUI interface) Slightly later the article notes that GUIs are not very reproducible (which I believe is a fair comment).
  - First line of discussion overstates nk's accessibility to all levels of experience since for users to validate and apply the different pipeline approaches on data requires some knowledge of processing + programming: 

> E.g., "Neurokit2 does require using Python. For the beginner program, the documentation also includes guidance on getting started with Python"

## Minor Issues
- [ ] Replace "alot of parameters” (line 114) with “several” or “multiple”. 
- [ ] Add current active users / collaborators (line 88)
- [ ] User interface usually shortened to UI (line 100), refers to the visual aspect of an interface
- [ ] Term **user friendly** (line 62): not wise to use this term, replace with term **user-centered which is instead related to the Design Approach used
- [ ] Term **professional** (line 71): Avoid using the term unless design guidelines and widely adopted conventions have been used and reported.
- [ ] Replace word "upload" with "submitting" (line 83) as former may suggest immediate feedback for newcomers not familiar with github
- [ ] Cite other software packages (line 75, page 4): describe citations that can be used for numpy, matplotlib, etc: https://www.scipy.org/citing.html
- [ ] Jupyter notebooks not Python notebooks (line 84, page 4)
- [ ] Unclear what peer-reviewed processing pipelines mean (line 90, page 5) is it implementations of these pipelines described in peer-reviewed literature or is it a peer-review system that reviews code for processing pipelines?
- [ ] Describe and spell out acronyms e.g., `RSP` (line 121, page 6 example) *start looking from page 5 line 64 onwards
- [ ] Duplication of word *three* (line 110, page 5)

cnboonhan94, 2020-08-11:
Nice

DominiqueMakowski, 2020-08-11:
Maybe we should put each "point" in a separate issue so we can also comment how we addressed said-point. It will be easier to later make the "response to reviewers" file that goes along the resubmission, in which we have to say to each point how exactly we have addressed it, indicate the lines of the change and paste the updated paragraph.

We could also link each issue in that file, so the reviewers can exactly see the commits referencing each of their point and the eventual discussions surrounding them. I mean it might be pushing open science too far but 🤷‍♂️

sangfrois, 2020-08-11:
These are constructive comments! Good job for putting this together in a comprehensible way. 

@DominiqueMakowski I have a suggestion that could answer to your last comment : would it be appropriate if I open an issue (or a PR?) that aggregates 2), 6). I see my input on binder/examples and on tutorials quite well together. we could also reference the issue (or PR) related to each point in the description here. Would that be suitable ? I'll let you judge and initiate the proper move accordingly.

We have to continue to push for open science, and not fall under pressure (that's my take on the subject 👍 ).

DominiqueMakowski, 2020-08-12:
@sangfrois yeah do go ahead. though I'd say we can open 2 separate issues for 2 and 6 and link the PR(s) to them accordingly (the same PR can be linked to both) but it will be easier for us to have as specific issues as possible.

Also, I think I'll create an `R1.md` file in /paper/ with our answers to reviewers that we can update alongside our fixes, It'll be easier than doing it at the end ☺️

sangfrois, 2020-08-19:
All right!! I will start by 2). 

Also : I've put together a google document so we could have a formatted version of original reviewer's comments.

[shared google doc](https://docs.google.com/document/d/1pC18MtedYd-WE_dQ1exjxNorgJWBabgU179HldZEYtM/edit?usp=sharing)

zen-juen, 2020-08-28:
Hey guys, do check out the individual issues opened for fixing the manuscript 😄

I tried to ensure one commit = one issue fixed so its easier to track the changes!

zen-juen, 2020-09-25:
@sangfrois Let us know what you think about the commented parts  regarding the examples and then we can go ahead and finalize everything https://docs.google.com/document/d/1pC18MtedYd-WE_dQ1exjxNorgJWBabgU179HldZEYtM/edit 😄

JanCBrammer, 2020-10-19:
@DominiqueMakowski , @zen-juen, @Tam-Pham, @sangfrois, as discussed, I just pushed my final edits (changed the corresponding sections in the Google doc as well.)

DominiqueMakowski, 2020-10-19:
perfect, thanks, we'll generate the manuscript and do a final pass tomorrow ☺️

closed 2020-11-17 by Tam-Pham