neuropsychology/NeuroKit #355: Capturing Data (e.g. ECG) Directly From Arduino Platforms Via Serial

roryyeung, 2020-08-14:
**Describe the solution you'd like**
I would like to integrate a s

**How could we do it?**
A description of actual ways of implementing a feature.

DominiqueMakowski, 2020-08-15:
Hi @roryyeung, sounds very exciting!

I'm not very sure to what extent it encompasses adding new functions to the software, in which case we'll have to see how to best do that. My recommendation would be that you can start adding an ["example"](https://github.com/neuropsychology/NeuroKit/tree/master/docs/examples) doc as a guide on how to integrate arduino and NK. (I believe that would be very useful, and I'll be personally looking forward to that!!). You can also as you mentioned add to the record quality signals tutorial.

If you need to create any specific functions to help that integration, you can start by defining them in the "Record data from arduino" tutorial, and from there we can see how to eventually move them directly to the software.

Is it what you had in mind?

roryyeung, 2020-08-19:
Sure, that works for me!

Obviously Serial allready has a lot of the functionality required, so it would effectively be a matter of defining a function that accepts the ECG parameters, sets up the connection to the Arduino (or any other similar chipset) accordingly.

I'll start putting it together. I'll let you know how I get along. With the tutorial etc.

csavur, 2020-11-17:
@roryyeung, I like the idea however I am not sure if it makes sense to integrate with NeuroKit. NeuroKit is an offline signal processing library. That's way, it would make more sense to record data separately then use NK for processing.

As a solution, I would suggest you to take a look at https://github.com/sccn/labstreaminglayer, LSL is a library that allows you to send and receive data in real-time (nearly) in a local network and capability of recording to XDF file. After data collection, you can load the XDF file and use NK to analyse the signals. 

This is the way I am collecting signals from multip devices in a local network and analyzing it with NR.  Let me know if you have any questions regarding LSL.

closed 2021-07-15 by stale[bot]