neuropsychology/NeuroKit #359: Pseudo-colouring of ECG to help detect abnormalities

DominiqueMakowski, 2020-09-01:
Alahmadi et al.  just published [a paper](https://journals.plos.org/plosone/article?id=10.1371/journal.pone.0237854) suggested coloring ECG signals so that abnormal components are more easily detected by the naked eyes. 

Could be interested to add as a feature to `ecg_plot()`, especially since its aesthetically quite nice and "original" (and hopefully wouldn't be too heavy as the coloring goes to an empty part - under the curve).

[image]

likith012, 2020-11-09:
When can we expect this feature to be available?

DominiqueMakowski, 2020-11-09:
No work in the pipes as of now, and it's not on the immediate roadmap 😕 that said, it should be too hard to code (mostly a matter of time for us), feel free to give it a try we can at least assist you!

likith012, 2020-11-10:
Sure, will look into that during my free time.

likith012, 2020-11-10:
Also, I am dealing with a heavily imbalanced 12 led ECG dataset (PTB xl), any suggestions on how to generate synthetic data for the undersampled ECG classes. I tried SMOTE but it didn't improve my performance score and GAN's is out of option(as they produce bad results for generating 1D signals but tend to give good results for generating spectrograms).

DominiqueMakowski, 2020-11-10:
I'm afraid that's outside of my domain 😕 good luck tho

closed 2021-07-08 by stale[bot]