neuropsychology/NeuroKit #362: Information about algorithms used in NeuroKit2

sitadrost, 2020-09-11:
Dear Developers,

Thanks a lot for sharing this project! I installed NeuroKit2 this morning, and so far I'm loving it! Just a quick question: where can I find more information about the algorithms used in this package? Reading the code gives some of this information, but say I'd like to know more about details, background, performance tests, etc. where should I look?

Many thanks,
Sita

DominiqueMakowski, 2020-09-11:
Hi @sitadrost that's a very good question. Currently it is true we don't have a place where we describe with enough details the algorithms (and probable sometimes we might even have missed some of the references).

Its a work in progress and the package has seen a rapid initial growth focused on adding features, and surely not enough documentation. and we hope to be able to add it little by little. That being said, we would appreciate a ton if for instance you managed to understand some of the  algorithms, if you could make a small contribution to slowly help us fill the missing bits ☺️

sitadrost, 2020-09-13:
Hi Dominic,
Thanks for getting back so quickly. That's too bad, that this information doesn't quite exist yet. If I come across a relevant reference I'll surely let you know.
Cheers,
Sita

closed 2021-07-21 by stale[bot]