neuropsychology/NeuroKit #368: Do ECG analysis functions require R peaks to be pointing upwards?

Miceuz, 2020-10-07:
We have tried to use NueroKit to process the ECG like this with very poor results:
[image]

[image]

Our cardiologist said that ECG signal like this is to be expected as a normal ECG on Lead I, orientation of electrical axis of a heart can result in R peak inverted like that. This particular signal was recorded with electrodes placed on chest in normal position for Lead I recording.

Is there a way to trick NeuroKit to correctly delineate such signal? We have tried inverting the signal, then R-peaks are detected correctly, then we have tried to use the detected peaks to delineate the signal, but accuracy is poor.

Can you please comment on this? I am an experienced programmer and I could put some effort into this project.

I am attaching both data and notebook illustrating the issue.
[neurokit-test.zip](https://github.com/neuropsychology/NeuroKit/files/5341150/neurokit-test.zip)

DominiqueMakowski, 2020-10-07:
Yes, neurokit expects for now the peaks to be oriented upwards. What I would suggest is simply to reverse your ECG vector (i.e., multiply it by -1)... hopefully it should work 😁

DominiqueMakowski, 2020-10-07:
Apologies, I read too quickly. Let us see what's going on :)

Miceuz, 2020-10-07:
Thanks for your answer @DominiqueMakowski yes, we have tried reversing the ECG - R peaks are detected correctly in this case, but delineation still looks wrong.

[image]

[image]

Tam-Pham, 2020-10-07:
Hi @Miceuz 

Could you try to apply the delineate function again but with a different delineate method, say `method=dwt`?

Just so that we know if this needs to be addressed for all methods. Thanks!

Miceuz, 2020-10-07:
Hi @Tam-Pham,

Yes, sure, 
# Results with DWT
[image]
[image]

# With CWT:
[image]

while this somehow crashes:
[image]

Tam-Pham, 2020-10-07:
It seems like DWT manages to identify the P-peaks and T-peaks a little better when the signal is original. Could you do me one last favour, to set the show_type to all for the DWT method? 
Thanks!

Miceuz, 2020-10-07:
Sure thing

[image]
[image]

Tam-Pham, 2020-10-07:
Alright. It seems like DWT is doing great here, with a few exception. This window is probably a bit too long for your QRS complex and that's why the messiness at the end (it includes the information of the adjacent complex).

We will investigate and try to improve the peak method. But for now, is the return of DWT sufficient for you?

Miceuz, 2020-10-07:
We are just evaluating NeuroKit right now, so this is totally fine. Thank you very much!

JanCBrammer, 2020-10-08:
> while this somehow crashes

@Miceuz, note that this has been fixed on `dev` (#347, version 0.0.41) already.

JanCBrammer, 2020-10-08:
Looks like in this case, only the R-peaks are inverted, not the P and T waves. This could lead to a scenario where after inversion, the R peaks are detected fine, but the delineation struggles to correctly detect P or T waves. @Tam-Pham, in how far does the delineation rely on non-inverted P and T waves?

DominiqueMakowski, 2020-10-08:
Indeed indeed, good catch @JanCBrammer! that's the cause of this issue I think. I wonder if we could selectively revert the R-peaks or - do the R peak detection on the reverted signal :and the delineation on the raw signal 🤔

JanCBrammer, 2020-10-08:
Although I believe that @Miceuz already worked around that by doing:

```
rpeaks = find_peaks(inverted_ecg)
waves = delineate(original_ecg, rpeaks)
```

DominiqueMakowski, 2020-10-08:
right - I need to change my glasses 🤓

Miceuz, 2020-10-08:
@JanCBrammer thanks, I will check it out.

Tam-Pham, 2020-10-08:
@JanCBrammer It actually matters. If you look at @Miceuz 's plot for original and inverted signals:

> [image]
> [image]

The delineation works alright with the original signal where P and T waves are NOT inverted. I guess for this particular case, users have to inverse the signal to identify R-peaks and use the original signal for the delineation. 

P/S: we should indeed merge the dev branch soon 😅 
P/SS: @DominiqueMakowski yah u need new glasses or slow down your reading 😄

JanCBrammer, 2020-10-08:
Thanks @Tam-Pham. So it looks like ideally we would find a way to detect inverted R-peaks and selectively re-invert them 😬

JanCBrammer, 2020-10-08:
@Miceuz, which method did you use for R-peak detection?

neurokit (default), pamtompkins1985, hamilton2002, christov2004, gamboa2008, elgendi2010, engzeemod2012, kalidas2017, martinez2003, rodrigues2020 or promac?

Miceuz, 2020-10-08:
@JanCBrammer  it was neurokit

I am attaching my python notebook and source data with updated DWT/CWT experiments

[neurokit-test.zip](https://github.com/neuropsychology/NeuroKit/files/5345765/neurokit-test.zip)

closed 2021-06-05 by stale[bot]