neuropsychology/NeuroKit #369: HRV Summary

johnnyzhang295, 2020-10-08:
Hello, I am looking for a summary or description of what each column represents in ecg_intervalrelated

(ECG_Rate_Mean, HRV_RMSSD, HRV_MeanNN...)

zen-juen, 2020-10-08:
Hi @johnnyzhang295 , thanks for reaching out! 

The brief descriptions of the output are available in our `ecg_intervalrelated()` docstrings [here](https://neurokit2.readthedocs.io/en/dev/functions.html#neurokit2.ecg.ecg_intervalrelated). However, as there are several heart rate variability indices that are also generated from `ecg_intervalrelated()`, you can find them described in more detail on this [readthedocs page](https://neurokit2.readthedocs.io/en/dev/functions.html#module-neurokit2.hrv). Do note that our hrv functions are organized across time-domain, frequency-domain and non-linear domain indices (so for example, the description of `HRV_RMSSD` would be housed under the [hrv_time()](https://neurokit2.readthedocs.io/en/dev/functions.html#neurokit2.hrv.hrv_time) function.)

johnnyzhang295, 2020-10-08:
Thanks. I am also wondering, @zen-juen , do you have a metric for **QT Interval Variability?**

zen-juen, 2020-10-09:
I'm afraid we don't have that for now though it seems like a promising direction to head towards! @Tam-Pham could you confirm? 😄

DominiqueMakowski, 2020-10-09:
> do you have a metric for QT Interval Variability?

This sounds relatively straightforward to add. @johnnyzhang295 would you have more details and info about how it is computed, how it is used etc.?

johnnyzhang295, 2020-10-09:
@DominiqueMakowski I do not know anything about how it would be computed, sorry. I am just interested in pulling that metric out for my data analysis

closed 2021-06-07 by stale[bot]