neuropsychology/NeuroKit #370: EDA, eda_analyze does not work.

zhaoyanxi, 2020-10-23:
When I use fuction eda_analyze, I find it does not work and the Error is UnboundLocalError: local variable 'features' referenced before assignment. Please try to fix it.

JanCBrammer, 2020-10-23:
@zhaoyanxi , please post a [minimal, reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) that exactly reproduces the issue. Also, please include the full traceback (i.e., error message) in your example. Then it will be easier for us to help you. If possible also upload the signal that produced the error.

bananakiu, 2021-03-15:
Reporting that this is the same case for rsp_analyze.

Same issue:  UnboundLocalError: local variable 'features' referenced before assignment.

JanCBrammer, 2021-03-15:
@bananakiu, can you post your script that causes the issue?

closed 2022-03-16 by stale[bot]