neuropsychology/NeuroKit #375: nk.rsp_intervalrelated doesn't work with epochs

johnnyzhang295, 2020-11-03:
It gives me this value when I provide my epoch: 

`ValueError: ('NeuroKit error: complexity_embedding(): dimension * delay should be lower than', ' the length of the signal.')`

There must be something wrong because I made an ECG epoch with the same event info and nk.ecg_intervalrelated worked just fine.

johnnyzhang295, 2020-11-03:
My setup: Both the ECG and respiratory gauge are going through Biopac data acquisition at 2000 Hertz.

Tam-Pham, 2020-11-04:
hi @johnnyzhang295 could you copy the whole error thread so that it's easier to pinpoint where the error comes from.

thanks!

DominiqueMakowski, 2020-11-04:
And ideally, send us the data that generates the error

closed 2021-07-02 by stale[bot]

zoubochao, 2021-08-13:
rsp_epochs = nk.epochs_create(rsp_df, events, sampling_rate=1000, epochs_start=0, epochs_end=epoch_duration)
nk.rsp_intervalrelated(rsp_epochs)

same error here， any solutions? thx

zoubochao, 2021-08-13:
@johnnyzhang295

zen-juen, 2021-08-13:
> rsp_epochs = nk.epochs_create(rsp_df, events, sampling_rate=1000, epochs_start=0, epochs_end=epoch_duration)
> nk.rsp_intervalrelated(rsp_epochs)
> 
> same error here， any solutions? thx

hi @zoubochao given that this issue is closed, could you open up a new issue along with a sample of your data + a reproducible script so that we can replicate this error? thanks 😄

zzaggreus, 2022-06-29:
Hi, is there any way to use this package to get time and freq domain variables of HRV from RR data (IBI data)? We do not have ECG data and we do have IBI data collected using Polar chestband. Thank you

Preston-Hunter, 2024-12-17:
Old issue But for others who have come across if, this can be in part caused by your epoch containing too few data points where data["RSP_Troughs"] == 1. Generally speaking if your epoch contains less than 4 such data points you will get the above error.

Dimension and delay defaults to 2 and 1 respectively. The signal in question is actually the intervals between the respiratory troughs data points. So 3 data points would result in a signal of length 2, which is not greater than 2*1 (dimension * delay)

I do not know if this was johnny's issue specifically. but it is something that I hope may help others