neuropsychology/NeuroKit #376: Reading data collected from BiTalino system

Tam-Pham, 2020-11-03:
@MeenaSiddharthan you mentioned that you tried to use NeuroKit to get the HR value of your ECG signal collected using Bitalino and you ran into some issue? 

Could you elaborate more about what errors did you get, the script that you used and also share with us a sample of your data here?

Let's investigate it together!

DominiqueMakowski, 2020-11-03:
Or via email if sensitive data

JanCBrammer, 2020-11-03:
OpenSignals (aka Bitalino) files that have been saved in `.txt` format can be read using `pandas.read_csv()` like [this](https://github.com/JanCBrammer/biopeaks/blob/5bfc545289f5bbf6387cc20b9c6ca202a12bb303/biopeaks/io_utils.py#L152).

DominiqueMakowski, 2020-11-04:
```python
import neurokit2 as nk

data, sampling_rate = nk.read_bitalino("data.txt")
nk.signal_plot(data[0:15][["ECGBIT", "RESPBIT"]], standardize=True)
```

[image]

@MeenaSiddharthan I have added a basic function (currently on the development version) for reading the files, but it seems that the columns are mislabled (ECG seems to be RSP and vice versa). I'm not sure if it's a problem of the recording itself (could the channels were misplaced during recording) or of the label retrieval.

MeenaSiddharthan, 2020-11-04:
@DominiqueMakowski Hi Dom thanks a bunch! Yeah I got that graph too with the flipped values. Oddly the columns were labelled correctly in the other .txt file from Bitalino we had. Soooo not sure what happened here. I should be able to figure out the HRV now!!

MeenaSiddharthan, 2020-11-05:
Hey! @DominiqueMakowski or @Tam-Pham so I've been trying to figure out HRV analysis on Neurokit and it seemed straight forward on the manual but I'm stuck trying to get the RMSSD values using hrv_time. So this is what I did after getting cleaning ECG signal and getting peaks.

hrvdata = nk.hrv_time(peaks, sampling_rate=100, show=True)
print(hrvdata)

I get the boxplot but I can't get the table of values. I get these errors:
TypeError: boxplot() got an unexpected keyword argument 'manage_ticks'
NameError: name 'hrvdata' is not defined

I'm not sure how to work back from this!

closed 2020-11-23 by Tam-Pham