neuropsychology/NeuroKit #378: signal_fixpeaks() results in infinity loop

richrobe, 2020-11-10:
I just had a very rare edge case where I wanted to correct R peak locations for HRV analysis using `signal_fixpeaks()` with the default parameters `(method='kubios', iterative=True)` and ended up in an infinity loop.

After some investigation I found out that it's caused by the iterative R peak correction approach (see below).

The while loop (difference between `n_artifacts_previous` and `n_artifacts_current`) never converges, but always jumps around between -1 and 1. I attached the file with the R peak indices: [rpeaks.txt](https://github.com/neuropsychology/NeuroKit/files/5520044/rpeaks.txt)

I think this only happens on very rare edge cases, so maybe an easy way to fix this would just be to introduce a `max_iterations` flag to `signal_fixpeaks()` as additional termination criteria?

```
    if iterative:
        # Iteratively apply the artifact correction until the number of artifact
        # reaches an equilibrium (i.e., the number of artifacts does not change
        # anymore from one iteration to the next).
        n_artifacts_previous = np.inf
        n_artifacts_current = sum([len(i) for i in artifacts.values()])

        previous_diff = 0

        while n_artifacts_current - n_artifacts_previous != previous_diff:
            print("HERE THE ERROR HAPPENS. This never converges.")

            previous_diff = n_artifacts_previous - n_artifacts_current

            artifacts, subspaces = _find_artifacts(peaks_clean, sampling_rate=sampling_rate)
            peaks_clean = _correct_artifacts(artifacts, peaks_clean)

            n_artifacts_previous = n_artifacts_current
            n_artifacts_current = sum([len(i) for i in artifacts.values()])
```

**To Reproduce**
```
import neurokit2 as nk
rpeaks = pd.read_csv("./rpeaks.txt", header=None)
nk.signal_fixpeaks(rpeaks.values)
```

**System Specifications**

```
- OS: Darwin ( 64bit) 
- Python: 3.8.3 
- NeuroKit2: 0.0.40 

- NumPy: 1.18.5 
- Pandas: 1.1.4 
- SciPy: 1.5.2 
- sklearn: 0.23.1 
- matplotlib: 3.2.2
```

DominiqueMakowski, 2020-11-11:
Hey @richrobe, nice catch! Maybe you can open a PR with a suggested fix if you have one, and we can see how it goes (and if it doesn't break tests ☺️)

JanCBrammer, 2020-11-11:
@richrobe, thanks for reporting this, very interesting! I could replicate the issue with your `rpeaks.txt`.

What I notice is that

```Python
n_artifacts_previous = np.inf
n_artifacts_current = sum([len(i) for i in artifacts.values()])

previous_diff = 0

while n_artifacts_current - n_artifacts_previous != previous_diff:

    previous_diff = n_artifacts_previous - n_artifacts_current

    artifacts, subspaces = _find_artifacts(peaks_clean, sampling_rate=sampling_rate)
    peaks_clean = _correct_artifacts(artifacts, peaks_clean)

    n_artifacts_previous = n_artifacts_current
    n_artifacts_current = sum([len(i) for i in artifacts.values()])
```

can be simplified to 

```Python
n_artifacts_previous = np.inf
n_artifacts_current = sum([len(i) for i in artifacts.values()])

while n_artifacts_current != n_artifacts_previous:

    artifacts, subspaces = _find_artifacts(peaks_clean, sampling_rate=sampling_rate)
    peaks_clean = _correct_artifacts(artifacts, peaks_clean)

    n_artifacts_previous = n_artifacts_current
    n_artifacts_current = sum([len(i) for i in artifacts.values()])
```

This (potentially) solves the convergence issue, since it does not depend on the sign of `previous_diff`. It solves the convergce issue in your example.

jukka, 2020-12-25:
I wonder if a better condition for stopping the iteration would be when the number of detected artefacts stops decreasing? That would guarantee eventual termination (can't decrease below zero) without the need to come up with some specific `max_iterations` limit.

jukka, 2020-12-25:
I implemented the above idea in https://github.com/jukka/NeuroKit/commit/07683f48360ce8918c53f73330e149b890c7c9f8, and verified that it indeed avoids the infinite loop on rpeaks.txt.

Unfortunately this alternative auto-stopping logic results in some quality loss at least as measured by RMSSD in the unit tests.

JanCBrammer, 2020-12-27:
Welcome @jukka and thanks for thinking along!

> Unfortunately this alternative auto-stopping logic results in some quality loss

Note that the stopping criterion you propose actually _improves_ the quality of the artifact correction on this particular unit test. See https://github.com/jukka/NeuroKit/commit/07683f48360ce8918c53f73330e149b890c7c9f8#r45446544.

Could you make a PR with the patch?

JanCBrammer, 2020-12-29:
@jukka, thanks for the patch! It'll be available in main once we release 0.1.0.

@richrobe, we can revisit this issue should we want to experiment with alternative stopping criteria.

closed 2021-02-08 by DominiqueMakowski