neuropsychology/NeuroKit #379: ECG systole/diastole start definition

sitadrost, 2020-11-12:
Just a quick question out of curiosity: among the results given by ecg_process are the starting points of ventricular systole and diastole, which, as far as I could see, you define as R-peak and end of T-wave (T-off), respectively. On what literature source do you base this? 

Looking around in textbooks and on the internet, there seem to be different definitions, especially for the start of ventricular diastole, which I've seen placed at the T-peak, at the end of the T-wave, and at some point roughly halfway in between. Not sure where these differences come from, perhaps it has to do with the peak of the T-wave being easier to reliably detect than its end, I don't know. Could you shed some light on this?

Many thanks!
Sita

Tam-Pham, 2020-11-12:
Hi @sitadrost 

Tbh, I'm not an expert either. I realize the arbitrary definitions of the beginning and the end of each phase and when deciding the thresholds, I refer to the relative pressure in atria and ventricles as shown in the [Wiggers diagram](https://en.wikipedia.org/wiki/Wiggers_diagram) below:

[image]

And as you can see, the end of the T-wave is when the ventricular pressure starts to flatten out and the ventricular volume starts to increase. Hence, we pick the end of the T-wave as the starting point for the ventricular diastole phase. The discrepancy could arise due to different references. For instance, while we refer to the pressure in the compartments, others might rely on the phonocardiogram information (which might suggest an earlier beginning of ventricular diastole). 

If you could find any other reliable sources to determine the thresholds, please share with us :)

sitadrost, 2020-11-12:
Hi @Tam-Pham ,

Thanks for that! This Wiggers diagram is one of my favourite resources actually; it gives such a nice, clear overview of all these signals at once (although I interpreted it as putting the start of diastole halfway between the T-peak and the end of the T-wave (looking at the solid vertical lines), but as you say, that depends on which signal you're looking at). 

Unfortunately I haven't yet found any authoritative source on this; I was hoping you might be aware of one, hence my question. Besides, the ECG signals I'm working with at the moment are mostly not of great quality anyway, so I'm afraid I'll have to stick with just the T-peak plus some offset for now, as it's the only feature that I can more or less reliably detect.

DominiqueMakowski, 2020-11-12:
> I interpreted it as putting the start of diastole halfway between the T-peak and the end of the T-wave (looking at the solid vertical lines), but as you say, that depends on which signal you're looking at).

To be fair I don't think that having hard boundaries make a lot of physiological sense, since it's a dynamical process anyway. That's why we also added the information about the cardiac phase completion which can be used a some index of phase "certainty" or "intensity" (i.e., near the boundaries of the phase, what phase we're in becomes less relevant)

sitadrost, 2020-11-12:
That's a fair point, makes sense. The thing is, I'm using the ECG signal to compute systolic and diastolic flow volume in coronary arteries, and as the flow rate signal tends to be quite steep here at the start of systole/diastole, the exact timing of systole/diastole start makes a lot of difference. But all things considered, I think I should just choose the definition that makes most sense for my signals, and consistently stick to that.

Anyway, thanks for your thoughts on this!

closed 2020-11-12 by DominiqueMakowski