neuropsychology/NeuroKit #380: A study on GAM-based approaches to EEG/ERP analysis

carldelfin, 2020-11-13:
After Googling a bit about [GAMs](https://en.wikipedia.org/wiki/Generalized_additive_model) and whether they'd be suitable for EEG/ERP analysis, I stumbled upon [this](https://neurokit2.readthedocs.io/en/latest/studies/erp_gam.html) page on the NeuroKit docs. I contacted Dominique, who suggested a study in the form of a tutorial on how to do analyze ERPs using GAMs (with R and Python).

**If you're interested in collaborating on this, let us know!** For starters, here's a few questions to ponder that may help us get going:

* What (in relatively simple terms) is a GAM (and/or other spline-based methods)?
* What is the added value of a GAM-based approach vis-a-vis current "standard" approaches (i.e., why should people bother)?
* How can the added value of GAMs be evaluated and presented to the reader?
* Does a GAM-based approach bring any additional (statistical, methodological) assumptions that need to be considered?
* (Bonus: Are Bayesian GAMs based on MCMC sampling feasible? Are other Bayesian approaches feasible?)

DominiqueMakowski, 2020-11-13:
Some thoughts regarding the "Bayesian perspective" to discuss the tradeoff of Bayesian sampling:

- Computational power issue with traditional MCMC
  - What are the advantages of it? Better inference?
  - Could variational algorithms be a viable alternative?
- The role of priors
  - It appears as a challenge to find informative prior 
  - Maybe priors could be used as a penalization parameter (i.e., have more narrow priors to avoid possible overfitting / parameters going crazy because of some extreme observations - e.g., bad channels or artifacts)

DominiqueMakowski, 2020-11-13:
> What (in relatively simple terms) is a GAM (and/or other spline-based methods)?

Yes this very important as GAMs are still fairly new, especially in psychology / neuroscience. 

So far I'd see two ways of starting, either we start by presenting GAMs and then be like "this appears as well-suited for ERPs", or we start by briefly mentioning the current ERPs typical approaches and their limitations, and then how GAMs could be a potential avenue to tackle these limitations

DominiqueMakowski, 2020-11-13:
Something like: 

### 1. GAMs and their relevance to ERP/EEG

- What (in relatively simple terms) is a GAM (and/or other spline-based methods)?
- What is the added value of a GAM-based approach vis-a-vis current "standard" approaches (i.e., why should people bother)?

### 2. Hands-on example of how to use them, interpret and report them
  
- How can the added value of GAMs be evaluated and presented to the reader?

### 3. Methodological Caveats

- Does a GAM-based approach bring any additional (statistical, methodological) assumptions that need to be considered?
- What users should be aware of, what are the important things to check / verify / justify (typically with GAM, the amount of wiggleness allowed is critical)

### Limitations and future developments

- Bayesian framework: is it feasible and why is it desirable?
  - Computational power issue with traditional MCMC
    - What are the advantages of it? Better inference?
    - Could variational algorithms be a viable alternative?
  - The role of priors
    - It appears as a challenge to find informative prior
    - Maybe priors could be used as a penalization parameter (i.e., have more narrow priors to avoid possible overfitting / parameters going crazy because of some extreme observations - e.g., bad channels or artifacts)

carldelfin, 2020-11-19:
Great, yep, that looks like the most promising structure!

I've started to do some literature search to see whether anyone has actually used GAMs for ERP analysis, and found [this](https://journals.plos.org/plosone/article?id=10.1371/journal.pone.0143328#pone.0143328.ref038), with code available [here](https://journals.plos.org/plosone/article/file?type=supplementary&id=info:doi/10.1371/journal.pone.0143328.s002), [this](https://www.frontiersin.org/articles/10.3389/fpsyg.2015.00077/full), [this](https://www.sciencedirect.com/science/article/abs/pii/S0093934X12000958?via%3Dihub), and [this](https://pubmed.ncbi.nlm.nih.gov/25132114/). The latter study reads a bit like a tutorial but isn't really that "hands-on". It also contains a fair bit of the math underlying GAMs, which may be useful (e.g., we can give a brief overview and refer to that study, for instance, and then focus more on practicalities).

There's also [this](https://repository.ubn.ru.nl/bitstream/handle/2066/162511/162511.pdf?sequence=1) study about different ways to analyze EEG data, and [these](http://www.let.rug.nl/wieling/Statistics/GAM-EEG) lecture slides. All in all, there's a lot of useful information in these studies that should help us get going with the introduction and methodological caveats/limitations. I believe that our study would be a good fit, with a more practical, hands-on approach compared to previous work.

DominiqueMakowski, 2021-01-26:
There is a [special issue](https://www.mdpi.com/journal/sensors/special_issues/Brain_Signals_Acquisition_Processing) in the sensor journal on "Brain Signals Acquisition and Processing". It could be a really great fit for that, but the deadline is 15 April 2021 so it might be a bit too soon 😁

carldelfin, 2021-02-08:
Sorry for the late reply! I agree, that does sound like a great fit, although yes, the deadline might be a bit tight... but we might as well try to get going? :-) What's the best way forward, just create an .Rmd and start writing?

closed 2022-02-03 by stale[bot]