neuropsychology/NeuroKit #387: Possible overflow in ecg_delineate

nx10, 2020-11-20:
I am processing a rather large ECG recording of 3.5 hours at 250Hz.
`ecg_peaks` is working as expected and gives me all R peaks:

```Python
_, info = nk.ecg_peaks(signal, sampling_rate=250, method="neurokit")
rpeaks = info["ECG_R_Peaks"]
```

However if I call `ecg_delineate` like so:

```Python
_, waves_peak = nk.ecg_delineate(
        signal, rpeaks, sampling_rate=fs, method='dwt')
```

the returned positions seem to overflow:

beat | R_peak | T_peak | T_on | T_off | P_peak | P_on | P_off | R_on | R_off
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
... | ... | ... | ... | ... | ... | ... | ... | ... | ...
4209 | 1073140 | 1073211 | 1073193 | 1073230 | 1073110 | 1073101 | 1073121 | 1073113 | 1073155
4210 | 1073410 | 1073481 | 1073464 | 1073500 | 1073380 | 1073371 | 1073390 | 1073382 | 1073424
4211 | 1073681 | 1073752 | 1073735 | 1073771 | 1073652 | 1073642 | 1073662 | 1073654 | 1073696
4212 | 1073942 | -1073463 | -1073470 | -1073456 | -1073580 | -1073586 | -1073573 | -1073546 | -1073523
4213 | 1074210 | -1073198 | -1073204 | -1073190 | -1073348 | -1073361 | -1073316 | -1073284 | -1073261
4214 | 1074483 | -1072930 | -1072936 | -1072922 | -1073077 | -1073094 | -1073056 | -1073021 | -1072982
... | ... | ... | ... | ... | ... | ... | ... | ... | ...

**To Reproduce**
Call `ecg_delineate` with a large signal array (as described above).

**Expected behaviour**
Component positions should always be positive. They seem to overflow.

**System Specifications**

- OS: Windows (WindowsPE 64bit)
- Python: 3.7.9
- NeuroKit2: 0.0.41

- NumPy: 1.19.1
- Pandas: 1.0.5
- SciPy: 1.5.2
- sklearn: 0.23.2
- matplotlib: 3.3.1

JanCBrammer, 2020-11-25:
@nx10, I can replicate the issue with a simulated ECG signal (3.5hrs at 250Hz) on 64bit Windows. Will investigate further if I find time.

JanCBrammer, 2020-11-27:
The overflow issue seems to originate here: https://github.com/neuropsychology/NeuroKit/blob/39dce3def09b7ffa03a9a51d9ff23fc392171cc7/neurokit2/ecg/ecg_delineate.py#L150-L154

In line 152, depending on the content of `peaks` it can happen that `np.array(peaks)` contains integers of type `np.int32`. The subsequent multiplication of `np.array(peaks)` with `desired_sampling_rate` (which defaults to 2000 Hz) then causes `np.int32` to overflow for values larger than 2147483647 (on my machine).

The maximum values that can be represented with `np.int32` on a specific machine can be determined using

```Python
np.iinfo(np.int32)
```
```
iinfo(min=-2147483648, max=2147483647, dtype=int32)
````

Here's a minimal reproducible example:

```Python
import numpy as np

def _dwt_resample_points(peaks, sampling_rate, desired_sampling_rate):
    """Resample given points to a different sampling rate."""
    peaks_resample = np.array(peaks) * desired_sampling_rate / sampling_rate
    peaks_resample = [np.nan if np.isnan(x) else int(x) for x in peaks_resample.tolist()]
    return peaks_resample

peaks = np.array([1899120, 1899160], dtype=np.int32)
sampling_rate = 250
desired_sampling_rate = 2000

peaks_resampled = _dwt_resample_points(peaks, sampling_rate, desired_sampling_rate)
print(peaks_resampled)
```
```
[-1986909, -1986589]
```

The issue can be solved by casting `peaks` to `np.int64`. See 2fb051df12a29dd3e74c0a67a9d61c414254d007.

More generally, this is probably not the only place in the respository where integer overflow is an issue. @DominiqueMakowski, we might consider introducing [type annotations](https://realpython.com/python-type-checking/), although it's quite a bit of work to retrofit those to the entire repository.

nx10, 2020-11-27:
Thank you for for fixing this that fast!
Type annotations would be great.

Tam-Pham, 2020-11-28:
@JanCBrammer thank you for looking into this so quickly 🥳. I think the fix is great!! But I'm wondering if converting the list with nan into a np.float64 would lead to a potential loss in precision. 

What if we initially convert all nan to a negative value (-1), then specify all values to np.int64. After all calculations, we then convert all negative values to nan?

JanCBrammer, 2020-11-28:
@Tam-Pham, looking at the private functions in `ecg_delineate()` I was wondering: is there a good reason to pass around NaNs internally? If not, we could more sparsely represent the features and would not have to deal with NaNs (i.e., simply cast everything to `np.int64`).

Tam-Pham, 2020-11-29:
I think the reason for keeping the nans around is to have an equal length of all features (rpeaks, ronset, offset, tpeaks...) in the output.

DominiqueMakowski, 2020-11-29:
Probably so that there is always one entry for each detected QRS, so it's easy to identify where are problems/undetected features

nx10, 2020-12-19:
It seems like there was a small mistake in your fix @JanCBrammer:

`numpy.ndarray.astype` will still copy with `copy=False` if the `dtype` argument [requirement is not satisfied](https://numpy.org/doc/stable/reference/generated/numpy.ndarray.astype.html). (Converting an int32 array to a int64 array always needs to copy.)

you can test this very easily with something like

```Python
>>> import numpy as np
>>> a = np.array([1, 2, 3],dtype=np.int32)
>>> b = a.astype(dtype=np.int64, copy=False)
>>> a.dtype
dtype('int32')
>>> b.dtype
dtype('int64')
```

I opened a small PR that fixes this.

JanCBrammer, 2020-12-20:
@nx10, thanks for catching and fixing this! Merging your PR.

closed 2021-02-22 by Tam-Pham