neuropsychology/NeuroKit #392: Bug in ECG windowing adjustment for high heart rates?

hoechenberger, 2020-12-04:
I might be mistaken, but intuitively I would say that the high heart rate adjustments when creating ECG segment windows is incorrect. Currently it's done in the following way:
https://github.com/neuropsychology/NeuroKit/blob/9ec5674b963cdc96709105a01a8af5c2c18c1ee5/neurokit2/ecg/ecg_segment.py#L87-L98
This means that the window gets **wider** with a higher heart rate – although heartbeat duration obviously **decreases** as heart rate goes up. So, in my opinion, the sign of `c` needs to be flipped.

But maybe I am missing something fundamental here?

Best wishes,

    Richard

hoechenberger, 2020-12-04:
After some more thinking about it, the idea is probably to prevent the epochs from getting **too** short – is that it?

Tam-Pham, 2020-12-07:
Hi @hoechenberger, 

Thanks for raising the issue 😄  You can find the discussion about the adjustment here: #176 

The adjustment is simply to expand the window a little for higher heart rates. We used a constant rather than a scaling factor for the sake of convenience. Hope it clarifies.

DominiqueMakowski, 2020-12-07:
That said, I do agree that our adjustment should be double checked and potentially improved as it was a rather hacky quick 😁

hoechenberger, 2020-12-07:
Oh that makes a lot of sense, thanks for the clarification. I guess I'll just close this issue for now :)

closed 2020-12-07 by hoechenberger