neuropsychology/NeuroKit #393: Reading Data

marcoistasy, 2020-12-07:
Hello!

I've been searching the `Neurokit` documentation for quite some time now and am unsure as to how to load collected data. The data is collect into Neuralynx format but parsed to produce simple `Numpy` arrays. How can I transfer these arrays into `Neurokit`?

DominiqueMakowski, 2020-12-07:
If you have numpy arrays, you can simply convert them to a Pandas' data frame and then use the functions: 

```python
import numpy as np
import pandas as pd
import neurokit2 as nk

# Generate random array of 3 columns named X for the sake of demonstration
X = np.random.random((3, 3))

# 1. Convert to a pandas' datadrame and add names to columns
data = pd.DataFrame(X, columns=["ECG", "PPG", "EDA"])

# 2. Run NK functions
nk.bio_process(ecg=data["ECG"], sampling_rate=200)
```

That said, we could potentially add a function to directly read Neuraynx data and return a formatted dataframe with appropriate names. For that, could you maybe attach one example neuralynx file as zip here so we can try on it? Thanks

marcoistasy, 2020-12-09:
Hi @DominiqueMakowski! Thanks so much for this. Unfortunately, I'm not at the liberty to attach share the file. However, I can definitely look into creating this dataframe and then submitting a pull request to the repo for the neuralynx format.

closed 2020-12-09 by marcoistasy