neuropsychology/NeuroKit #401: Processing for Pupil dilation (Pupillometry)

csavur, 2020-12-30:
**Describe the solution you'd like**
NK is able to process most of the physiological signals, however, Pupil dilation is also used in neuroscience and other fields. Therefore, it would be a plus for NK to have Pupil processing ability.

**How could we do it?**
Here is an open-source project that only targets the pupillometric data.

https://github.com/ihrke/pypillometry 

What do you think? I think that would really nice to have in NK toolkit.

DominiqueMakowski, 2020-12-31:
That sounds like a great idea, and could be a good addition to the package. But looking at the impressive `pypillometry` package, I wonder what could we add to it, as it looks quite comprehensive and well-documented.

Maybe we could start by using it as an (optional) "dependency" (i.e., have a code chunk [like that](https://github.com/neuropsychology/NeuroKit/blob/9ec5674b963cdc96709105a01a8af5c2c18c1ee5/neurokit2/eeg/mne_to_df.py#L34-L40) in the functions) and having wrappers around it. This way, we would allow for a "neurokit-y" API at little cost (we could have functions like `pupil_process()`, `pupil_clean()`, `pupil_plot()` that would be essentially wrappers around pypillometry), which would in turn facilitates combined analysis of pupillometry with other signals. And later, we could always refactor / addin more features etc. (note that my knowledge of pupillometry is pretty much null so I am quite clueless)

If that's something you're interested to look into, feel free to open a PR with a new "pupil" submodule and maybe some basic functions like _plot() and _clean(). Are there any "simulation" algorithms outthere to simulate artificial pupillometry data? Which we could use for testing etc.

csavur, 2021-01-04:
@DominiqueMakowski, I totally agree with you. As you said we will start with as a dependency and then we will write our wrapper for now. I have some work to finish in a couple of weeks then I will spend some time implementing the wrapper. 

I am not sure if there is a simulation out there. I do not have too many experiences with pupil data as well, however, it is something I want to use in my research.

If anyone wants to help that would be really nice.

closed 2021-12-30 by stale[bot]