neuropsychology/NeuroKit #403: Missing dependency: mne

axl89, 2021-01-09:
**Describe the bug**
When following [this step of the docs](https://neurokit2.readthedocs.io/en/latest/functions.html#neurokit2.eeg.eeg_diss), the `mne` dependency is missing.

**To Reproduce**
```
(neuro) axel@axel-bravo15:~/tmp$ python3
Python 3.8.7 (default, Jan  9 2021, 20:39:51) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import neurokit2 as nk
>>> eeg = nk.mne_data("filt-0-40_raw")
Traceback (most recent call last):
  File "/home/axel/.pyenv/versions/neuro/lib/python3.8/site-packages/neurokit2/eeg/mne_data.py", line 25, in mne_data
    import mne
ModuleNotFoundError: No module named 'mne'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/axel/.pyenv/versions/neuro/lib/python3.8/site-packages/neurokit2/eeg/mne_data.py", line 27, in mne_data
    raise ImportError(
ImportError: ("NeuroKit error: mne_data(): the 'mne' module is required for this function to run. ", 'Please install it first (`pip install mne`).')
```

**Expected behaviour**
The `mne` package should be added as a package dependency. In fact, manually installing it fixes the issue:

```
(neuro) axel@axel-bravo15:~/tmp$ pip install mne
Collecting mne
  Downloading mne-0.22.0-py3-none-any.whl (6.9 MB)
     |████████████████████████████████| 6.9 MB 9.2 MB/s 
Requirement already satisfied: scipy>=0.17.1 in /home/axel/.pyenv/versions/3.8.7/envs/neuro/lib/python3.8/site-packages (from mne) (1.6.0)
Requirement already satisfied: numpy>=1.11.3 in /home/axel/.pyenv/versions/3.8.7/envs/neuro/lib/python3.8/site-packages (from mne) (1.19.5)
Installing collected packages: mne
Successfully installed mne-0.22.0
WARNING: You are using pip version 20.2.3; however, version 20.3.3 is available.
You should consider upgrading via the '/home/axel/.pyenv/versions/3.8.7/envs/neuro/bin/python3.8 -m pip install --upgrade pip' command.
(neuro) axel@axel-bravo15:~/tmp$ python3
Python 3.8.7 (default, Jan  9 2021, 20:39:51) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import neurokit2 as nk
>>> eeg = nk.mne_data("filt-0-40_raw")
>>> 
```

**Screenshots**
N/A

**System Specifications**

- OS: Linux (ELF 64bit) 
- Python: 3.8.7 
- NeuroKit2: 0.0.42 

- NumPy: 1.19.5 
- Pandas: 1.2.0 
- SciPy: 1.6.0 
- sklearn: 0.24.0 
- matplotlib: 3.3.3

JanCBrammer, 2021-01-10:
Welcome @axl89 and thanks for your contribution. For now, MNE is an _optional_ dependency. See #22 for some background. At the moment, Neurokit throws an **error** when a user without MNE in their environment uses one of the few modules that depend on MNE. Maybe it's more helpful to throw a **warning** instead? This would maybe more accurately reflect the fact that MNE is optional, i.e., missing by design.

DominiqueMakowski, 2021-01-10:
Hi @axl89 and yes, we decided to keep mne as optional to avoid the automatic downloading of mne, especially since it's used only in few features (contrary to the other deps which are used everywhere) ☺️ 

> Maybe it's more helpful to throw a warning instead?

Well but the function would fail anyway if it requires mne to do something and if mne's not installed 😕 

```
ImportError: ("NeuroKit error: mne_data(): the 'mne' module is required for this function to run. ", 'Please install it first (`pip install mne`).')
```

The printing of the message seems ugly with the parentheses etc. though 🤔

axl89, 2021-01-10:
Ah, got it. Thanks for the feedback! And great project, by the way! 💯

DominiqueMakowski, 2021-01-10:
No worries, looking forward your next PRs!

closed 2021-01-10 by DominiqueMakowski