neuropsychology/NeuroKit #405: ERROR used function load data in my own data

Wan-umaidah, 2021-01-18:
**Question and context**
I loaded my data in this path>>  C:\Users\User\anaconda3\Lib\site-packages\neurokit2\data.  Is it correct??
Then, I tried to load my data named "try_epoch" by using this function data = nk.data("try_epoch"), but it defined ERROR as attached image1
How should I solve this problem?? 
[image]
[image]

DominiqueMakowski, 2021-01-18:
Hi @Wan-umaidah, the `nk.data()` function is actually to load "preexisting" example datasets (that are included in the package). If you want to read your own dataset, you will need to use the appropriate function. For instance, if your data comes from biopac's system, and is named `try_epochs.acq`, you will have to run `read_acqknowledge("try_epochs.acq")`.

Wan-umaidah, 2021-01-18:
I want to performed Even-related ECG analysis
I recorded ECG at sampling rate = 1k, Participant was exposed in the condition of fixation (1) , Neutral (2), drug (3) randomly for 8 minutes (8x60x10000= 4800000 data points), each condition for 6 sec.
So I exported the EEG raw data from Lab chart .txt to .csv
I read the data and use find event function as attached picture but there are few errors. please correct me if I'm wrong.

[image]
[image]
[try_epoch.xlsx](https://github.com/neuropsychology/NeuroKit/files/5828736/try_epoch.xlsx)

DominiqueMakowski, 2021-01-18:
If I understand, your conditions are marked as 1, 2 or 3 at the onset of the condition correct? So there's no need to run `nk.events_find()` which is made to find onsets from a continuous "events" channel (which you don't have). 

I added a function, `nk_events_create()` to facilitate the extraction of events in your case.

1. Be sure to reinstall neurokit from the *dev* branch (https://neurokit2.readthedocs.io/en/latest/installation.html#neurokit) to have that new function
2. I ran the following

```python
import neurokit2 as nk
import pandas as pd
import numpy as np

data = pd.read_excel("try_epoch.xlsx")

x = nk.as_vector(data[["Condition"]])
           
onset = np.where(~np.isnan(x))[0]  # The index of non-empty values
conditions = x[~np.isnan(x)]  # The non-empty values themselves

events = nk.events_create(events_onsets=onset, event_conditions=conditions)
events
```
```
{'onset': array([     0,   6000,  12000,  18000,  24000,  30000,  42000,  54000,
         66000,  78000,  90000, 102000, 114000, 126000, 138000, 150000,
        162000, 174000, 186000, 198000, 210000, 222000, 234000, 246000,
        258000, 270000, 282000, 294000, 306000, 318000, 330000, 342000,
        354000, 366000, 378000, 390000, 402000, 414000, 426000, 438000,
        450000, 462000, 474000], dtype=int64),
 'duration': array([    0,  6000,  6000,  6000,  6000,  6000, 12000, 12000, 12000,
        12000, 12000, 12000, 12000, 12000, 12000, 12000, 12000, 12000,
        12000, 12000, 12000, 12000, 12000, 12000, 12000, 12000, 12000,
        12000, 12000, 12000, 12000, 12000, 12000, 12000, 12000, 12000,
        12000, 12000, 12000, 12000, 12000, 12000, 12000], dtype=int64),
 'label': array(['1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '12',
        '13', '14', '15', '16', '17', '18', '19', '20', '21', '22', '23',
        '24', '25', '26', '27', '28', '29', '30', '31', '32', '33', '34',
        '35', '36', '37', '38', '39', '40', '41', '42', '43'], dtype='<U11'),
 'condition': array([1., 2., 1., 2., 1., 3., 2., 3., 3., 2., 2., 3., 2., 2., 3., 2., 3.,
        2., 2., 3., 2., 3., 2., 3., 3., 2., 2., 3., 2., 3., 3., 2., 2., 3.,
        3., 3., 2., 3., 2., 3., 3., 2., 3.])}
```

You can then proceed to the rest of the analysis

Wan-umaidah, 2021-01-19:
I installed pip install neurokit2. It defined that there are no attribute of events_create().
So I uninstalled neurokit2 and reinstalled with C:\Users\user> pip install https://github.com/neuropsychology/neurokit/zipball/dev but it error showed on cmd
[image]

DominiqueMakowski, 2021-01-19:
Weird, it might be because the checks are not passing on devs currently, we'll fix that and let you know so that you can try again

Wan-umaidah, 2021-01-21:
Please let me know.

Tam-Pham, 2021-02-22:
hi @Wan-umaidah 

Could you install NeuroKit from the master and try again? We have recently merged the dev branch to master.