neuropsychology/NeuroKit #406: Implementation of BioSPPy ECG filtering in ecg_clean

sitadrost, 2021-01-19:
Just out of curiosity: looking at the code in _ecg_clean_biosppy and comparing it with the ECG filtering code used in the BioSPPy package, I noticed that your implementation is different (among others, you use filtfilt, where the original code uses lfilter with initial conditions).

Does this make any difference, can you explain? I haven't managed to get BioSPPy working yet, will it give the same results?

DominiqueMakowski, 2021-01-19:
As far as I'm aware we test it here:

https://github.com/neuropsychology/NeuroKit/blob/9ec5674b963cdc96709105a01a8af5c2c18c1ee5/tests/tests_ecg.py#L47-L57

And I remember it to be equivalent. But feel free to double check it :)

sitadrost, 2021-01-19:
Thanks for that, good to know that they're equivalent. Could you also explain how/why?

DominiqueMakowski, 2021-01-19:
as far as I remember, basically internally it boils down to the same functions, but written differently (especially the filters). But my memory is very bad so I don't remember the details 🤷 😅

closed 2021-03-10 by Tam-Pham