neuropsychology/NeuroKit #407: ECG signal quality classifier by Zhao (2018)

DominiqueMakowski, 2021-01-25:
This SQI takes makes use of four different quality indexes. Each one of the four indexes is classified as either optimal, suspicious or unqualified and these are then used to classify the signal according to the following function:
[image]

Using one detector makes it impossible to extract one of the features, the matching degree between the QRS, sSQI. You could adjust the decision rule and use only the other indexes three indexes (power of baseline, power of QRS and kurtosis of the signal). Any particular reason for that?

_Originally posted by @TiagoTostas in https://github.com/neuropsychology/NeuroKit/issues/124#issuecomment-766424249_

DominiqueMakowski, 2021-01-25:
Current implementation in biosppy by @TiagoTostas: https://github.com/PIA-Group/BioSPPy/pull/75

DominiqueMakowski, 2021-01-25:
It is a threshold classifier based on several indices (referred to as Signal Quality Indexes - SQI):

- Matching Degree of R Peak Detection (qSQI): The corespodnance of two different R-detection methods, in their case: *Hilbert and dynamic adaptive threshold based on R-wave detection* and *wavelet transform*.

> for wavelet transform, the input ECG signal was decomposed by discrete wavelet transform (DWT) with four layers, and the wavelet coefficients for each layer were obtained. The R point is the singularity of the ECG waveform. The extraction of the R point is completed by the correspondence between the singularity of the signal and the positive and negative modulus maximum of the wavelet coefficients to the zero point (Zhen et al., 2008)

- Power Spectrum Distribution of QRS Wave (pSQI)
- Variability in the R-R Interval (cSQI) (apparently redundant)
- Skewness sSQI and Kurtosis (kSQI)
- The Relative Power in the Baseline (basSQI)

@TiagoTostas indeed as far as I understand, it seems a bit arbitrary to require the use of 2 independent R-peak detection methods, like which ones should we use? How much does it impact the classifier? 🤔 

For now I think it'd be easier to drop the qSQI index, as it requires the user (or us) to select two detectors, for which the rationale, the assumptions and the consequences are unclear, and stick with the rest.

TiagoTostas, 2021-01-25:
From experience, I used the method by Kalidas and PanTompkins. Since they were the detectors that I found to be more robust, with detected peaks within the accepted windows.

It is true that this index is completely dependent on the detectors and it may not bring any additional information about the signal quality. Thus, dropping it may seem a good option.

Yes, the cSQI and sSQi are not used.

zen-juen, 2021-04-20:
Zhao's method has the simple heuristic approach and a fuzzy evaluation approach so dropping the indices will have some impact when using the fuzzy approach (mentioned [here](https://github.com/neuropsychology/NeuroKit/pull/410#issuecomment-822979269)) but for now in neurokit we'll drop the index involving detector comparisons and document this in our docstrings!

Oh yes also @TiagoTostas when implementing zhao's classifier in neurokit I noticed just one small error in biosppy's code in which the UbJ and UbI are swapped!

```
        # UbI
        if bassqi <= 85:
            UbI = 1
        else:
            UbI = 1.0 / (1 + np.power((bassqi - 85) / 5.0, 2))

        # UbJ
        UbJ = 1.0 / (1 + np.power((bassqi - 95) / 2.5, 2))
```
[image]

DominiqueMakowski, 2021-08-20:
closing as I'm think it's implemented

closed 2021-08-20 by DominiqueMakowski