neuropsychology/NeuroKit #409: Implementing detrending to the rri time series?

noa7291, 2021-02-04:
First of all,  this is a great package and very easy to use! In many papers it is reccomened to detrend the rri time series before analysis, but i couldn't find how to do it in neurokit. additionaly, for sanity check, I want to compare my results from neurokit to the results i get from kubios. Right now the results are comparable only if i disable the detrending option in kubios, so i'm guessing that it is not implemented in neurokit default analysis procedure. 
I've tried to use signal_detrend on the rri time series after the _hrv_get_rri but it resulted with NaN on all the measured of frequency domain analysis and the results of the time domain were really different and didn't make any sense. 
any thoughts how to implement detrending in the analysis pipeline?

Tam-Pham, 2021-02-04:
Hi @noa7291 

Thank you so much for highlighting the issue to us. To help us quickly reproduce and diagnose the issue, could you share your current processing script (the one that you try to apply `signal_detrend()` on) as well as the dataset that you are analyzing?

Many thanks!

noa7291, 2021-02-04:
this is the code of hrv_get_rri

`def my_hrv_get_rri(peaks=None, sampling_rate=1000, interpolate=False, **kwargs):
    rri = np.diff(peaks) / sampling_rate * 1000
    if interpolate is False:
        return rri

    else:

        # Minimum sampling rate for interpolation
        if sampling_rate < 10:
            sampling_rate = 10

        # Compute length of interpolated heart period signal at requested sampling rate.
        desired_length = int(np.rint(peaks[-1]))

        rri = signal_interpolate(
            peaks[1:],  # Skip first peak since it has no corresponding element in heart_period
            rri,
            x_new=np.arange(desired_length),
            **kwargs
        )
        rri = nk.signal_detrend(rri, method="polynomial", order= 1)
        return rri, sampling_rate`

the only change i've tried is to add the detrend function to the interpolated signal. for the time domain analysis - i tried two versions - 1) set interpolate = True and using the above script and 2) set interpolate = False and move the detrend function  right after the line - "rri = np.diff(peaks) / sampling_rate * 1000"

zen-juen, 2021-02-09:
Hello @noa7291 sorry for the late response! Could you point us to the paper that recommended detrending of the RRI time series? We just want to check if the detrending is done on the raw signal or the RRI itself. Thanks!

noa7291, 2021-02-09:
for example in the smooth priors method - 
Lipponen, J. A., & Tarvainen, M. P. (2019). A robust algorithm for heart rate variability time series artefact correction using novel beat classification. Journal of medical engineering & technology, 43(3), 173-181.‏

https://www.tandfonline.com/doi/citedby/10.1080/03091902.2019.1640306?scroll=top&needAccess=true 

i think it is done on the rri series.

Tam-Pham, 2021-02-22:
Dear @noa7291 

Upon reading the documentation of [Kubios](https://www.kubios.com/downloads/Kubios_HRV_Users_Guide.pdf), I do recognize that a lot of the calculations of the HRV parameters are being done on the detrended IBI, except the mean RR, mean HR, and max HR values. There are quite a few detrending options in Kubios and the default seems to be the smoothness priors method (smoothing parameters = 500). 

For NeuroKit, we do perform a detrending step on the RRIs before calculating the different frequency domain analyses. However, we are currently doing a simple constant detrending (subtracting mean) and have yet to provide users with options to choose a different method of detrending. Since the detrending of RRI in `signal_psd()` is currently hardcoded and there is no option to turn in off, your script might have detrended the signal twice (once with linear detrend and one with constant detrend)

> the only change i've tried is to add the detrend function to the interpolated signal. for the time domain analysis - i tried two versions - 1) set interpolate = True and using the above script and [...]

I wouldn't suggest setting the interpolation to False as interpolating and detrending are 2 different steps. To better compare the performance of NeuroKit and Kubios, especially for the frequency domain analysis, I would suggest you to try setting the detrending option in Kubios to a polynomial with order 0 (though I'm not sure if it supports this option) so that the detrending is comparable. 

In the meanwhile, we would discuss whether to include different options of detrending for RRIs in the pipeline.

DominiqueMakowski, 2021-02-22:
What are the benefits / issues with detrending RRIs?

closed 2022-08-15 by stale[bot]