neuropsychology/NeuroKit #412: Computing HRV indices with missing data

danibene, 2021-02-08:
Hi! 

I would like to compute HRV indices for heartbeat data, given peaks that I have already detected (so not e.g. raw ECG data). 
```
# Compute HRV indices
hrv_info = nk.hrv(peaks, sampling_rate=100, show=False) 
```

I understand that NeuroKit takes a dataframe containing the samples, with each peak being 1 and the other samples being 0, for example:

```
ECG_R_Peaks
0	0
1	0
2	0
3	0
4	0
5	0
6	0
7	0
8	0
9	0
10	0
11	0
12	0
13	0
14	0
15	0
16	0
17	0
18	0
19	0
20	0
21	0
22	0
23	0
24	0
25	0
26	0
27	0
28	0
29	1
...	...
50652	0
50653	0
50654	0
50655	0
50656	0
50657	0
50658	0
50659	0
50660	0
50661	0
50662	0
50663	0
50664	0
50665	0
50666	0
50667	0
50668	0
50669	0
50670	0
50671	0
50672	0
50673	0
50674	0
50675	0
50676	0
50677	0
50678	0
50679	0
50680	0
50681	0
```

Now in my case, there are some anomalies in the peak detection that I would like to discard for the HRV analysis. Is there a way to indicate that in the input dataframe so that these anomalies are excluded when computing the HRV indicies? 

Many thanks,
Danielle

zen-juen, 2021-02-09:
Hi there @danibene thanks for reaching out! We'll look into this asap - meanwhile, is there any reproducible data that you can share with us?

zen-juen, 2021-02-09:
@danibene Could you elaborate on what anomalies that you derived after the peak detection? And when you say discard these anomalies, do you mean just dropping the sample points from the data completely? Thanks 😄

danibene, 2021-02-09:
Hi @zen-juen thanks so much for your quick response :)

Here is an example of a vector with the times at which each peak was detected:

```
import numpy as np
example_peak_times = np.array(
    [
        0.119274376417238,
        0.679297052154197,
        1.20428571428572,
        1.77424036281180,
        2.32253968253968,
        2.85260770975057,
        3.35117913832200,
        3.94757369614513,
        4.57070294784580,
        5.16630385487528,
        5.67140589569161,
        6.21131519274377,
        6.77292517006802,
        7.36725623582767,
        7.95437641723356,
        8.54471655328798,
        9.10018140589570,
        9.65478458049887,
        10.2223129251701,
        10.7868480725624,
        11.4124036281179,
        12.0131972789116,
        12.5928117913832,
        13.1535600907030,
        13.7065759637188,
        14.2671201814059,
        14.8328344671202,
        15.4220861678005,
        16.6072108843537,
        17.0678231292517,
        17.6607709750567,
        18.2908616780045,
        18.8431292517007,
        19.3832879818594,
        19.9221995464853,
        20.4735147392290,
        21.0520634920635,
        21.6962811791383,
        22.3354875283447,
        22.9748299319728,
        23.6463945578231,
        24.3152607709751,
        24.9865532879819,
        25.6258049886621,
        26.2452380952381,
        26.8763945578231,
        27.4955782312925,
        28.1114058956916,
        28.6122675736961,
        29.2399773242630
    ]
)
```

Here is the BPM plotted, without anomalies removed:

```
import matplotlib.pyplot as plt
example_ibis = np.diff(example_peak_times); # interbeat intervals
example_ibi_times = example_peak_times[1:len(example_peak_times)]; # peak times corresponding to each interval
example_bpms = 60./example_ibis # beats per minute
plt.plot(example_ibi_times,example_bpms) # plot original
```

[image]

I assume that the suddenly low BPM (around 50) is the result of a false negative in the peak detection, so I want to discard that part of the interbeat interval/BPM vectors:

```
from scipy import stats
zscores = stats.zscore(example_bpms) 
nan_ibis = np.where(np.logical_or(zscores > 3, zscores < -3), np.nan, example_ibis) # replace anomaly with NaN value
nan_bpms = np.where(np.logical_or(zscores > 3, zscores < -3), np.nan, example_bpms) 
plt.plot(example_ibi_times,nan_bpms) # plot with anomaly removed
```

[image]

Now to answer your question about what I mean by "discard". I think for time-domain HRV indices like RMSSD, this would just mean excluding that interbeat interval from the calculation, though I'm not sure how missing data should be dealt with for the other types of features. Does that make sense?

zen-juen, 2021-02-10:
For now, our `nk.hrv_time()` which computes time-domain HRV indices doesn't accommodate the discarding of IBIs because theoretically the anomalous peaks would have been discarded during the initial cleaning phase of `nk.ecg_peaks()`. But if you would like to specifically discard IBIs based on a threshold you can modify the nk.hrv_time() function like so (I added the **# Discard** section where you discarded the IBIs based on zscores of the bpms):

```
def hrv_time(peaks, sampling_rate=1000, show=False, anomalies=False):

    # Sanitize input
    peaks = _hrv_sanitize_input(peaks)

    # Compute R-R intervals (also referred to as NN) in milliseconds
    rri = _hrv_get_rri(peaks, sampling_rate=sampling_rate, interpolate=False)
    
    # Discard
    bpms = 60./rri # beats per minute
    zscores = scipy.stats.zscore(bpms) 
    rri = np.where(np.logical_or(zscores > 3, zscores < -3), np.nan, rri)
    rri = rri[~np.isnan(rri)]
        
    diff_rri = np.diff(rri)

    out = {}  # Initialize empty container for results

    # Mean based
    out["RMSSD"] = np.sqrt(np.mean(diff_rri ** 2))

    out["MeanNN"] = np.nanmean(rri)
    out["SDNN"] = np.nanstd(rri, ddof=1)
    out["SDSD"] = np.nanstd(diff_rri, ddof=1)

    # Normalized
    out["CVNN"] = out["SDNN"] / out["MeanNN"]
    out["CVSD"] = out["RMSSD"] / out["MeanNN"]

    # Robust
    out["MedianNN"] = np.nanmedian(rri)
    out["MadNN"] = mad(rri)
    out["MCVNN"] = out["MadNN"] / out["MedianNN"]  # Normalized
    out["IQRNN"] = scipy.stats.iqr(rri)

    # Extreme-based
    nn50 = np.sum(np.abs(diff_rri) > 50)
    nn20 = np.sum(np.abs(diff_rri) > 20)
    out["pNN50"] = nn50 / len(rri) * 100
    out["pNN20"] = nn20 / len(rri) * 100

    # Geometrical domain
    bar_y, bar_x = np.histogram(rri, bins="auto")
    out["TINN"] = np.max(bar_x) - np.min(bar_x)  # Triangular Interpolation of the NN Interval Histogram
    out["HTI"] = len(rri) / np.max(bar_y)  # HRV Triangular Index

    if show:
        _hrv_time_show(rri)

    out = pd.DataFrame.from_dict(out, orient="index").T.add_prefix("HRV_")
    return out
```
and then based on the `example_peak_times` you have provided, convert them first into samples and then calculate the HRV indices

```
# Set sampling rate of data
sampling_rate=1000

# Convert example_peak_times to example_peak_samples
example_peak_samples = np.round(example_peak_times*sampling_rate, 0).astype(int)

# Compute hrv
hrv = nk.hrv_time(example_peak_samples, sampling_rate=1000, show=True)
```
Assuming that your sampling rate is 1000 you should then get an output like this! You can also do a comparison of the indices calculated before and after the **# Discard** section was added into the function to see if it meets your expectations
```
hrv
Out[8]: 
   HRV_RMSSD  HRV_MeanNN   HRV_SDNN  ...  HRV_pNN20  HRV_TINN  HRV_HTI
0  48.999783       582.0  47.528715  ...  54.166667     211.0      3.2
```

closed 2021-03-10 by Tam-Pham

danibene, 2022-02-04:
Hello, 

First of all, thank you so much for the detailed response & sorry for my very late response! 

I'm re-opening this because I think there should be adjustments made for specific features & I noticed that [a similar question has been raised since](https://github.com/neuropsychology/NeuroKit/issues/541).

For the features that are computed based on successive differences (e.g. RMSSD in hrv_time or SD1SD2 in hrv_nonlinear), I would exclude the differences between non-successive intervals. Here is an example code based on the current version of [hrv_time](https://github.com/neuropsychology/NeuroKit/commit/e13d8081437fca876fb27191c723bc8d3b6abcb0):

```
# -*- coding: utf-8 -*-
import matplotlib.pyplot as plt
import numpy as np
import pandas as pd
import scipy.stats

from neurokit2.stats import mad, summary_plot
from neurokit2.hrv.hrv_utils import _hrv_get_rri, _hrv_sanitize_input

def hrv_time(peaks, sampling_rate=1000, show=False, check_successive=False, **kwargs):
    """Computes time-domain indices of Heart Rate Variability (HRV).
     See references for details.
    Parameters
    ----------
    peaks : dict
        Samples at which cardiac extrema (i.e., R-peaks, systolic peaks) occur.
        Can be a list of indices or the output(s) of other functions such as ecg_peaks,
        ppg_peaks, ecg_process or bio_process.
    sampling_rate : int, optional
        Sampling rate (Hz) of the continuous cardiac signal in which the peaks occur. Should be at
        least twice as high as the highest frequency in vhf. By default 1000.
    show : bool
        If True, will plot the distribution of R-R intervals.
    Returns
    -------
    DataFrame
        Contains time domain HRV metrics:
        - **MeanNN**: The mean of the RR intervals.
        - **SDNN**: The standard deviation of the RR intervals.
        -**SDANN1**, **SDANN2**, **SDANN5**: The standard deviation of average RR intervals extracted from n-minute segments of
        time series data (1, 2 and 5 by default). Note that these indices require a minimal duration of signal to be computed
        (3, 6 and 15 minutes respectively) and will be silently skipped if the data provided is too short.
        -**SDNNI1**, **SDNNI2**, **SDNNI5**: The mean of the standard deviations of RR intervals extracted from n-minute
        segments of time series data (1, 2 and 5 by default). Note that these indices require a minimal duration of signal to
        be computed (3, 6 and 15 minutes respectively) and will be silently skipped if the data provided is too short.
        - **RMSSD**: The square root of the mean of the sum of successive differences between
        adjacent RR intervals. It is equivalent (although on another scale) to SD1, and
        therefore it is redundant to report correlations with both (Ciccone, 2017).
        - **SDSD**: The standard deviation of the successive differences between RR intervals.
        - **CVNN**: The standard deviation of the RR intervals (SDNN) divided by the mean of the RR
        intervals (MeanNN).
        - **CVSD**: The root mean square of the sum of successive differences (RMSSD) divided by the
        mean of the RR intervals (MeanNN).
        - **MedianNN**: The median of the absolute values of the successive differences between RR intervals.
        - **MadNN**: The median absolute deviation of the RR intervals.
        - **HCVNN**: The median absolute deviation of the RR intervals (MadNN) divided by the median
        of the absolute differences of their successive differences (MedianNN).
        - **IQRNN**: The interquartile range (IQR) of the RR intervals.
        - **pNN50**: The proportion of RR intervals greater than 50ms, out of the total number of RR intervals.
        - **pNN20**: The proportion of RR intervals greater than 20ms, out of the total number of RR intervals.
        - **TINN**: A geometrical parameter of the HRV, or more specifically, the baseline width of
        the RR intervals distribution obtained by triangular interpolation, where the error of least
        squares determines the triangle. It is an approximation of the RR interval distribution.
        - **HTI**: The HRV triangular index, measuring the total number of RR intervals divded by the
        height of the RR intervals histogram.
    See Also
    --------
    ecg_peaks, ppg_peaks, hrv_frequency, hrv_summary, hrv_nonlinear
    Examples
    --------
    >>> import neurokit2 as nk
[... 167 lines cut ...]

# Set sampling rate of data
sampling_rate=1000

# Convert example_peak_times to example_peak_samples
example_peak_samples = np.round(example_peak_times*sampling_rate, 0).astype(int)

# Compute hrv, without checking for successive RR intervals
hrv = hrv_time(example_peak_samples, sampling_rate=1000, show=False, check_successive=False)
print(hrv["HRV_RMSSD"])

# Compute hrv, checking for successive RR intervals
hrv_check_successive = hrv_time(example_peak_samples, sampling_rate=1000, show=False, check_successive=True)
print(hrv_check_successive["HRV_RMSSD"])

```

Which gave me the following output:

```
0    48.999783
Name: HRV_RMSSD, dtype: float64
0    45.793013
Name: HRV_RMSSD, dtype: float64
```